iNavFlight/inav #7675: Glide Time and Range Calculation

bishko, 2021-12-20:
## Current Behavior
This option is not available at the moment.

## Desired Behavior
The idea is to have a calculation running in the background, such that when the airplane begins to have a negative vertical speed, the FC calculates how much time you have approximately before you reach the ground level. 
Given a ground speed, you can also calculate the glide range of the aircraft. 

## Suggested Solution
For now, I have experimented a little bit with the programmable logic and I have come up with the following:

### **Calculating Glide Time**
**Altitude [cm] / Vertical speed [cm/s] = Glide time [s]**
logic 0 1 -1 17 2 12 2 13 0
**Converting Glide time from [s] to [min]**
logic 1 1 -1 17 4 0 0 60 0
**Setting GVAR0 to Glide time in [min]**
logic 2 1 -1 18 0 0 4 1 0
### **Calculating Glide Range**
**Converting Ground speed from [cm/s] to [m/s]**
logic 3 1 -1 17 2 9 0 100 0
**Converting Ground speed from [m/s] to [m/min]**
logic 4 1 -1 16 4 3 0 60 0
**Time [min] x Converted ground speed [m/min] = Glide range [m]**
logic 5 1 -1 16 4 1 4 4 0
**Converting Glide range from [m] to [km]**
logic 6 1 -1 17 4 5 0 1000 0
**Setting GVAR1 to Glide range in [km]**
logic 7 1 -1 18 0 1 4 6 0

I am aware that this is probably not the best way (e.g. rounding errors, etc.), but since there are no decimal places and there is a limit to the value of GVARs, I had to convert Glide time and Glide Range to [min] and [km], respectively.

In addition, the caluclaiton above is in [km] which is not too useful. Ideally, the units should change just like the home distance for example, once you go above 1km.

If you think that having this is a good idea and believe that it is worth to program the code for a new OSD feature as opposed to displaying GVAR0 and GVAR1, please include something like this in future versions of INAV, as part of the OSD. 

## Who does this impact? Who is this for?
Primarily, his would be a good feature for soaring.

The fature would also be great for people who would like to fly off with a glider in a particular direction, and then use the altitude they gained to glide back without power. 

Similarly, this gives people who would like to catch thermals, the opportunity to gauge how much time and what distance they can glide using the elevation they have gained. It could for example be used for thermal-hopping.

Most importantly, this fature could be integrated as part of a FailSafe type of situation, where the battery has drained below a certain level. If sufficient elevation has been gained, i.e. Glide_Distance > Distance_To_Home, then the RTH could work without engine power, and attempt to bring back the model safely with the last bit of juice left in the battery, rather than attempting to land in a remote location. Usually servos still work after the ESC has stopped working.

## Additional context
PS: This is 123AMG from the YouTube live stream.

MrD-RC, 2021-12-27:
I've started looking at this. But not tested anything yet.

https://github.com/iNavFlight/inav/tree/MrD-glider-assistance-OSD-elements

https://github.com/iNavFlight/inav-configurator/tree/MrD-Add-OSD-elements-for-gliders

MrD-RC, 2021-12-28:
Initial coding is done. But I need to test in flight.
[image]

avsaase, 2021-12-28:
I think you're going to need some filtering on these values. You divide by a noisy variable that is close to zero which will amplify the noise in the output. For the glide slope element, I used a 0.5Hz cutoff frequency.

For the glide range, I would add the same `glideSeconds > 0` check as you did for the glide time because when you are climbing, `glideSeconds = 0` and the glide range is infinity instead of zero.

MrD-RC, 2021-12-28:
@avsaase  I did see the filter in the glideslope. I guess it's the vario that's noisy, coming from the baro?

Good point about the range. I've added that in.

avsaase, 2021-12-28:
Maybe it is not really noise but just that the vertical speed has a large range that includes zero. For example, it can go from -2m/s to -0.1m/s in one second, so then the glideslope/distance/time increases by a factor of 20. Especially around the zero vertical speed point, the division results in rapidly changing values.

MrD-RC, 2021-12-28:
Makes sense. I’ve added the filter to the vertical speed.

bishko, 2022-02-26:
MrD-RC, thank you very much for adding this OSD item! It looks exactly as I imagined it! I really appreciate it! 
Unfortunately I have not had a chance to upgrade my frimware and enable this feature yet, but towards the end of March, I should be able to offer some feedback at least. :))

MrD-RC, 2022-03-01:
> MrD-RC, thank you very much for adding this OSD item! It looks exactly as I imagined it! I really appreciate it! Unfortunately I have not had a chance to upgrade my frimware and enable this feature yet, but towards the end of March, I should be able to offer some feedback at least. :))
It's not been added yet. I'm still working on it, as I'm not too happy with how it's working. I need to try it on a calmer day with a more stable craft to get some better measurements.

PR #8081 (2022-07-28): Add glide time, glide distance, and climb efficiency OSD elements
This PR adds OSD elements for glide time, glide distance (Fixes #7675), and climb efficiency.

Requires Configurator change https://github.com/iNavFlight/inav-configurator/pull/1539

[Demo](https://youtu.be/vEOpl1yXouk)
closed 2022-07-28 by MrD-RC