iNavFlight/inav #7686: Matek h743-slim fails to connect to configurator

ryet9, 2021-12-26:
## Current Behavior
Will not connect to configurator, DFU mode connects seamlessly, 3.0.2 works as intended every time

## Steps to Reproduce

1. flash 4.0.0 with full chip erase 
2. connect USB  **many times**
3. watch it time outs **many times**
4. cry deeply **many times**

## Expected behavior
a connection to the configurator

## Suggested solution(s)

## Additional context
the same FC with firmware 3.0.2 will connect every time

---

can't get to CLI

- Matek h743-slim
- INAV version string:  4.0.0

ZePapou, 2021-12-28:
I have quite the same behavior with this H743 slim v1.5 (Inav V4.0)
- when **no** SDCARD inserted : everything fine
- when SDCARD inserted : incorrect boot (blue led not flashing) - no connection with configurator
It was working with 3.0.2
I did not find anything on the net
Any help will be appreciated :)

DzikuVx, 2021-12-29:
Tested both cases with my H743-slim and everything works fine 

[image]

No problems with flashing, SD card, etc

@ryet9 please verify if you have correct STM32 VCP driver installed
@ZePapou check if SD card is correctly placed and it's working with other FCs