iNavFlight/inav #7688: Matek H743 Slim no current values in OSD  since update to INAV 4.0 Stable

terryfritz99, 2021-12-26:
## Current Behavior
Matek H743 Slim flight controller isn't reporting a value for current in telemetry since update to INAV 4.0

## Steps to Reproduce

1.Update firmware from INAV 4.0 RC2 to INAV 4.0 Stable
2.
3.
4.

## Expected behavior
My setup had been reporting the value for current flow in RC1 and RC2 releases.

## Suggested solution(s)
 No suggestion

## Additional context

https://pastebin.com/cuX3MRKG

---
****
INAV/MATEKH743 4.0.0 Dec 15 2021 / 15:18:48 (c20a9852)
# GCC-10.2.1 20201103 (release)

- FC Board name and vendor: Matek H743 Slim
- INAV version string:INAV/MATEKH743 4.0.0 Dec 15 2021 / 15:18:48 (c20a9852)
# GCC-10.2.1 20201103 (release)

terryfritz99, 2021-12-27:
Sorry: forgot to say I have two quads to the same specs and both show the same problem.

ryet9, 2021-12-28:
Interesting - I have two of these & they wont even connect after flashing with 4.x, yet 3.x is fine