iNavFlight/inav #7707: High CPU load on fresh build with SD card installed

wx4cb, 2021-12-31:
Brand new F765 WSE with a 16 gig UHC3 sd card in the sd card slot, i get 75% cpu usage and cannot get into anything  whhen the card is inserted. card removed evrything is fine.

tested with multiple sized and speeds of card, including working BB SD cards from other airframes with the same controller.

thoughts?

DzikuVx, 2021-12-31:
[image]

More details would be needed. Everything seems to work just fine over here

DzikuVx, 2021-12-31:
[image]

wx4cb, 2021-12-31:
i know, that's whatI dont understand... I can't even get to the CLI when it does it as the configurator just takes absolutely ages. i'm going to try again later today

DzikuVx, 2021-12-31:
@wx4cb it's super important that SD card is correctly placed in the slot. If there is any dirt inside it will cause problems

clemmmdawg, 2022-01-01:
I had a similar issue with my F765 Wing FC this morning. CPU usage was spiking to 1900% and trying to use the SpeedyBee configurator kept giving timeout errors. [Here ](https://www.youtube.com/watch?v=NG6F_ZyjyQo) is a video I got of the CLI while I was waiting to be able to type in the "tasks" command to see what was causing the CPU spike.

This was with a fresh 16GB PNY card and the issue was resolved simply by removing the card. The card now has a 0 byte FREESPAC.E file. Could iNav be having an issue with writing to the new card?

wx4cb, 2022-01-01:
it could just be that the cpu is tied up writing the freepsace file. I reformatted the card in guiformatter to make sure that it had a valid fat32 filesystem on it, put it back in the FC and just left it run... after about 5 minutes it stopped complaining and usage went back to normal.

This could be just a side effect. but doesn't explain why it was doing ti with a card that was already used for BB from another 765 wing.

clemmmdawg, 2022-01-01:
>put it back in the FC and just left it run... after about 5 minutes it stopped complaining and usage went back to normal
Just did the same thing with the same results. Plugged in via USB and left alone for 5 or so minutes. Came back and everything is working perfectly. I can also confirm the FREESPAC.E file on the card is now 4GB.

MorrisR2, 2022-01-01:
Should we add a message or something when it's doing this, so the user knows what's going on?
Something like "Creating FREESPAC.E, please wait 5 minutes"?

wx4cb, 2022-01-01:
> Should we add a message or something when it's doing this, so the user knows what's going on? Something like "Creating FREESPAC.E, please wait 5 minutes"?

that would be an idea, or even just (and i dunno the right term so i'll "linuxify" it) "nice" the process so that it doesn't hog the cpu

MorrisR2, 2022-01-01:
> 

It would be handy if we could "nice" the process, which tells the scheduler adjust the priority. Unfortunately, there is no process. And there is no scheduler.  It looks like similar COULD be done by spawning a protothread and calling PT_YIELD() at the appropriate place.

0crap, 2022-01-02:
> > 
> 
> It would be handy if we could "nice" the process, which tells the scheduler adjust the priority. Unfortunately, there is no process. And there is no scheduler. It looks like similar COULD be done by spawning a protothread and calling PT_YIELD() at the appropriate place.

That would make the mess even worse....

0crap, 2022-01-02:
Tested on a Matek F405-STD and I don't see this issue.

```
# version
# INAV/MATEKF405 4.0.0 Dec 15 2021 / 14:53:05 (c20a9852)
# GCC-10.2.1 20201103 (release)
```

Phil-MC, 2022-12-08:
I have a user on Discord who is having this problem with INAV 5.1 on a SPEEDYBEE F405 V3. He says the quad flies great on default without the SD card inserted but is unusable as soon as it is inserted. I will see if he has high CPU usage. Perhaps the above solution will work for him...

Pedrolomel, 2023-01-27:
Related issue:

https://github.com/iNavFlight/inav/issues/8733#issue-1557271328

closed 2023-10-27 by DzikuVx

Phil-MC, 2023-10-28:
Thanks PawelSent from Outlook for Android