iNavFlight/inav #7712: Landing mode can send the multirotor UP rather than down

MorrisR2, 2022-01-01:
This ZIP file contains the blackbox log (last flight shows the problem), the FPV video with OSD, and the config dump:
https://drive.google.com/file/d/1iU-_5WKC81tqQp6C36wR3m4Pn-eui2Pr/view?usp=sharing

[##Current Behavior

During RTH landing (navOnEnteringState_NAV_STATE_RTH_LANDING), the craft descends initially, then at around 8 meters it shoots UPWARDS. This pattern repeats - descends to 7-10 meters, then goes back up again.

The altitude at which it reverses direction doesn't *seem* to precisely match the Slow Down Altitude or the Final Approach Altitude.

## Steps to Reproduce

I'm not sure why mine does this and others don't.  My steps are:
1. Take off from my front yard
2. Fly several meters away from home
3. Engage RTH with a switch

After having this problem more than once, I added a message to try to help troubleshoot it.
The video shows an OSD message "landing error 3".  The message "landing error 3" means this code was triggered that I added to navOnEnteringState_NAV_STATE_RTH_LANDING:

```
        float old_z = posControl.desiredState.pos.z;
        updateClimbRateToAltitudeController(-descentVelLimited, ROC_TO_ALT_NORMAL);
        if ( posControl.desiredState.pos.z > old_z + 1) {
            landing_error = 3;
            }
        }
```

## Expected behavior
In landing mode, the target altitude should decrease, not suddenly increase.
That is, the craft should go down, not up, during the landing sequence.

## Suggested solution(s)

There are actually two solutions needed.  Mainly we'd like a solution that would prevent it from happening by addressing the root cause.  Because the root cause is not yet known, I've implemented this solution to sanity-check the value and mitigate the problem:

```
        float old_z = posControl.desiredState.pos.z;
        updateClimbRateToAltitudeController(-descentVelLimited, ROC_TO_ALT_NORMAL);
        // Sanity check - go down, not up, during landing
        if ( posControl.desiredState.pos.z > old_z + 1) {
            landing_error = 3;
            if (navGetCurrentActualPositionAndVelocity()->pos.z < 200.f) {
                posControl.desiredState.pos.z = old_z - 1;
            } else {
                posControl.desiredState.pos.z  = old_z - 2;
            }
        }
```

## Additional context

---

- FC Board name and vendor: OMNIBUS F4 (AIO) V6
- INAV version string: # INAV/OMNIBUSF4V6 3.0.2 Dec 19 2021 / 17:01:53 (4af5bca0)

The same problem was seen with 4.0.0, as I recall. Reviewing the changes between 3.0.2 and 4.0.0.0, I don't see any relevant changes.

Jetrell, 2022-01-02:
I've noticed similar discrepancies with RTH land as well.

My quads all had their landing settings tuned -

`nav_land_maxalt_vspd = 430 `            _Descend from home altitude at -4.3m per sec_
`nav_land_slowdown_maxalt = 500`     _At 5m above the ground, start slowdown_

`nav_land_slowdown_minalt = 150`      _At 1.5m above the ground, be a touchdown minalt_vspeed ( 0.5m per sec)_
`nav_land_minalt_vspd = 50 `               _Touchdown speed_

This used to work consistently on this particular quad. Now with 4.0, I can't get it to slowdown with these settings...  Or it touches down then flies back up again.  And if it does land ( 1 time in 5) it very often doesn't auto disarm anymore, because it still thinks its in the air.

The issue seems to be caused by poor altitude calculation, which is making landing hit and miss. 

I loaded a build that I had from back on _September 18th_ and it works consistently. So the problem seems to have been introduced after that time.

I have also noticed that the Navigation positional accuracy is lacking consistency. This particular quad would fly back to home and get within 1m of the home position and then start descent within a few seconds. 
Now it can sit up there for 20+ seconds before it creeps its way to with 1m of home.  And this doesn't seem to be wind related. 

I have increased the _Vel_XY_ P and I  by 1 and reduced the D by 3.   This has helped... but it still has its moments.

MorrisR2, 2022-01-02:
@Jetrell  Is that Sept 18 build from master?
That's very roughly around the time the landing code had some changes. I don't *think* that is responsible, but it's possible. I'd love to compare the source code that works vs the earliest that doesn't work.

Jetrell, 2022-01-02:
> @Jetrell Is that Sept 18 build from master? That's very roughly around the time the landing code had some changes. I don't _think_ that is responsible, but it's possible. I'd love to compare the source code that works vs the earliest that doesn't work.

Yes it was. Here is that file if you would like to compare it with 4.0.
[inav_4.0.0_MATEKF722MINI.zip](https://github.com/iNavFlight/inav/files/7798466/inav_4.0.0_MATEKF722MINI.zip)

When I did the back to back compares. I did 5 RTH Land flights of both, and rebooted between flights. They were all at the same time and location.
The Sept 18th build, landed and disarmed every time. With good Altitude accuracy +-2m.
While 4.0 build only landed once. That after `nav_land_minalt_vspd`  occurred at about 7m from the ground (visual line of sight). Which was 1m on the OSD altitude.  Needless to say, it took a while to touch down.

MorrisR2, 2022-01-02:
@Jetrell  Just to verify:
3.0.2 was released on Sept 15.
4.0.0 was release December 17.
The file you linked is named inav_4.0.0.  Is that the right file?  Doesn't seem right for Sept 18.
Also I guess you just downloaded the compiled hex, not the code, right?

Jetrell, 2022-01-02:
Its a nightly test build that one of the iNav developers release on his site- http://seyrsnys.myzen.co.uk/inav_dev/
It had a patch that I wanted to test out at the time.

Using 4.0. I notice the times that the quad wouldn't slow down at all, and bounced back up into the air, the Altitude would show it rise again, while the vario was showing an increasing negative value, even when it started to shoot back up into the air. as high as -3.5m per second.  That doesn't seem right, but I haven't had this issue happen before. Lucky this is a small quad, it was practically a crash impact.
Does yours do anything like that?

MorrisR2, 2022-01-02:
I just rewatched my fpv video with OSD and sure enough the vario shows high negative as the altitude shows it's increasing.

I didn't have that much foam on my baro.  I suspect it's _possible_ that a change to the code made it react this way to wind noise in the baro.  How's your foam?

I have a full-proof way of finding the exact code change that makes the difference for you, if you're up for some testing. I can build a hex file halfway in-between Sept 18th master and 4.0.0.  See if that works for you. If so, try one halfway between that point and 4.0. We'll find the exact change that causes the problem for you.

Note to self:
The changes are https://github.com/iNavFlight/inav/compare/d8c8fac0c8fb34405ffaccb6628ca43c9edeae64...4.0.0

Jetrell, 2022-01-02:
I have small cell foam rubber over my baro. 

I also have a few more development builds from Sept 22, Oct 31 and Nov 7 that I can try.  But I wont be able to do so in the near future. Today was my day out :-)

Maybe @breadoven maybe able to chime in. 

In case anyone else reads and is wondering. I also tried the SPL06, the equivalent of the DSP310. They both performed the same.

MorrisR2, 2022-01-02:
Funny you mention @breadoven  I was wanting to talk to them.  Checked Discord - Breadoven isn't online.

breadoven, 2022-01-02:
Looking at the log there seems to be significant vibration at higher motor speeds which is possibly affecting measured vertical velocity (and probably altitude). Vertical velocity in the log is generally negative even when the baro is indicating a climb (assuming the baro is correct). This will obviously cause problems with the quad thinking it's descending faster than it is causing it to throttle up to the point it actually climbs when it should be descending.

Not sure why this should be an issue now when it wasn't before, something changed affecting position estimator ?

MorrisR2, 2022-01-03:
@breadoven  One thing that may have made a difference is the descent code was rewritten around six months or so ago. The newer code may be more easily affected by whatever physical problem. Or there might be a math issue with negative numbers, zeroes, etc. Not sure.

What are your thoughts about my proposed couple lines of code for sanity check - during landing descent the target position should never increase, it should never go higher than the target was for the last loop. Seem like a reasonable sanity check?

Also Breadoven, is there a way to send you a private message, Discord or something? I wanted to mention something not you that doesn't need to be part of this issue.

Jetrell, 2022-01-03:
> 
> Not sure why this should be an issue now when it wasn't before, something changed affecting position estimator ?

@breadoven Thanks for the feedback.
Just from what I've read and experienced myself and from the people I fly with. There certainly seems to be position estimator issues. Both with planes and multicopters.

Fixed wing Poshold loiter is under and over shooting the `nav_rth_home_altitude` target, by as much as 20m, regardless of crosswinds, as it always did.

While position targeting seems to be okay. It appears to be slow, causing strange behavior in missions at some waypoints. 

To add to this. I have also noticed discrepancies in GPS home positioning in 4.0.
e.g. The GPS can have a fix and be showing Zero meters for the home location. Then once it arms, that value can jump up to 4m away.

breadoven, 2022-01-03:
> @breadoven One thing that may have made a difference is the descent code was rewritten around six months or so ago. The newer code may be more easily affected by whatever physical problem. Or there might be a math issue with negative numbers, zeroes, etc. Not sure.
> 
> What are your thoughts about my proposed couple lines of code for sanity check - during landing descent the target position should never increase, it should never go higher than the target was for the last loop. In other words "never try to go up during descent". Seem like a reasonable sanity check? It seems to solve the problem for me.
> 
> Also Breadoven, is there a way to send you a private message, Discord or something? I wanted to mention something to you that doesn't need to be part of this issue.

Does your sanity check solve the problem entirely ? I'd have thought you would still have a problem with the erroneous climb rates confusing throttle control causing erratic vertical speed control ?

You could contact me on Discord, I look in every now and again.

physje, 2022-01-09:
Same experience here with a Matek 722SE.

Last week, went back to INAV 3.0.1 and made a basic setup which fly fine.
Made a backup of the settings (diff), loaded INAV 4.0.0 on the FC and restored the saved settings from INAV 3.0.1.

RTH on 'low' altitudes is executed as expected, at higher altitudes the same problem as described above is seen.

If needed I can provide the backlog-files and setting-file.

No problem :
# INAV/MATEKF722SE 3.0.1 Jul  4 2021 / 14:31:07 (753c4479)
# GCC-9.3.1 20200408 (release)

problem :
# INAV/MATEKF722SE 4.0.0 Dec 15 2021 / 15:12:41 (c20a9852)
# GCC-10.2.1 20201103 (release)

MorrisR2, 2022-01-09:
> Does your sanity check solve the problem entirely ? I'd have thought you would still have a problem with the erroneous climb rates confusing throttle control causing erratic vertical speed control ?

It seems to completely resolve the problem. Before, it would set the new target altitude higher than the previous target altitude, initiating a climb. With the sanity check, it doesn't try to climb.

b14ckyy, 2024-03-20:
looks like this was fixed? If not @ me and I reopen.

closed 2024-03-20 by b14ckyy