iNavFlight/inav #7716: Launch and Idle Throttle do not match set value

Mateyhv, 2022-01-03:
An Inav user brought my attention to the fact that "launch throtte (us)" was overshooting the user set parameter. He is on 4.0 and I tested it on 3.0.2 with the same outcome so apparently the error leaked upstream.

From my test on 3.0,2 without any kind of expo on throttle.

Launch Throttle set to **1850us**, real resulting throttle was 100% aka >**2000us**.
Launch Throttle set to **1800us**, real resulting throttle was   94% aka **1940us**.

Equaly Idle Throttle shows some difference:
Idle Throttle set to **1420us**, real resulting throttle was 47% aka **1470us**.

For some reason both settings do not match the input value.
Thanks

MrD-RC, 2022-01-03:
That looks about right to me. With a range of 1000 - 1850:
- 1850 is 100%
- 1800 is 94.1%
- 1420 is 49.4%

I’m sure the throttle command uses that percentage. It doesn’t use uS.

breadoven, 2022-01-04:
Throttle percentage is based on `max_throttle `and [idle throttle](https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#throttle_idle) not 0 to 2000, i.e. it's the percentage of the range from the set idle value to max throttle setting.

See also https://github.com/iNavFlight/inav/pull/7435.

Mateyhv, 2022-01-05:
Hello Darren! My min max thr range is 1200 - 1800 but thats Autothrottle shouldn't be related to launch throttle in any way. Yes I assume 1850 uS to be 85% of full throttle. Today I tried 1770 uS (77%) and got 90% launch throttle, confirmed by the BBox. Looks rather strange.

I might have a clue though. I had Sag Compensated Vbat activated and the results after doing the maths are within 1% of the throttle given by the FC in both cases. The data is from the Bbox log with a 5s lion battery wich as expected sags quite a bit:

Launch idle:                                                      Launch Throttle:
1420uS (set) -> 47% (exec. by FC)                    1770uS (set) -> 90% (exec. by FC)
19.90v (real) -> 20.51v (compensated)             17.25v (real) -> 20.50v (compensated)

If someone can prove my assumption is correct there are pros and cons of having the Vbat Compensation active during Launch and we may consider them if there is some interest.

Mateyhv, 2022-01-13:
Apparently as @breadoven said the problem comes from the max and min throttle which has to be set to 1000 and 2000 for a correct scaling.. Yet to do some tests but for now I will close the issue. Thanks

closed 2022-01-13 by Mateyhv