iNavFlight/inav #7717: OSD Switch Indicator (for Flaps, Gear, etc.)

stuwe911, 2022-01-04:
I'd like to suggest the creation of a OSD expansion so that it is possible to display a switch setting in the OSD. In order to keep it simple I suggest to have given channels where this applies and users will have to change their servos accordingliy i.e.:

CH7-CH9 2-position switch
CH10-CH12 3-position swich

This would allow users to visualize special functions in OSD like Flaps, Gear, Cargo Door, etc.
A simple graph could then be displayed in OSD and you have visual control of your setting without having to touch the switch

[image]

[image]

a1axx, 2022-01-04:
This is cool, 

There are characters that could be used already, I am sorry, but can't remember what these are!
[image]

StuweFPV, 2022-01-04:
can't see any characters on your pic that would serve the purpose.. looks more like AHI to me :)
even better would be if there are some letter attached to it like F=Flaps, G=Gear in front of the symbol. Similar to what we now have when displaying the PID profile selection with P.
Even if there are usable characters available i never saw any switch in the OSD to turn either a given function like Flaps on or a given channel. how would you display and assign your characters in the OSD?
cheers mate

MrD-RC, 2022-01-04:
Yep, thats the AHI. I think you’d only need the icons for the 3 position switch. They would only show the high and low positions for 2 way switches. 

It would be simpler to implement the channel number and the switch icon. Once you start adding letters, the setup becomes more complex. Perhaps a couple of options for flaps, gear, doors, then select the channel number? It would need to be simple for setup on the OSD page though.

Mateyhv, 2022-01-05:
Also have to be considered how many OSD spaces have to be used for the switch display. Here is the available list:

1 OSD char - 3 new icons needed - economical on OSD space, but probably the image will be too small?
2 OSD char - 6 new icons needed - middle size which might be good but icon hungry.
3 OSD char - 2 new icons needed - highly visible, fewer icons needed but eats some OSD space.

StuweFPV, 2022-01-05:
> Also have to be considered how many OSD spaces have to be used for the switch display. Here is the available list:
>
> 1 OSD char - 3 new icons needed - economical on OSD space, but probably the image [...]

good point thank you. i guess that kicks out the option of a letter in front of it. I think most users will use max. 3x this feature and it should be fairly "detectable" by the icon itself. i.e. gear is the one with only 2 pos. flaps the one with 3. and the other one, somewhere stacked i a special corner of the osd would be the "special function" whatever that might be. cam recording or cargo bay open.
I would actually have to see and test before making a final call which size is best. i understand the limit of the char in osd available by the font but form your reading option #2 would probably be the nice middle ground :)

Mateyhv, 2022-01-06:
I did some icons for 2 character which I also think is a good compromise of space usage and visibility on the OSD. You can scroll throuhg the different positions to see how it looks which I think is quite good. I began with separators which look good for individual switches but when indicators are stacked they look messy, so I eliminate them. You can have a look at both stacks with and w/o separators to understand what I mean. Simbols are oversized so they serve equally well for 2 and 3 position switches.
[image]
[image]
[image]
[image]
[image]
.

StuweFPV, 2022-01-06:
wow!! that's very good work! I guess they could work but to be honest i would prefer just a filled square like in my drawing. I found that fine stuff is hard to see sometimes especially when video reception is not at its best. just keep them in the corner of my eye and not really looking at it or just a very brief glance. however this is sure going to work. i like the circle if you use up/down. for flaps that would be weird. which font is that in/based on? convertable to any font? I use clarity medium.
cheers mate!!

Mateyhv, 2022-01-06:
It is not based on an existing font but can fit any. Initially I was about to do the squares but preferred the signs for several reasons: 

1. they are clrearly identified by the position inside the rectangle AND by the shape so even in a glance you will grasp whats going on, specially when several are stacked together. Squares alone are Ok but stacked create checkerboard images making them more awkward to read at a glance.
2. View is less obstructed, as 80 percent of the rectangle is tranparent but that doesn't impact the information recevied.

The circle uneqivocaly identifies the center position, of course can be changed for a dash or a square. The signes does not identify flaps, gear, etc, they just gives you the position of the switches. The triangles I will probably enlarge a bit to increase visibility, don't worry about size they are larger than most of the icons available.

Here is an example with squares and with signs to compare
[image]
[image]
:

StuweFPV, 2022-01-06:
yes two very valid points. i was not planning on stacking them. i would probably only use two of those symbols since my plane only has one set of gear :) but offering different version would give users a whole range of options they could use. if it's no biggie to change that then i would prefer for the 2-pos switch arrow-up and arrow-down. for the 3-pos the circle is not ideal if the middle position represents a active setting. maybe the circle at the top, in the middle a 1 and at the bottom a 2? representing the flap positions.
if you could merge them into clarity medium i'd be very happy to test them in real conditions and provide a feedback/video.
thanks!

OptimusTi, 2022-01-06:
Something more like this:
[image]

This could be cool for Programming logic too. Global vars?

Mateyhv, 2022-01-06:
When designing features for Inav you have to think in general so that the feature fits anyone and given the scarce free chars available its impossible to fit every case eg: flaps, gears, cameras, lights, custom functions... some wont stack the indicators, others will. In short you have to design the features globaly. At any time you can change the icons you want and compile them for your own needs.

I don't remember where I got the font compiler, it was some time ago but probably @MrD-RC can help you as he reworked the fonts recently.

I have done one more set making the triangles bigger and adding a square in place of the circl, so now the same icons fit nicely 3 and 3 position switches.

The big problem you have now is to find someone who wants to write the code for Inav and the configurator... :-)

[image]
[image]
[image]
[image]

StuweFPV, 2022-01-06:
> Something more like this: [image]
> 
> This could be cool for Programming logic too. Global vars?

I love that!!! super cool!! I wish we had color...

Mateyhv, 2022-01-06:
> > When designing features for Inav you have to think in general so that the feature fits anyone and given the scarce free chars available its impossible to fit every case eg: flaps, gears, cameras, lights, custom functions... some wont stack the indicators, others will. In short you have to design the features globaly. At any time you can change the icons you want and compile them for your own needs.
> > I don't remember where I got the font compiler, it was some time ago but probably @MrD-RC can help you as he reworked the fonts recently.
> > I have done one more set making the triangles bigger and adding a square in place of the circl, so now the same icons fit nicely 3 and 3 position switches.
> > The big problem you have now is to find someone who wants to write the code for Inav and the configurator... :-)
> > [image] [image] [image] [image]
> 
> that's always hard to please everyone and everyone's needs.. especially in osd - the liking is so individual. ok I'll ask Darren if he can help me out with the designer. yes the switch is another issue.. I thought just a font upload.. but you're right. i have to be able to activate it.. I think Darren did something on LQ's. I'll ask. thanks for your effort.

Don't worry, if this feature sees the light I will make a set of icons for you exactly as you wish ;)

StuweFPV, 2022-01-06:
ok thanks. so compiling is the issue now... let's see what Darren says. he's super busy and probably many other priorities. maybe i'll start learning to compile myself.. who knows. it's cool if you can do your own stuff

MrD-RC, 2022-01-06:
> Something more like this: [image]
> 
> This could be cool for Programming logic too. Global vars?

That would be pretty cool and only need three characters. I like @Mateyhv icons with the triangles too.  But being 2 OSD squares high, it would need 6 characters. But also I think it would make it less attractive from a usability perspective. I think L S (Letter Symbol) inline, taking up 2 OSD blocks would be more usable and better for OSD real estate. But the triangles and circle looks cool. 

Compiling the fonts is a piece of cake. You just run a file and it does everything. Can’t remember the name of the file, as I set it up on my batch file. 

I did start looking at code. I think I have an implementation in my head now.

@OptimusTi the first 4 GVars can already be shown on the OSD.

StuweFPV, 2022-01-06:
Hurra MrD-RC added it as a feature request - problem solved :) thank you for that!! I think many people will use that.
For simplicity I think Optimus's icon would be a good start even though the arrows also have their advantages. A letter in front would be perfect so it's clear what it is but not mandatory. I'm sure you have a great idea in your head.

no worries about the compiler - you're much faster and better. I'd rather be your test pilot :)
cheers mate!!

OptimusTi, 2022-01-06:
> That would be pretty cool and only need three characters. I like @Mateyhv icons with the triangles too.  But being 2 OSD squares high, it would need 6 characters. But also I think it would make it [...]

Yes, using 5 characters but I see what you mean by 3 characters. As for gvars I was thinking about operating the symbols with them but nah. It would be simpler operating them by assigning a channel and checking the values.

Mateyhv, 2022-01-07:
OptimusTi icon looks cute but its misleading, it has color which improves visibility, its not dimensioned and by no means its easy to spot on the OSD in difficult conditions as StuweFPV asked. And its 5 font chars vs 6 font chars. Thats why the double character was preferred although its character expensive. Here are both versions in real size. Of course they should be tested on the OSD to check for real sensations
[image] [image]
.

StuweFPV, 2022-01-07:
i'm sure the colors are misleading - would've been so nice though :) in the option you show i think the smaller one is too small on here but we have to see it in the goggles before making a call. I like the idea when you can see the "empty" fields as well - making it clear which position active. framed squares either filled or empty would just be good enough if you don't use the arrows.

MrD-RC, 2022-01-07:
@Mateyhv it would be interesting to see how well the first option with the squares shows up on the OSD. To me, that is the better solution as it uses less font characters and can be used horizontally. I think with the 2 screen character version, it would have to use a vertical orientation, with the function character above the switch indicator . Which would be a little odd, and could make positioning more difficult.

i have an idea for the characters, but won’t be able to draw it up until tonight. It may not work though. Was thinking about having the active vs inactive as different proportions. So what’s happening may be more visible with a single screen character.

OptimusTi, 2022-01-07:
> OptimusTi icon looks cute but its misleading, it has color which improves visibility, its not dimensioned and being a single character is not easy to spot on the OSD in difficult conditions as StuweFPV asked. And its 5 font chars vs 6 font chars. Thats why the double character was preferred although its font expensive. Here are both versions in real size. Of course they should be tested on the OSD to check how well they actually look.
> 
> 
> 
> [image] [image]

That's not color. Blue is transparent. Based off existing symbols.

MrD-RC, 2022-01-08:
Here are a couple of thoughts for switches. The grey colour is the transparency.
[image] [image]
Seeing them next to each other, I quite like the 2nd one. It looks more like a transmitter switch. But would need to see how it looks in the goggles. What do you guys think?

Examples with function:
[image][image] Flaps
[image][image] Gear

MrD-RC, 2022-01-08:
Another example, using the triangles and circle on a single character
[image][image]

Mateyhv, 2022-01-08:
I like the second one either. It has the advantage of shape change which coupled with the black and white vertical line gives a clear indication of the position. The F and G are nice but limiting to two functions only and need 3 icons for each letter. 

On the last one I understand what you have done extending the range of the 3D stick but clipping the shape does not work well: the stick looks much smaller on the edges, and if taken as a triangle it points to the opposite direction of where the stick which is counterintuitive. Not sure about the black dashes but the vertical line on the upper icons make an easier guess on the sticks position. Good job Darren 👍

MrD-RC, 2022-01-08:
Yeah, I think the 2nd one looks the best. I’ll try it on the OSD. But based on size next to the letters, I think it will be fine for visibility. 

@Mateyhv I don’t get what you mean by the letters limiting to 2 functions, or needing 3 icons for each letter? The F in the example is just to show that the switch is for the flaps. So you would have the F followed by the switch indicator. The switch indicator would be mapped to an RC channel. My thought is to have both the letter and RC channel selectable by the pilot. Maybe have 4 switch indicators that could be enabled. This should have plenty of scope.

Mateyhv, 2022-01-08:
@MrD-RC excuse me my bad, On the phone looked like the letters had a frame but its just a regular font. Great

MrD-RC, 2022-01-08:
The plan is going ok at the moment. Hit a bit of a snag with the configurator. The data-setting method seems to hang when the data type is char, so I need to look in to that. But this is the plan.

[image]
On the left, the regular toggles to enable or disable the OSD elements.

[image]
Then on the right, the settings area to choose a letter and channel that the switch is on.

On thing's for sure, I need to sort out that alignment with the inputs vs select boxes on the right side of the OSD page. Don't know how I've not noticed that before. It looks pretty crap with that misalignment.

MrD-RC, 2022-01-08:
Sorted the MSP stuff. Strings weren't being processed at all, so it just threw an exception 🤣 The configurator side is done. Just the OSD stuff, then we can see how it looks.

[image]
[image]

MrD-RC, 2022-01-08:
[You may find this interesting...](https://youtu.be/_8EkgQCx12E)

StuweFPV, 2022-01-08:
wow now that's some nice work in no time! well done Darren! looks super cool and the option to assign your own channel and letter opens this up for many users with all sorts of applications.
I really like the 2-pos switch. indicating nicely with the pear shaped form up and down.
on the 3-pos switch i'd say this is good for a regular application where you have off in the middle and up/down represent a function. for flaps however it is less ideal as the neutral position is at the top. should be 0 there and then the pear shape forms in the middle and at the bottom. now this is very specific and would almost require a special flaps switch meaning create a 3rd type which will probably complicate the whole thing. on one side i say leave it as it is. on the other hand i think flaps is probably the most widely used "additional" function out there. even if it's only flaperon.. i'll have that on about 70% of my planes whereby gear only on 10% these days. tough call - up to you. 
great work an i'm happy to test up in the air to see how visible and useful this really is!

MrD-RC, 2022-01-08:
I get what you’re saying about flaps. But this has to be universal. It’s would add a whole load more to get that done, as well as take away more resources. Even though the icons may not be exactly what would fit that specific purpose. It would still be easy to see the status of the flaps as it is.

StuweFPV, 2022-01-08:
yeah fully agree. let's keep them as universal as you can so users will use it. it surely gets the job done! nice work
now lets test up in the air. i need F405/722 targets :) :)

Mateyhv, 2022-01-08:
Excellent job Darren! I have to admit I like the result very much! Now wondering for blank space instead of letter is it enough to left the field blank? And in case its blank is the space usable? Thanks

StuweFPV this is not a specific visual style switch, its just shows the switches position on your TX regardless of their function. But you definitely can exchange them for any icon you like, you can take my rectangle icon above, separate in three and substitute the ones Darren did.

StuweFPV, 2022-01-08:
@Mateyhv 
ok thank you understood. i still need Darren to produce the hex and the configurator :) I have planes ready waiting and can test tomorrow up in the air. feedback video incl :)

MrD-RC, 2022-01-19:
@Mateyhv The ranges are the same as in the programming:
- **HIGH** > 1666
- **MID** between HIGH and LOW
- **LOW** < 1333

I don't quite follow about the blank spaces. Back then, there was only one character. Now you can have up to 4 characters. The switch is on the right, and the text is right aligned. So that the characters will be closest to the switch. Maybe I could add an alignment option, though this would not be represented in the configurator.  Possibly could be, but I'd have to have a little think.

Mateyhv, 2022-01-19:
@MrD-RC thanks, yes the range is as expected. The blank space question was simply if a letter is not used can multiple switches be put close together side by side? I guess the configurator will show the overlapping positions in red but not a big deal if that works.

MrD-RC, 2022-01-19:
Perhaps that’s something that can be checked on the firmware side. To only write the switch to the OSD if there is no text.

MrD-RC, 2022-01-19:
The Configurator side works. I could potentially make the OSD field a variable length; As it can be updated on the OSD, so the layout will always be right.

https://youtu.be/R8MRLwLB4Qo

MrD-RC, 2022-01-20:
Got it working. Nice and customisable now. Ignore the slight glitch when removing the LT. This will only happen during editing, and wouldn't ever be visible in flight. To be honest. Unless people have a live OSD while editing, it would never appear.

https://youtu.be/A7vcS6twd2w

Mateyhv, 2022-01-20:
Excellent!!! 👌

a1axx, 2022-01-21:
Awesome work. Il be finding a use for these!!

StuweFPV, 2022-02-12:
So sad to see that this new feature did not make it into 4.1 as it is all ready to go.. I could've made good use of that.

MrD-RC, 2022-02-12:
There are font changes, so needs a major version increase with the new version numbering system. Plus 4.1 has broken it 🤣 I need to get it fixed.

MrD-RC, 2022-02-12:
Thanks mate, I make the easy changes. But it allows the real devs to work on things that need an understanding of complex maths or electronics. I wish I knew more of both, but thats the way it is. 

I spent a couple of hours (well, about 8) working on the units conversions. Vertical speed will be in m/s or ft/s going forwards.

happy12, 2022-03-26:
This feature will be so awesome! Thanks for making it!

PR #7734 (2022-04-19): Add switch position indicators to the OSD
This PR adds four switch position indicators to the OSD. Fixes #7717 
Requires configurator change https://github.com/iNavFlight/inav-configurator/pull/1429

Each switch position indicator can be enabled or disabled like any other OSD element. They are also configurable on the right of the screen. This allows the pilot to assign a letter to the switch indicator, and the channel that it watches.

[This is a demo of the PR working](https://youtu.be/_8EkgQCx12E)

As a side effect, strings are now handled by the MSPHelper for the data-setting method of com [... cut]
closed 2022-04-19 by MrD-RC