iNavFlight/inav #7719: Blackbox times out with Matek F405mini with inav 4.0

rickgode, 2022-01-04:
Current Behavior
Selecting the blackbox logs in inav 4.0 configurator on a Matek F405 mini just times out.  (I have two and it does it on both) It works fine in inav 4.0 on a Matek F405SE and a FlywooF411V2 FC.

Steps to Reproduce

1.  Update Matek F405 to inav 4.0
2.   Try and access the blackbox tab in inav configurator 4.0
3.   Watch it just say Waiting for data... and time out.

Expected behavior
Should be able to access the blackbox logs tab and download the blackbox files

## Suggested solution(s)

## Additional context
inav 4.0 dump https://pastebin.com/NviVrzv3 

- FC Board name and vendor:  INAV/MATEKF405 4.0.0 Dec 15 2021 / 14:53:05 (c20a9852)
- INAV version string: GCC-10.2.1 20201103 (release)

stronnag, 2022-01-04:
Duplicate of https://github.com/iNavFlight/inav-configurator/issues/1420

stronnag, 2022-01-04:
What does CLI `get blackbox_device` report?

rickgode, 2022-01-05:
# get blackbox_device
blackbox_device = SPIFLASH
Allowed values: SERIAL, SPIFLASH, SDCARD

jc9aj, 2022-01-11:
I'm having the identical issue with the new Matek F405 MiniTE. The 4.0.0 Configurator shows "Dataflash: free 16.0MB" in the header but when I click the blackbox page just says "waiting for data..."  
```
# version
# INAV/MATEKF405TE 4.0.0 Dec 20 2021 / 14:09:08 (GITDIR-N)
# GCC-10.2.1 20201103 (release)

# get blackbox_device
blackbox_device = SPIFLASH
Allowed values: SERIAL, SPIFLASH, SDCARD

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216
Paritions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=0
```

Flashed it with BF 4.2.10 and also tried INAV 3.0.2 and all seems to work in those builds

rickgode, 2022-01-15:
Okay, so I really think this one is a bug.  @jc9aj if you type in "set blackbox_device = SDCARD" it will open up and allow you to enter the blackbox tab, then when you select "On-board dataflash chip" in the dropdown, it will time out and just say "waiting for data..."

It also says that our flight controller does not have a compatible dataflash chip available when you finally get in the Blackbox menu, so there's that.

Rick
[image]
y

mulfo44, 2022-01-26:
I had the same issue with the miniTE with not being able to get into the menu tab.  Matek has a fix they compiled last night with a new firmware target that gets the black box menu working again.  Email them and they'll send you the target firmware, I don't think they uploaded it yet on their website.  I tried uploading it here but .hex files aren't supported on this.

rwindsorjr, 2022-03-08:
Same issue with 4.1.0 using the Maytek Site.

rickgode, 2022-04-09:
Just following up again.  I updated to inav 4.1.0.  @mulfo44 I haven't tried Matek's fix, I've only used the builds in configurator.

Ricky

rwindsorjr, 2022-10-11:
I have used the Matek fix, from the support page of their site and it works fine . The Black Box file is safe and useable. INAV 4.01 works well now been flying it. Just need to adjust the PIDS a bit. Used this FC to replace a HGLC 722 mini FC . on a REKON5

helicocrasher, 2022-11-27:
@rwindsorjr can you please indicate where to find the matek F405 mini blackbox fix?  Board code MKF4?
Matek F405 mini is hanging in configurator blackbox tab.
Thanks

rwindsorjr, 2022-12-01:
sorry slow to respond . It says right in the question.  it was rolled into the Matek Target if I remember correctly been a while. I started out with INAV 4.0 . It was a PITA . mine worked fine after that.  I never got it tuned. IM Done with Quads . Came from fixed wing. And returned. Cant waste the time and money.  Working on a AtormRC
dolphin.        405 mini is more reliable than the HGLRC 722.  Sorry .

rickgode, 2022-12-01:
Yeah, but it's not, the target listed on Matek's website is the same one you download in the inav configurator.  I'm on 5.1 and the blackbox still times out.  I think this FC is discontinued by Matek so it might not get much more development here.

helicocrasher, 2022-12-02:
@MATEKSYS could you please indicate how to patch inav F405mini for blackbox to work.
I guess the knowledge is available F405miniTE had the same issue

rwindsorjr, 2022-12-03:
I dug it out and no help. I had converted it to External Logger. Because the built in was to small.   It worked before but now the FC overheats on the USB power within minutes.  I was told about the fix on Discord .  Sorry Had a few strokes and it may come to me but not now.

closed 2023-10-27 by DzikuVx

rickgode, 2025-01-23:
Good news!  It looks like @mmosca figured out what the deal was and came up with a fix.  Hopefully this fixed MatekF405 target will get merged.  Here is a video of it working.

https://youtu.be/-r8S9HklKRQ

Ricky