iNavFlight/inav #7720: WP-Mission as RTH option

StuweFPV, 2022-01-04:
## Current Behavior
As we get more and more RTH options it makes it even harder to forsee which one to pick. It's not only about getting home safely but also avoiding certain areas like villages, freeways, mountain rigs, etc. When i fly longer range i know where i go and i know which way home at which alt would be best. selecting one of the existing RTH options seems always a compromise and guesswork. In my case it's often so that when coming home straight i would crash into a mountain ridge and loose video signal but i could nearly glide around the rigde, keeping my video whilst descending. Current RTH can't do that.

## Desired Behavior
Let RTH fly a predefined WP route (not backtracking a existing one). Any WP-Route should be selectable as the "RTH-Route" for flexibilty

## Suggested Solution
Add RTH option "RTH-WP" and let me select a mission number. I could have 2-3 predefined routes coming home which ideally are selectable from the stick-menu prior to flight. I could define the exact track the plane flies, the decent rate until home is reached. then proceed with standard RTH Home operation (loiter, land, etc.). 
One additional option i suggest would be "go last WP past" or "go nearest" depending on my mission i could select the best one suited.

## Who does this impact? Who is this for?
Especially long range pilots that fly in complicated areas but also pilots that fly in dense areas with lots of villages, power lines, roads or whatever you want to avoid.

## Additional context
To me this seems to be by far the safest way to get my plane home when flying in more challenging areas. Adding more complexity and options to the RTH function does not necessarily make it safer but increases the chance of a missconfiguration. With more WP options and routes available now, this would make a good use of it. It would be simple to understand for everyone, reliable and straight forward but it would also require more planing prior to the flight or when changing flying area. But this should not replace the existing RTH options so one can always revert back to standard settings.

JulioCesarMatias, 2022-01-04:
have you tried Safe-Home?

StuweFPV, 2022-01-04:
yes but that has nothing to do with the way a plane comes home. safehome only changes where home is. a bit more sophisticated from the old "home offset"

Jetrell, 2022-01-04:
@StuweFPV a similar feature was suggested a while back. #962

JulioCesarMatias, 2022-01-04:
> yes but that has nothing to do with the way a plane comes home. safehome only changes where home is. a bit more sophisticated from the old "home offset"

so what you want is something like ArduPilot's Smart RTL, right?

StuweFPV, 2022-01-05:
> @StuweFPV a similar feature was suggested a while back. #962

yes i've read that but it's different. that's why i explicitly wrote "no backtrack". I want it simple. you go to a place, know where you fly and know the best way home. setup a WP-mission and mark this one as RTH-WP mission. with backtrack.. imagine i go flying for 45min somewhere and then i get a failsafe. the plane would fly those 45min back and home even if home migto only be 10min away. otherwise you'd have to add some complicated logic by picking out the "nearest to home points" and creating a track based on where you've been. I'd like a very simple solution. All it takes make WP a RTH option, lets you define which mission is active for RTH and select "nearest or last" to define where the plane should start the way back.

StuweFPV, 2022-01-05:
> > yes but that has nothing to do with the way a plane comes home. safehome only changes where home is. a bit more sophisticated from the old "home offset"
> 
> so what you want is something like ArduPilot Smart RTL, right?

can't say - don't know ArduPilot that well.

StuweFPV, 2022-01-05:
Let me try to visualize this with a example but this is similar to how i really fly 80% of the time. Lets say I'm at 0m alt on take off. the green path is my flying path to get to that place at 1'000m. I need to climb to maintain LoS because that ridge on the left, which is let's say 800m, is blocking the valley. I can fly nicely in the green circle but not below 950m. if i then have a failsafe the current RTH will take the red path. at the first cross i will loose video signal and at the second the plane will crash into the mountain. assuming i have set linear descent. if i keep the alt or even climb before RTH then i will be 1'000m above my head and in airspace of real planes - if the battery makes it. waiting for the plane to loiter down 1'000m and praying that i don't get a gyro drift or some wind gusts or whatever else (have you ever tried letting your plane RTH-loiter more than 10-times? - then you know what i mean)
If now the RTH-WP could follow the amber line (WP's), keeping just enough alt for the video to stay alive but descending, staying away from trees in case it makes a emergency landing so that i have a chance for recovery and maybe a plane in one piece then it would be all good. reaching home 100m above my head and land her without goggles. In my real life scenario i could even glide home if the battery is that dead - enough for FC and servos - but only if it takes the right path.

[image]

breadoven, 2022-01-05:
This wouldn't be RTH so much, more of a WP failsafe mode or WP mode with an option to start at the nearest WP rather than the first (used in the case you get lost/disorientated and manually trigger WP mode to get back home).

Could be done relatively easily just using the current WP mission loaded before arm. More difficult to implement if you wanted to have a RTH-WP mission and a normal WP mission active at the same time.

Mateyhv, 2022-01-05:
Very interesting indeed. Looks helpful in mountaineous terrain where you can easily go behind a terrain feature and the only clear path is to return backwards rather than stright back home.

However its also restricts your flying area and great caution has to be taken when planning where you CAN fly. Eg: if in the above picture the pilot decides to cross the ridge to the left of the red line, in the event of a WP-RTH it will be a dissaster hitting the ridge, while a stright RTH will bring you back safely. In this regard a stright RTH is much more flexible.

StuweFPV, 2022-01-05:
> This wouldn't be RTH so much, more of a WP failsafe mode or WP mode with an option to start at the nearest WP rather than the first (used in the case you get lost/disorientated and manually trigger [...]

I see your point that having 2 WP-missions "active" could be a challenge for implementation. Of course this would be a loss since i use regular WP-missions quite a bit. But if i'd have the option i would trade the "normal" mission vs a RTH-WP mission most of the time. nevertheless having both options would be the "perfect" solution.

start at "nearest" is only in certain cases ideal. in other scenarios going back to the "last nearest" that you've passed on your way up would be better. hence the request for letting you decide before flying which one to pick.

I did not write this request just out of the blue. i've put quite some thought in it based on years of mountain flying with many RTH incidences and various RTH configurations. I thought a lot about what would be a easy but reliable way to bring the plane home safely. but this is not only about bringing the plane home then sometimes it just will not make it due to battery or a mechanical defect, it's also about directing the plane over a save path and in some cases accessible terrain in case it goes down on the way home. just last year a friend lost a plane during RTH and it's always a very awkward feeling when driving up to the area and you look out for any siren and blue lights.. what did it hit? we knew the direct route home was over houses and roads... I'd rather have the plane not making it home but go down in a river or field than getting closer to home but crash into the neighbour village. so big safety aspect behind this idea for everybody not flying only in the desert.

appreciate your flagging this as a "feature request"!

StuweFPV, 2022-01-05:
> Very interesting indeed. Looks helpful in mountaineous terrain where you can easily go behind a terrain feature and the only clear path is to return backwards rather than stright back home.
>
> However [...]

yes i fully agree. straight RTH gives you more flexibility and less to think about. but that's not the request. if you suddenly decide lets go over there and have a look, it could be fatal. but flying longer range in mountains you're always well advised preparing your flights properly, knowing your area and the altitudes and stick to the flight plan. this is not for the spontaneous pilot to goof around but more for the "mission" pilot. and these flights do require proper planning for many other reasons. it's not the "serves it all" solution. that does not exist in my view. that's why we have so many RTH options now. it's a addition to the existing ones for a very specific type of flying with lots of safety in mind that could help others as well (i.e. city flyers) as described above.

MrD-RC, 2022-01-06:
I think this would be a great addition. I guess on the failsafe page you’d have RTH and Waypoint RTH as options?

I’ve no idea how the missions work or anything about their implementation. Do they have header information, so could be defined as standard or RTH in the header? I guess it would also have to fall back to regular RTH too if an RTH mission couldn’t be found.

I can definitely see this as being useful to RC pilots though.

StuweFPV, 2022-01-06:
yes that would be nice having it on the RTH page as a option with the sub-option "nearest or last passed". A very good point! yes regular RTH would have to kick-in if there is no "RTH mission" loaded or invalid. thanks for your feedback!

Mateyhv, 2022-01-07:
> I think this would be a great addition. I guess on the failsafe page you’d have RTH and Waypoint RTH as options?
>
> I’ve no idea how the missions work or anything about their implementation. Do they [...]

Shouldn't RTH and RTH-WP be a separate modes? As I see it stright-in RTH should always be available.

StuweFPV, 2022-01-07:
> 

do you mean in the Adjustment tab? i never configure RTH there - always failsafe. I set RTH in the failsafe tab

StuweFPV, 2022-01-07:
gents i was just playing around a bit with the programming. as a plan B or alternate solution in case this idea never makes into a release. what do you think?

- If i set failsafe to "do nothing"
- In adjustments define CH5 to WP mission 1600-2000pwm
- if I get my RSSI chanel's value of <900pwm / or CRSF LQ <40% and set this on LC1
- then channel override CH5 to 1900pwm if LC1 is active
(if seen the option "is failsafe" in the programming tab but can't figure out how to just ask the system if its actually in failsafe. surely there is a better way of programming but this is only about the idea)

would it then not trigger the active WP mission in case of a signal loss?? I know this is not proper failsafe management and state of the art but would the effect not be the same? of course having the problem that it would start on WP1 no matter what which is far from ideal unless there is a option "start WP nearest"?
quick and dirty - but do you guys think this could work?

MrD-RC, 2022-01-07:
> Shouldn't RTH and RTH-WP be a separate modes? As I see it stright-in RTH should always be available.

Of course, it could be set up as modes too. But I see the faulsafe part as more essential, along with the fallback to regular RTH. While you’re still in control of the craft, you can fly it back however you want. But either way, once the code is there, it being activated via failsafe or mode are both possible.

StuweFPV, 2022-01-07:
yeah and how do i get out once link is established again.. basically re-write the entire failsafe logic. not a very good idea. rather live with what we have and hope for a proper new solution officially.

MrD-RC, 2022-01-07:
> do you mean in the Adjustment tab? i never configure RTH there - always failsafe. I set RTH in the failsafe tab

No, on the failsafe page. Thinking about it. It may be better to keep RTH as the failsafe method. But have a checkbox with an optikn like

Jetrell, 2022-01-07:
@StuweFPV  I like the concept of having another means of RTH. Not just a straight line.
But the only part that concerns me with the current idea, is that it has to be planned as carefully as any waypoint mission for it to work correctly. 
And the specific flight plan has to be based on the possibility of RX signal loss at the location when WP_RTH would take over.  
I don't personally see many people using it because of that complexity of setup. But the concept is still good!

Could the same thing be accomplished by using the new `On the Fly WP_Planner,` when its altitude PR is added?
This would be more user friendly and plot the coarse and altitude on the path you took to get to the location that RX signal is lost.
Obviously, it would require the WPs to be thinned along the way, while always maintaining the highest points and points that a coarse deviation took place.

I know that could make for a long trip home, but the odds of completing the whole WP_RTH mission is unlikely. Its more about getting us out of a tight spot, until RX signal or video returns.

StuweFPV, 2022-01-07:
> > do you mean in the Adjustment tab? i never configure RTH there - always failsafe. I set RTH in the failsafe tab
> 
> No, on the failsafe page. Thinking about it. It may be better to keep RTH as the failsafe method. But have a checkbox with an option like “Use Waypoint RTH if available”. That way, it’s implying that regular RTH should still be set up and that the waypoint RTH will only be used if one is available, and meets safety parameters

yeah i like that. having WP-RTH as prio one and if that fails, mission not loaded, then regular RTH is activated based on the existing settings. only sub-option to add would be "nearest or last-passed".

JulioCesarMatias, 2022-01-07:
I don't like the idea of ​​a WP-RTH, the UAV doing in Fail-Safe a user-programmed RTH, that's not cool for me. I like the idea of ​​doing something like ArduPilot Smart RTL better.

stronnag, 2022-01-07:
Someone could always revive Alberto's `agh_smart_rth` branch. When last tested in 2018-06 it was quite promising, but required further work. In particular the tendency to faithfully reproduce take off and fly the model into the same terrain.

JulioCesarMatias, 2022-01-07:
in my opinion, a WP-RTH mode that runs based on user configures, would not be cool. I think something like ArduPilot's Smart RTL should be worked on in iNav.

Mateyhv, 2022-01-07:
It can be useful as a second type of RTH, another mode so you can select with a flick of a switch which RTH will be executed but where the pilot is always in control of the situation can turn it off and ammend any unwanted/unexpected situation that may arose.

The more I think the more I see failure situations in a failsafe sceneraio where the system picks the closest point in a random direction to initiate the mission to home. So I wouldn't pick this for a failsafe. It can work very well in certain scenario but be a complete mess in another. But as I said it can be cool for a return on pilots input.

StuweFPV, 2022-01-08:
> 

what was that all about? can you link some reference? thanks

StuweFPV, 2022-01-08:
> Someone could always revive Alberto's `agh_smart_rth` branch. When last tested in 2018-06 it was quite promising, but required further work. In particular the tendency to faithfully reproduce take off and fly the model into the same terrain.

what was that all about? can you link some reference to read up on it? thanks

StuweFPV, 2022-01-08:
> It can be useful as a second type of RTH, another mode so you can select with a flick of a switch which RTH will be executed but where the pilot is always in control of the situation can turn it off [...]

or the other way around like MrD stated. WP-RTH primary and standard RTH secondary. of course having a switch (if you have spares) would even be nicer. but i don't want to overcomplicate the request. just 2 features added - keep it simple so it will have a chance to materialize.

of course this solution would fail in certain situations but so do the current ones. there is no perfect solution for every situation with RTH. but in my opinion this comes close if you stick to your flight plan. at least you can influence the path to avoid obstacles. how do you do that now? and it would not be random which WP to start with. either closest or last one past - again depends on your mission and the way you've plotted the WP-RTH.

Jetrell, 2022-01-08:
> what was that all about? can you link some reference to read up on it? thanks

https://github.com/iNavFlight/inav/tree/agh_smart_rth

StuweFPV, 2022-01-08:
thanks for these links. interesting project and i would not mind seeing this fly but as a alternate solution. what does not match my idea of the WP-RTH:

- it's again relying on the FC to plot on the fly. trusting my software to work correctly
- requires again to kinda plot the way up with positions and altitude with witch the software can work with
- another scenario where i don't have full control and just hope for the best
- additional workload for the FC during flight (not sure how much it would be but just another task on top which is useless 98% of the time)
- does not take into account obstacles like villages, roads, power lines unless avoided on the way up

i've seen inav and my FC's do strange things that remain a mystery to me until today. computers do that sometimes. I would like to see a robust solution that removes the reliance on a "intelligent" machine in my plane. more lines of code mean more possible errors. I'd like to see something very simple but robust that gives me the most control i can have of the way the plane performs RTH. I know that inav can fly WP-missions quite well, all i want is to reuse that feature and list it as a option of a failsafe action. with the additional option to select "nearest" or "last passed" WP to start with. that's all. keep it as simple as we can so it will work the way expected when we need it the most.

JulioCesarMatias, 2022-01-09:
https://ardupilot.org/copter/docs/common-rally-points.html

StuweFPV, 2022-01-09:
yes it kinda goes into this direction. except i don't want the plane just to go to the nearest WP, loiter and land. I would like to fly the entire mission until the last WP then goto home. But the idea is very similar and the reason for introducing this feature almost identical.

closed 2022-12-01 by StuweFPV