iNavFlight/inav #7721: OSD on goggle is not showing Average Cell Voltage

brownviper, 2022-01-04:
## Current Behavior

## Steps to Reproduce

1. battery cell voltage in setup tab shows the battery voltage
2. selected battery cell voltage on OSD tab
3. on the goggle osd I see mA/(can not make out the symbol) ESC 3
4. it not showing the average cell voltage 

## Expected behavior
I m expecting to see 3.90V

## Additional context

https://pastebin.com/BDxaqzzS
---

# version
# INAV/MATEKF722SE 3.0.2 Sep 15 2021 / 10:24:04 (66011184)
# GCC-9.3.1 20200408 (release)

- FC Board name and vendor:
- Lumenier LUX F7 Ultimate Flight Controller (Dual Gyros)
- INAV version string:

avsaase, 2022-01-04:
> 3\. on the goggle osd I see mA/(can not make out the symbol) ESC 3

This suggests you don't have the correct font installed.

StuweFPV, 2022-01-04:
I agree. it works fine for me and i always use cell voltage. upload the font before any diff

brownviper, 2022-01-05:
I v used all the available fonts, with and without Lipo power.
however, when I used iNav 3.0.2 the dont get the font issue

StuweFPV, 2022-01-06:
maybe you wana double-check the setting: osd_stats_min_voltage_unit
default would be BATTERY but you want CELL

avsaase, 2022-01-06:
> maybe you wana double-check the setting: osd_stats_min_voltage_unit default would be BATTERY but you want CELL

That's only for the disarm summary.

StuweFPV, 2022-01-06:
yeah sorry my mistake. it was late... just an idea. but maybe still worth a try to check if it works in the stats in order to narrow down the issue?

MrD-RC, 2022-01-08:
@brownviper  Are you saying that you're having this issue on 3.0.2, as indicated by the diff? If so, I can honestly say that I've never seen this issue. What version of configurator are you using?

brownviper, 2022-01-20:
sorry for the late reply guys,
the issue about the font was mainly showing in version 4. even with uploading different fonts!
when I dropped to 3.02 all seem to work

MrD-RC, 2022-01-20:
You were using configurator 4.0 with firmware 4.0 I assume? The configurator and firmware major version numbers need to match.

closed 2022-05-01 by DzikuVx