iNavFlight/inav #7722: H743 Wing + SDCard + 4.0.0 = "No configuration received within 10 seconds, communication failed"

LinkinL0gs, 2022-01-06:
## Current Behavior
When an SDcard is in the SDcard slot of my H743 Wing it is impossible to connect to over USB, yea I KNOW ABOUT THE VCP and the FS this and FS that. I'd spent the better part of the past 2 weeks considering giving up RC altogether as a hobby after banging my head against a wall for longer than i'd like to admit.  Then this evening i decided to remove the sdcard and VOILA, Magic, I can connect again and configure the Quite Awesome 4.0.0.  I then made sure to follow the instructions for setting up the blackbox at inavfixedwinggroup and as soon as i plugged back in "No configuration received within 10 seconds, communication failed" Until such time that I removed the SDcard and plugged it back in.  Boom! connected no problemo. 

## Steps to Reproduce

1. Insert SDcard into H743
2. Attempt to connect via USB through windows configurator
3. You won't be able to, I promise you...
4. Unless it's something to do with my particular SDcard: Team 500x 32GB

[image]

## Expected behavior

I'd been expecting to be able to connect to the FC using USB like i had been in 3.0.1, and yea i know the sdcard hadn't been available at that point but i was still able to connect with the card in the slot, i liked it being there cause i didn't wanna lose it before it was supported. 

## Suggested solution(s)

No clue 🤷 But I appreciate all yinz' work

## Additional context

[](https://pastebin.com/nrqrC0PH)
[image]

I feel bad for blaming this on STM and lashing out at them on twitter  = ( 
https://twitter.com/linkinlogs/status/1473151161398091781

---

# INAV/MATEKH743 4.0.0 Dec 15 2021 / 15:18:48 (c20a9852)
# GCC-10.2.1 20201103 (release)

- FC Board name and vendor: Matek H743 Wing
- INAV version string: 4.0.0

LinkinL0gs, 2022-01-07:
Having read the High CPU load thread in issue thread #7707 I'm going to give their work around solution a try here and see if that resolves what i'm dealing with!  *crosses fingers* will circle back afterwards.

bkleiner, 2022-01-16:
Closing in favor of https://github.com/iNavFlight/inav/issues/7759 please comment there if your issue was resolved by using a different sdcard or by waiting longer for the sdcard to init

closed 2022-01-16 by bkleiner