iNavFlight/inav #7726: Inav 4.0.0 - Unexpected motor stop and drone crash - defective battery or something else  ?

alccrl, 2022-01-07:
### Discussed in https://github.com/iNavFlight/inav/discussions/7725

<div type='discussions-op-text'>

<sup>Originally posted by **alccrl** January  7, 2022</sup>
Hi,
I was in doubt about opening an issue on inav 4.0.0 because I cannot reproduce the problem in short times, but while I was piloting in ACRO mode in a very soft way and no wind, quad not so far from me, so in complete control of what it was going on, suddenly the quad pointed to ground and crashed. !!!
I reacted putting quad into POS_HOLD, but it was too late, quad was already in the mud (luckily - not crashed on the rocks).
I attach here to this post
- taranis opentx log
- blackbox log
- short footage from goggles fpv recording
- screenshot  of blackbox explorer with an arrow indicating the crash point

Equipment used: 
AutoBuilt Fpv Long Range QAV-250 frame with Matek F722-STD FC and Taranis X9-Lite RC link (868-900 Mhz)), 6 inch propellers. Inav 4.0.0 official release. 5.8 Ghz video link (vtx).
[image]
[THE_BEAST-2022-01-07.csv](https://github.com/iNavFlight/inav/files/7831123/THE_BEAST-2022-01-07.csv)
[LOG00101.TXT](https://github.com/iNavFlight/inav/files/7831124/LOG00101.TXT)

https://user-images.githubusercontent.com/78967144/148597096-dbcd677f-5264-45c4-af25-32eed8e07063.mp4

YOU CAN SEE something like a HOLE in the bottom state bar, at about time 00:30. it seems that sw stopped control of motors ...

Any help about this ?
Could it be a defective battery casuing the problem ?

Any help is appreciated !!!
Thanks in advance ...

Alfredo</div>

alccrl, 2022-01-07:
I add also INAV 4.0.0. configuration file
[INAV_cli_20220107_214650.txt](https://github.com/iNavFlight/inav/files/7831274/INAV_cli_20220107_214650.txt)

Alfredo

Mateyhv, 2022-01-07:
I believe you had some mechanical failure or something heavy got loose. Twice you had a situation where the rear motors increased power and forward almost stopped, the second time started about 4,36 seconds before the fast pitch movement and keep increasing until the front motors stopped. This situation is clearly audible in the DVR.

The fast powerless inverted pitch proves there was a heavy imballance transferring a big weight to the rear, to the point of almost stopping the front motors to maintain attitude which it miraculously did. At that anomalous point probably some protection kicked in and stopped all four motors. Most probably thats what happened.

alccrl, 2022-01-08:
Thanks for Your tips.
Quad is quite balanced, and it did fly again after the crash. I wanted to be sure that crash was not damaging components, so I cleaned the mud from the quad and it flew again well. So I don't think that a sudden change of weight took place.
Sometime it happens that a bird attacks quads, but neither I did, or my son did recognize birds in the neighborhoods at the momnt of the crash.
I attach a picture of the quad, You can see that it is quite balanced.
[image]

I am suspicious that the (graphene) battery is defective. Before flight SkyRc balanced charger signaled it already charged after  two days of storing it, it sounds strange, because generaly batteries loose charge, not gain charge after a few days, but I cannot demonstrate this idea based on the available flight logs.

So, I am waiting for myself or somebody else to have some more ideas about this behaviour.

Alfredo

Mateyhv, 2022-01-08:
The battery does not affect the dynamics. If you look at the motor graph and the DVR you will see that the quad is absolutely steady in its attitude when the front motors almost stopped. The only way for that to happen physicaly is for CG to have moved back. If it was a battery failure all motors would have suffered equally. If you ask me I would rather put normal blades which are more effective and better ballanced than folding ones.

alccrl, 2022-01-08:
I am not expert in Weight&Balance, but a sudden change in CG forces could happen, I guess,  if CG is located very high in the vertical axis and the drone changes its pitch attitude from back to front or viceversa, example if I try to relevel the quad that is pointing the ground trying to move it again toward the sky, pulling the pitch stick toward my body. The battery in the picture is mounted quite tall, vs other batteries I use that are more flat and longer. So. I again guess, motor stops because of a too high energy saving in motor rpm. in other words, the quad is flying keeping rpm at minimum because I want to fly long range, not acrobatic. Some sort of iper-tuning toward low rpm.
Sorry if I speak an heuristic language, I am an electronic engineer, I am not prepared to discuss aerodynamics stuff.

Alfredo

DzikuVx, 2022-01-09:
You have pulled Throttle below `min_check`. Because before that THR never crossed 1300us mark, Airmode was not active

closed 2022-01-09 by DzikuVx

alccrl, 2022-01-09:
Hi 
[image]
Pawel,
how to avoid this ? airmode is enabled (see screenshot)...
What is it missing in configuration ?

Alfredo

DzikuVx, 2022-01-09:
Nothing is missing. You have to raise the throttle at least once for Airmode to kick in. Or lower https://github.com/iNavFlight/inav/blob/master/docs/Programming%20Framework.md to `1100` or so

alccrl, 2022-01-09:
Thanks, I'll have a look at docs.

Alfredo

alccrl, 2022-01-09:
Just a doubt:

I thought that airmode is always active and assures that motors do not stop.
After arming, for sure, throttle does change, otherwise drone does not takes off...
So, is this behaviour (stop) related to changing flight mode <<during>> flight ?
So, do flags and thresholds be reset if You change mode during flight ?
( I generally start in angle, go to acro or pos_hold afterward)
If I start directly with acro mode at taking off, can stop yet happen during flight ?
This is very dangerous and unsafe, it it can happen.

Alfredo

alccrl, 2022-01-12:
Issue #7726 has been closed because an explanation exists for this behaviour.
If behaviour is as expected this does not mean that behaviour is good, some problem about requirements might arise that suggest a different behaviour or solution.
So, I consider issue #7726 still open ...

MrD-RC, 2022-01-12:
No, this is a closed matter. Airmode only activates once the throttle has been raised above the threshold. That is a documented feature. 1300 should have been fine. But https://github.com/iNavFlight/inav/pull/7735 lowers the default threshold. 

Airmode keeps stabilisation active at low throttle, once activated.

alccrl, 2022-01-13:
I did not change any default, at least I did not wanted to change them.
I have yet to go into details of logs, but when I took off I think I passed the threshold for airmode to be activated, at least once.
If flags and airmode are reset when mode is changed during flight, this is a bug, because throttle could be already over the limit and piloting cannot assure that You go low and up again.
If I well understood, if You go lower and airmode has been reset, motors stop...
I do not want to be boring, so I will not go on, my hope was to solve this issue without going into the code.
If motors may stop during flight because of the sw, I think I have to change sw ....

Thank You All for Your patience, but if quad falls at 1km distance from home point, I would not be so happy ...

Alfredo

digitalentity, 2022-01-13:
This is a configuration issue. Typically you should set your airmode threshold to a value **below** your hover throttle. This way you simply won't be able to take off without passing the theshold for airmode.

Also note that this threshold doesn't exist for automated modes in the default settings - they keep motors running regardless of your throttle stick position.