iNavFlight/inav #7729: Add "FW level trim" to Adjustments

StuweFPV, 2022-01-08:
## Current Behavior
As with the new servo autotrim you should no longer adjust the board alignment like in previous versions in order to keep the plane level when flying in autonomous modes. Instead you should use the "Fixed Wing level trim" in the mechanics or the CLI "fw_level_pitch_trim". I've always leveled my FW's in flight by changing the pitch and roll level via adjustments. The new method is not yet available in the adjustments menu but the "old" ones of board alignment still are.

## Desired Behavior
Being able to properly level my plane in flight again

## Suggested Solution
Ad "fw_level_pitch_trim" to the Adjustments menu to be used in flight via switch. Perhaps remove the "old" 2 board alignment options from the Adjustments menu in order to prevent false configurations.

## Who does this impact? Who is this for?
Anyone who would like to level his plane in flight via switch

## Additional context
I found the changable unit of 1.0 often too high for a perfect tuning. Rarely I had to change my board by 5 degrees but rather by i.e. 1.2. I would prefer the changing unit to be lower at least 0.5 or even better 0.2 since the code in CLI does accept 0.1 changes.

StuweFPV, 2022-01-08:
cheers mate!

MrD-RC, 2022-01-09:
Duplicate of #7113

StuweFPV, 2022-01-09:
indeed it is. nicely spotted and good that you've flagged that issue. i still try to figure out what pitch board adjustment is here for these days if we're not supposed to use it anymore. however we need the new FW level trim up in the air. every airplane has it's own AoA for level flight and since we don't have proper flight manuals we have to "fly out" the AoA.
thank you MrD!

PR #8028 (2022-05-14): Added FW level trim to in flight adjustments
Fixes #7729
Fixes #7113

Configurator PR: https://github.com/iNavFlight/inav-configurator/pull/1517

[Demo](https://youtu.be/K9SInOTxUTk)
closed 2022-05-14 by DzikuVx