iNavFlight/inav #7732: Akk ultra long range vtx

calehunt, 2022-01-08:
I have a speedy bee f7 v2 running inav4.0 using smart audio to control my vtx , akk ultra long range vtx, it has 5 power levels, using smart audio I can access power levels 1-4 but when I switch to 5 it goes back to the first power level ? I’m just wanting the first power until first arm then full power for the rest of the flight , the gui will allow me to select that option but doesn’t work correctly.

Jetrell, 2022-01-08:
iNav doesn't presently have support for this VTX. 
Here a a reference of the issue that was brought up earlier. #6530

Jetrell, 2022-01-08:
iNav doesn't presently have support for this VTX.
Here is a reference of the issue that was brought up earlier. #6530

calehunt, 2022-01-09:
That issue was on inav 2.6 , inav 4.0 has 5 power levels in the gui , I’m sure I’m not the only person that is or will be using this vtx and being able to have it at a low power until armed would be great !

Jetrell, 2022-01-09:
The IRC Tramp protocol has 5 power levels. TBS Smart Audio has only had 4 levels.  AKK seem to have an exception to the SA protocol.

Jetrell, 2022-01-09:
The IRC Tramp protocol has 5 power levels. TBS Smart Audio only has 4 levels.  AKK seem to have an exception to the SA protocol.

SebKister, 2022-01-24:
You can recompile your firmware making this change in vtx_smartaudio.c:

const char * saPowerNames[VTX_SMARTAUDIO_MAX_POWER_COUNT + 1] = {
    "----", "250 ", "500 ", "1000", "2000", "3000"
};

// Fill table with standard values for SA 1.0 and 2.0
saPowerTable_t saPowerTable[VTX_SMARTAUDIO_MAX_POWER_COUNT] = {
    {  250,   24 },
    {  500,  27 },
    { 1000,  30 },
    { 2000,  33 },
    { 3000,  35 } 
};

And change in vtx_smartaudio.h:

#define VTX_SMARTAUDIO_DEFAULT_POWER_COUNT 5

calehunt, 2022-01-24:
I’m nowhere near being able to compile my own firmware :/ was hoping for a simple workaround

SebKister, 2022-01-24:
Tell me which FC you need it for and I'll compile a release_4.0.0 with the mods

calehunt, 2022-01-24:
That would be awesome ! The flight controller I’m using is a matek f405 wing

SebKister, 2022-01-24:
Here you go :-) 
[https://1drv.ms/u/s!AiAVr-uNNYbxp_sn5Hu1DQ68PgLD1w?e=mG05fd](url)
Let me know if it works as it should.

calehunt, 2022-01-24:
For some reason this link doesn't work?

SebKister, 2022-01-24:
It should directly start the download of the file, not open a page. Here's another link with a ZIP file containing the firmware:
https://1drv.ms/u/s!AiAVr-uNNYbxp_sqVYMNNvM-J0vKCQ?e=pivf7y
[inav_4.0.0_MATEKF405SE](https://1drv.ms/u/s!AiAVr-uNNYbxp_sqVYMNNvM-J0vKCQ?e=pivf7y)

calehunt, 2022-01-24:
It’s working now thanks ! I’ll flash this evening:)

closed 2022-02-23 by DzikuVx