iNavFlight/inav #7752: New Target for JHEMCU GHF405 PRO AIO

scottrea123, 2022-01-14:
Hope that support for this device can be added. Currently supported by Betaflight:

https://github.com/betaflight/unified-targets/blob/master/configs/default/JHEF-JHEF405PRO.config

scottrea123, 2022-01-15:
[JHEF405PRO.zip](https://github.com/iNavFlight/inav/files/7873863/JHEF405PRO.zip)
Ok, figured this one out - src is attached. Tested ok in the configurator.

Odin263, 2022-01-15:
@scottrea123 
Where did you get that from? why is the black box deactivated? Because the wrong pins are specified for SPI2?  Look here:
[en.DM00037051.pdf](https://github.com/iNavFlight/inav/files/7874851/en.DM00037051.pdf)

Do you have the target and is it working for you now?

scottrea123, 2022-01-15:
I modified an existing target with info from the beta flight code. I have the board and can confirm it does work for the mimimim that I have setup.
Only currently have the receiver connected but the gyro baro and radio all work.

scottrea123, 2022-01-15:
based on the betaflight config, i think BB flash and OSD is on SPI3
I changed the "CS PINS" accordingly. Does this look better?

#define USE_SPI_DEVICE_3
#define SPI3_SCK_PIN            PC10
#define SPI3_MISO_PIN           PC11
#define SPI3_MOSI_PIN           PC12

#define USE_MAX7456
#define MAX7456_SPI_BUS         BUS_SPI3
#define MAX7456_CS_PIN          PB14

#define ENABLE_BLACKBOX_LOGGING_ON_SPIFLASH_BY_DEFAULT
#define USE_FLASHFS
#define USE_FLASH_M25P16
#define M25P16_CS_PIN           PB3
#define M25P16_SPI_BUS          BUS_SPI3

Odin263, 2022-01-17:
@scottrea123
If everything works, then it's good. I just wanted to help and create a target but couldn't test it. Now that you have one, that's settled.

scottrea123, 2022-01-20:
Hi - having some trouble with flight, wondering if the target.c is set up correctly? currently have:

const timerHardware_t timerHardware[] = {   
    DEF_TIM(TIM3, CH3, PB0,  TIM_USE_MC_MOTOR, 0, 0),
    DEF_TIM(TIM3, CH4, PB1,  TIM_USE_MC_MOTOR, 0, 0),
    DEF_TIM(TIM5, CH4, PA3,  TIM_USE_MC_MOTOR, 0, 0),
    DEF_TIM(TIM5, CH3, PA2,  TIM_USE_MC_MOTOR, 0, 0)   	
};

Not sure about the last two parameters, any advice?

armsp, 2022-05-15:
@scottrea123 and @Odin263 I also want to flash INAV on it to fly...any updates on this? Any luck with a firmware that seems to work?

Noramas, 2022-07-15:
Any progress on JHEMCU GHF405 PRO target to be added to iNav?

VendettaPrime, 2022-12-08:
Any progress?

amdgigabyte, 2023-01-20:
@scottrea123 I am working on this Target too recently, Even though JHEMCU offer a inav 5.0.0 hex [here](https://share.weiyun.com/h4Z2va2f), But I found this aio does not have scl/sda pads and inav does not have resource remapping ,So I want to change the scl/sda definitions to make it work。

ltwin8, 2023-01-21:
@amdgigabyte do you have the binary for download? the file you linked does not download

ltwin8, 2023-01-22:
take a look at #8726

amdgigabyte, 2023-01-26:
> @amdgigabyte do you have the binary for download? the file you linked does not download

I am still working on it，I wanna contact the factory tech support for help， caus my binary still did not work properly.

ltwin8, 2023-01-26:
@amdgigabyte please take a look at #8726

closed 2024-03-22 by DzikuVx