iNavFlight/inav #7756: Random motor output to 0

charlesmarseille, 2022-01-15:
## Current Behavior
7" quad arm dips randomly, uncorrelated to throttle, angle or flight mode.
I witnessed another instance where both front motors got 0% (not triggered at the same time, but overlapping intervals).
[image]

## Expected behavior
smooth flight, relatively equal motor output through normal range of motion.
If motor desyncs, I expected the motor output to go to 100% and not 0 (since the arm dips).

- FC Board name and vendor: matek f722se
- INAV version string: 
INAV/MATEKF722SE 3.0.2 Sep 15 2021 / 10:24:04 (66011184)
GCC-9.3.1 20200408 (release)

Odin263, 2022-02-11:
The sum of the buoyancy of all motors should remain constant even when corrected. As a result, an engine sometimes has to drop to 0, otherwise the quad would shoot up uncontrollably.

closed 2022-03-13 by DzikuVx

charlesmarseille, 2022-03-13:
The arm litterally dips towards the ground and the quad must recover. 
This is not a normal behaviour, as can be seen in the logs.