iNavFlight/inav #7763: Need PINIO and USER1 mode for Matek F405SE

donne1226, 2022-01-18:
## Current Behavior
[image]

## Desired Behavior

## Suggested Solution

## Who does this impact? Who is this for?

## Additional context

Mateyhv, 2022-01-22:
@donne1226 are you able to compile the target yourself with PINIO?

donne1226, 2022-01-24:
@Mateyhv I do watched a [tutorial video](https://youtu.be/2zElnakFQRU), I'm worry about the future upgrade if I compile it myself

sevrugin, 2022-03-18:
Hi! Have you tested PINIO2? 
I tried to use your patch with 4.1 version
On Configurator I assigned USER2 to the switch. It switched when I change the switch position
#resource 
A08: PINIO1 OUT
B15: PINIO2 OUT

# get pinio_
pinio_box1 = 0
pinio_box2 = 41
pinio_box3 = 255
pinio_box4 = 255

After firmware was flashed I tried to measure the voltage between G and S8 - no differences between ON and OFF switch position

sevrugin, 2022-03-18:
Updated
I just set 
pinio_box1 = 47
pinio_box2 = 48

donne1226, 2022-08-30:
This issue has been solved on version 5.1.  many thanks for developer's hard work

closed 2022-08-30 by donne1226

Refick79, 2022-12-17:
> огромное
Привет. У меня так и не получается настроить pinio в inav.   Полетник f405 se пытаюсь повесить на tx4 управляющий пин камеры и не получается(( прошит inav 5.1. Помогите разобраться.