iNavFlight/inav #7765: iNAV Target Request for iFlight Succex-D Mini F7 2-6S Flight Controller

MeapVR, 2022-01-18:
## Current Behavior
As far as I can tell there is no iNav target yet for this flight controller. I did try to flash a custom .hex file from another feature request but that ended up being for the Succex-E which is not my flight controller and the gyro did not work with that firmware.

## Desired Behavior
A firmware target for the Succex-D Mini F7. I have a GPS on board.

## Suggested Solution
Added iNav support for the Succex-D Mini F7 flight controller.

## Who does this impact? Who is this for?
Anyone that has the iFlight Nazgul Evoque F5D Digital version (DJI Caddx Vista) which is a very popular drone on the market right now.

## Additional context
If anyone can help make this happen I would highly appreciate it. I can't wait to use iNav on my Nazgul for my long range flights using my Li-Ion pack. I have provided a dump of my Nazgul Evoque F5D Digital as attachment to this post.

[BTFL_cli_Nazgul_F5D_20220102_152939.txt](https://github.com/iNavFlight/inav/files/7890199/BTFL_cli_Nazgul_F5D_20220102_152939.txt)

Odin263, 2022-01-18:
Hello @MeapVR , I think you got something wrong there. As far as I know, the D in Succex-D stands for double gyro. The target for this is available in iNav:
[image]

what you need is the target iFlight BLITZ F7, which is also already available in iNav:
[image]

MeapVR, 2022-01-18:
Thank you. I will check this out.

closed 2022-01-21 by MeapVR