iNavFlight/inav #7772: MatekF411WSE only using the first two PWM outputs for servos (INAV 4.0, MatekF411 FD_SFTSRL1 target)

Painless360, 2022-01-21:
Reproduced on two Matek F411WSE boards flashed with the MatekF411 FD_SFTSRL1 target, INAV 4.0. 

Setup as both 'PLANE with DIFFERENTIAL' and regular 'PLANE' with no change to the symptom. 

First two servo connections (marked S3 and S4 on the FC) work as expected but S5 and S6 do not give out any PWM signals. Re-ordering the servo table so that different outputs appear on different servos make no difference. Only S3 and S4 (as marked on the FC) will move the servo. 5V power is on the other two non functional outputs so the servo is powered.

I've re-flashed and wiped the config as well as reproduced on a brand new Matek F411WSE.

To reproduce: flash a Matek F411WSE with the MatekF411 FD_SFTSRL1 target and configure as normal for plane as above. Plug in servos and only those connected to S3 and S4 (as marked on the FC) will respond.

Dump and images available on request.

MrD-RC, 2022-01-21:
Hi Lee. That sounds like the wrong target. They should use the MATEKF411SE_FD_SFTSRL1 target.

Painless360, 2022-01-21:
That is the target used. Dump here - https://www.dropbox.com/s/wxt8s2wjfc9j9d3/P360_dump.txt?dl=0

MrD-RC, 2022-01-21:
Thanks for clarifying. There is also a MATEKF411_FD_SFTSRL1 which it sounded like you may have been using from the original post.

snaewe, 2022-01-25:
Can't confirm. Just did a clean flash of  MATEKF411_FD_SFTSRL1 4.0 to a F411-WSE, setup 'full'' Airplane, all servos move.
Loaded the above dump, same result: all good.

MrD-RC, 2022-01-28:
Do the servos work with the regular target?

closed 2022-05-11 by DzikuVx

Senders76, 2022-06-11:
Did you ever get this sorted out? 
  I've got the same firmware flashed to the same board on 4.1 
Mixer is setup for a wing with differential thrust. I have no pwm output on S3 or S4.  I have not yet tried s5 or s6.
   I've got 4.93v on both s3 and s4.  I've desoldered both servos and used a servo center tool to confirm servos are good.