iNavFlight/inav #7773: RTH Critical bug: stick rolling to extreme values

krl91, 2022-01-22:
## Current Behavior
https://youtu.be/y2EjF46OeT4
[LOG00002.TXT](https://github.com/iNavFlight/inav/files/7918217/LOG00002.TXT)

## Steps to Reproduce
5 inches quad with iNav 3.0.1

1. Calibrate compass
2. Check the compass is correctly pointing to the north
3. Start flying in Angle Mode 
4. Continue flying in Acro Mode
5. Activate RTH
6. Pray or get the control back asap
7. if praying: crazy toilet spirals at landing step
8. manual landing or deArm

## Expected behavior
the right stick goes crazing : it is rolling itself to extreme values in clockwise, that is making the drone "dancing" sometime it stabilized sometimes not

## Suggested solution(s)
If this behavior (stick rolling to extreme values) is wanted that is ugly and dirty, this bug has to be catch for security.

## Additional context
Dump:
https://pastebin.com/j8fJyUhp

---

# Version

- FC Board name and vendor: (original) Omnibus F4 v5
- INAV version string: INAV/OMNIBUSF4V3 3.0.1 Jul 19 2021 / 00:35:46 (cb6726e2)
GCC-9.3.1 20200408 (release)

tiriad, 2022-01-22:
I have also noticed in version 4.0 and specifically in FS two times, that when the RTH is activated, it makes a very strong roll to the right and the motors throttle up to 100% within 1 second and then stabilises horizontally. I find it uncomfortable because it seems to lose control.

Mateyhv, 2022-01-22:
The teenager music makes it look good

krl91, 2022-01-22:
I already had this issue in a previous flight:
https://youtu.be/v2rRzVQkarQ after 45 seconds

tiriad, 2022-01-22:
i have caught the bug in fpv, i have gone from 3.0.0 where it worked fine, to 4.0 so the bug could be between 
https://youtu.be/DWkcEwhLWuE

krl91, 2022-01-23:
I just add an experience to check the compass isolation, the drone is fixed (manually), throttle move randomly between 0% and 100%) in two positions:
[LOG00001.TXT](https://github.com/iNavFlight/inav/files/7921074/LOG00001.TXT)
[LOG00002.TXT](https://github.com/iNavFlight/inav/files/7921075/LOG00002.TXT)

krl91, 2022-01-31:
magnetometer issue, sorry

closed 2022-01-31 by krl91