iNavFlight/inav #7775: bug/feature? Servos freeze while in the menus

wx4cb, 2022-01-24:
INAV v4.0 release, servos freeze in MAX positions when entering the OSD menu. this can cause un-needed stress on servos attached to the rudder and pitch channels by driving them (potentially) over the limits of the servo or control surface in the case of binding.

fix: set the servos to be centered to their mid range when in the OSD menu.

Affects: All fixed wings

OptimusTi, 2022-01-26:
Found out why. I'll give it a crack and post a PR if I can make it work.

OptimusTi, 2022-02-04:
> Found out why. I'll give it a crack and post a PR if I can make it work. 

Can't make it work. Can't find a way to reset the servos after entering the cms. 😟

wx4cb, 2022-02-04:
@OptimusTi  couldnt you just freeze them at 1500 ?

OptimusTi, 2022-02-04:
I thought of doing that before line 769 but I couldn't figure it out. Maybe doing a little auto servo test to make it look cool

wx4cb, 2022-02-04:
> I thought of doing that before line 769 but I couldn't figure it out. Maybe doing a little auto servo test to make it look cool

somewhere around line 1313

wx4cb, 2022-02-04:
@OptimusTi  not sure if you can do this but maybe (at lines 1311-1313 etc
```
    if (!cmsInMenu) {
        // Detect menu invocation
        if (IS_MID(THROTTLE) && IS_LO(YAW) && IS_HI(PITCH) && !ARMING_FLAG(ARMED)) {
             THROTTLE = YAW = PITCH = 1500;
            cmsMenuOpen();
            rcDelayMs = BUTTON_PAUSE;    // Tends to overshoot if BUTTON_TIME
        }
```

MrD-RC, 2022-02-04:
I’m sure the servos used to be disabled in the CMS. But were re-enabled when servo adjustments were added to the CMS?

OptimusTi, 2022-02-04:
It's the other way around. Until 2 years ago, servos were active in the cms.

OptimusTi, 2022-02-04:
> @OptimusTi  not sure if you can do this but maybe (at lines 1310-1315) etc
> 
> ```
> 
>     if (!cmsInMenu) {
> 
>         // Detect menu invocation
> 
>         if (IS_MID(THROTTLE) && IS_LO(YAW) && IS_HI(PITCH) && !ARMING_FLAG(ARMED)) {
> 
>              THROTTLE = 0;
> 
>              YAW = PITCH = 1500;
> 
>             cmsMenuOpen();
> 
>             rcDelayMs = BUTTON_PAUSE;    // Tends to overshoot if BUTTON_TIME
> 
>         }
> 
> ```

Nope. That's not it.

JulioCesarMatias, 2022-11-08:
https://github.com/iNavFlight/inav/pull/5319

wx4cb, 2022-11-08:
> #5319

guess this is finally going to be fixed lol

JulioCesarMatias, 2022-11-08:
> > #5319
> 
> guess this is finally going to be fixed lol

do you want to test?

wx4cb, 2022-11-08:
> > > #5319
> > guess this is finally going to be fixed lol
> do you want to test?
to be honest, i havent got the time right now as im in the middle of major projects at work, and moving house

closed 2024-03-20 by b14ckyy