iNavFlight/inav #7778: [REQ] Staged Autolaunch for heavy crafts + auto loiter - maybe nav supported launch in the future

b14ckyy, 2022-01-24:
## Current Behavior
Currently the Autolaunch feature is pretty dumb. It sets a target angle and just goes full madness on throttle. Yes this works well and is reliable but especially for bigger crafts with heavy weight but still lots of power a more sensible way to launch would be nice. As a reference I'd like to take a Passenger plane. 

## Desired Behavior
Like a real passenger plane that has a TOGA power setting, similar things could be applied for INAV Autolaunch by implementing a launch boost throttle and split the launch into 2 phases. Actually 5 phases in total. 

1. Pre-Throw: As it is right now Autolaunch gets initiated, pilot raises throttle, Idle is applied if set. 
2. Launch Boost: Pilot throws the plane, the motors go to full power (or a necessary power level to quickly reach a stable speed)
3. Climb phase: Throttle is reduced to protect batteries, motors and ESC and conserve power and only apply enough throttle to savely climb at the set climb angle
4. End phase: as it is right now, leveling out and gently transition into cruise throttle
5. Loiter phase: after the launch is completed, the plane should loiter if any other flight mode is set than RTH, WAYPOINT or POSHOLD until the pilot actively inputs stick commands. 

## Suggested Solution
This would need 2 or 3 new values for autolaunch depending on implementation. The Boost phase with high power applied could be simply a timer. Like 2000ms boost and then it switches into the climb phase with lower throttle. The time can be adjusted. The boost throttle value could keep its name Launch throttle and the new lower value would be Climb Throttle. 
If a GPS is on board it could be used to determine a specific ground speed to switch to climb throttle. As we NEVER launch with tail wind, an airspeed sensor won't be necessary. If no GPS is installed or fails, the timer is used automatically.

## Who does this impact? Who is this for?
Every pilot flying big cruisers and especially twin engine planes with high takeoff weight that also have a massive amount of thrust. These planes need to get to a stable flight speed quickly with lots of power but then can easily climb at much lower power levels. For example my X-UAV Clouds setup pulls 115A peak on launch out of lipos and 100A from the LiIon pack. This causes a lot of stress to the batteries and it climbs like a rocket at more than 90kph with 15-20m/s climb rate. This causes a lot of wasted power. after reaching a stable cruise speed of 60-70kph it could use just 40-50A to continue the climb at a reasonable climb rate. 

## Additional context
In the future this could be advanced even more with a navigation-based climb phase and use the 3D Cruise Mode after the boost. Controlling the throttle based on P2T and active climb rate control. It would also allow the plane to maintain heading during climb so it does not drift off course in crosswind situations, leading to a more stable and safer launch pattern. 
**Of course this needs tuning and can't be used on a maiden or a untuned navigation system so there need to be an option to switch between Nav based climb and angle based climb as dewcribed above.**

wx4cb, 2022-01-24:
tbh, can't you pretty much do that with the current settings for launch angle, max height, and judicious use of the timings ?

yes i realise that it'll use launch throttle, but once it's at it's altitude it goes back to cruise throttle does it not ?

MrD-RC, 2022-01-24:
@wx4cb that’s the whole point though Charles. It doesn’t need to use launch throttle to get up to altitude, it only needs a climb throttle. For example the launch throttle of 2000. But once the safe ground speed is reached and the aircraft is just climbing out, the throttle could be reduced to 1700. The when the launch completes, level off and use the cruise throttle.

b14ckyy, 2022-01-24:
yes it does after launch but let's say my launch target altitude is 100m, the plane will climb to these 100m at full throttle and unnecessary high airspeed in a very short time or with a lower launch angle it will be very far away very quickly and reduce the time to be ready to take over. It would be nice to get the extra push on launch to get to cruise speed as fast as possible but then climb at that constant cruise speed instead of way faster than necessary. any extra speed above cruise will burn extra power from the battery that could be used during flight later and stress the batteries.

or take planes like a mini drak or big drak for example that you usually launch nearly vertical and needs a high thrust to weight ratio to get ot a stable speed. But as soon as the climb is stable there is no reason to climb that rapidly anymore and shoot out of LOS within seconds. But instead it could climb much slower and shallower at low throttle instead of pulling 100+ Amps.

wx4cb, 2022-01-24:
> @wx4cb that’s the whole point though Charles. It doesn’t need to use launch throttle to get up to altitude, it only needs a climb throttle. For example the launch throttle of 2000. But once the safe [...]

then just reduce the launch "window" or add max alt etc?

JulioCesarMatias, 2022-01-24:
about two weeks ago, I was working on an equation to make better use of the energy during the ascent (in this case what you mentioned in behavior 3), but I still couldn't reach a final equation

wx4cb, 2022-01-24:
in all seriousness though, your going to have the issue of True airspeed vs Ground speed... cus don't forget, you can have a ground speed higher than your true airspeed so then if it chops the throttle, the plane will stall

JulioCesarMatias, 2022-01-24:
> about two weeks ago, I was working on an equation to make better use of the energy during the ascent (in this case what you mentioned in behavior 3), but I still couldn't reach a final equation

an equation for preventing stall during ascent is also used, but as I said in the comment above, I couldn't come up with an ideal math for iNav

b14ckyy, 2022-01-24:
> about two weeks ago, I was working on an equation to make better use of the energy during the ascent (in this case what you mentioned in behavior 3), but I still couldn't reach a final equation

This is actually pretty easy. as long as you keep your airspeed at an optimal rate, you have the optimal climb power for energy saving. So if your plane is massively overpowered and has it's sweet spot at 60kph cruise speed than a vertical climb with a speed of 60kph or 17m/s would be optimal for efficient climb. Although it sounds counter intuitive

b14ckyy, 2022-01-24:
> in all seriousness though, your going to have the issue of True airspeed vs Ground speed... cus don't forget, you can have a ground speed higher than your true airspeed so then if it chops the throttle, the plane will stall

this is ruled out as we NEVER EVER launch into a tail wind. So our airspeed will always equal or be higher than the ground speed. Therefore it is safe to use this as a trigger for the "climb phase"

MrD-RC, 2022-01-24:
> then just reduce the launch "window" or add max alt etc?
> 
> btw lets not clutter up the thread :D

But by reducing the launch window, you aren’t getting up to the launch altitude. You’re having to compromise to save power and potentially component health. Not to mention the full power section could only be 2 or 3 seconds. Not even long enough to get the goggles on. 

I think the settings should be CLI only though, as an advanced feature.

wx4cb, 2022-01-24:
> this is ruled out as we NEVER EVER launch into a tail wind. So our airspeed will always equal or be higher than the ground speed. Therefore it is safe to use this as a trigger for the "climb phase"

not necessarily.... you could be flying backwards :D and not to mention that could be launching down wind :D but i do get your point.

Mateyhv, 2022-01-25:
As @JulioCesarMatias pointed out we have discussed something very similar to what is proposed at point 3 however the solutions we got never passed the theoretical checks for every possible situation. The main idea came from my Dolphin which after a 4 seconds launch reaches almost 130kmh which is basically a lost energy because usualy after the launch the overspeed is reduced to cruising level. The idea was instead of converting the power into kinetic energy, to convert it into potential energy gaining altitude so as in real life in case of some failure to circle back to the landing strip. Even in the event of reverting from FPV to LOS its easier to spot the plane higher but closer than shooting away at high speed and thus easier to bring down. The main defeat of this idea is that probably 95% of inav foam fun planes can just do with 25º or 30º climb and a bit of tunning to have the speed on check. Anyway here are both option:
1) The dumb one: pilot sets a speed and a second climb angle. Once that speed is reached the new climb angle is adopted.
2) The smart one: pilot inputs the stall speed only. At (stall_speed x1.6) pitch is gradualy increased proportional to the speed gain to the point where speed is not increasing anymore, which idealy will happen around (stall_speed x 2). On that same principle if speed decreases pitch will drop thus preventing any dangerous situation and as a way of self control.

This method is good for heavy planes that need a shallow launch angle but once have picked some momentum and safe speed can use all available power to maximize the climb. Of course this phase is to be done on fixed TOGA throttle. Reducing throttle to maintain a lower speed is not an option since high power is safety in this phase.

wx4cb, 2022-01-25:
@Mateyhv what you're saying is you want a "best rate" and a "best efficency" climb :D pretty much what we do in the airlines :D

Mateyhv, 2022-01-25:
@wx4cb exactly ;)

JulioCesarMatias, 2022-01-25:
> @Mateyhv what you're saying is you want a "best rate" and a "best efficency" climb :D pretty much what we do in the airlines :D

yeeeeeeeeeeeeeeeeeeeeeeeeeeees

JulioCesarMatias, 2022-01-25:
And also the lack of interest on the part of the official Dev's of iNav, makes me not make an effort to arrive at an ideal math for this resource. Some Dev's (you know who they are) are fixed-wing users, and they are only interested in their ideas, others' ideas are kind of excluded and not viewed or discussed. I mean, it's discouraging.

b14ckyy, 2022-01-26:
@Mateyhv I see your point. Having a climb angle change is indeed a nice option to gain speed first and then find the sweet spot. 

But.... I still want to be able to reduce the power after the initial speed is reached. Let me bring my Clouds as an examble. At full throttle this plane with 3.4kg AUW can climb vertical at 25m/s. even with a shallow 20° climb angle it reaches the target altitude of 100m in under 5s and a peak airspeed of more than 110kp/h and even with a steeper angle it will barely drop below that. It just makes no sense to keep the throttle applied at full power until launch alt is reached. 

Thus would rule out your 1. option already as it is basically the same as it is now with just a delayed rotation point. 
Your Option 2. sounds a bit better but still would not adopt the throttle and in case of very powerful planes lead to insane climb rates and if you limit the max angle it ends up in the same problem. That's why the reduced power when reaching cruise speed should be preferred as this is the main reason. And the last option, using altitude control for a controlled climb rate with autothrottle (after tuning) like you do in 3D Cruise mode, would eliminate all these issues completely. 

Of course the new autolaunch variants I suggest here will need some more adjustments than currently. But the new paramters could be just 0 (disabled) by default so it will launch the same way as it did before at first. And can be tuned later.

rts18, 2022-01-27:
> @b14ckyy 
 _And the last option, using altitude control for a controlled climb rate with autothrottle (after tuning) like you do in 3D Cruise mode, would eliminate all these issues completely._

That idea would certainly be the most efficient when tuned correctly.