iNavFlight/inav #7780: Soaring mode over Course hold issue

Mateyhv, 2022-01-25:
Today I tested the soaring mode coupled to course hold (TRK on the OSD). It works as expected, the plane maintained level flight and the nose dropped at reduced throttle and rised when throttle was rised.

The issue is that when roll is applied. Instead of gently rolling what happens is that full level stabilization is activated and the nose goes immediately to level attitude. Apparent no roll motion is executed, only the pitch is leveled. Once the roll input is off the nose falls as expected on Soaring mode. I repeat this several times at zero throttle to left and right and always got the same result a quick nose up to level flight.

To show the roll input applied at the end of the video ACRO is selected and the same roll command is executed this time as expected.

https://user-images.githubusercontent.com/53102558/151066680-530373fe-844f-4874-9a32-fba7e026ea2a.mp4

breadoven, 2022-01-26:
Soaring mode is disabled if the pitch/roll stick isn't centered. This is so position adjustment still works as normal giving some ability to change direction/altitude. This adjustment has always seemed a bit sluggish to respond which might explain the apparent lack of affect when rolling to change course. How does it respond with Course Hold if Soaring is off ?

Mateyhv, 2022-01-27:
Definitely roll is slugish on Course hold alone as if there was a center stick dead zone and then responds as in angle. 
I mentioned it because feels strange to soar at zero throttle and for slight trajectory corrections on roll, to level the pitch suddenly which also can induce stall.
@breadoven do you want me to close this issue?

closed 2022-09-04 by breadoven