iNavFlight/inav #7781: Matek F405 miniTE DSHOT 300/600 not working

mulfo44, 2022-01-26:
Motors don't work with DSHOT 300 or 600 on my new build quad, however, DSHOT and RPM filtering works perfectly fine with Betaflight.

Setup: Matek F405 miniTE FC with Mamba F30 4in1 BLHELI S ESC (DSHOT 600 supported) flashed with Bluejay version 16.  (no versions of Bluejay work by the way)

The only way I can get the motors to work and fly with INAV is to flash the ESC with BLHELI S without DSHOT support and use multishot or oneshot125.  Then the motors will work and I can fly, but I would really like to be able to use DSHOT and filtering.

INAV 4.0.0
inav_4.0.0_MATEKF405TE

version
# INAV/MATEKF405TE 4.0.0 Jan 25 2022 / 12:51:12 (GITDIR-N)
# GCC-10.2.1 20201103 (release)

StuweFPV, 2022-01-26:
The same happened to me with the Matek F722-Wing. Dshot not working but Oneshot was fine

rts18, 2022-01-27:
@mulfo44 You may want to bring this issue up with Matek here - #7656 . 
The F405TE Target has not presently been merged into iNav.

FNNCFPV, 2022-02-24:
My flight controller doesnt even communicate with my esc, using same fc. only getting do do do... not getting last 2 esc beeps.

mulfo44, 2022-02-25:
Matek says it's a firmware issue.  I guess wait for an update and keep looking on their website for new firmware, mine doesn't even work with oneshot right.

FNNCFPV, 2022-02-28:
> Matek says it's a firmware issue. I guess wait for an update and keep looking on their website for new firmware, mine doesn't even work with oneshot right.

I found a solution. I flashed betaflight to the fc to establish communication with esc using dshot600, I then flashed my esc using BLHeliS to an older firmware that had multishot next to the name. I then flashed iNav back into my fc from then my esc was communicating using multishot. You'll just need to calibrate your motors. Let me know if you need some help with this I can hop on discord to help you through the process.

closed 2022-05-12 by DzikuVx