iNavFlight/inav #7789: OSD Elements Incorrect 4.1RC1

DataStorageCS, 2022-01-30:
## Current Behavior
HDZero connected to H743 Wing V2. OSD elements in goggles don't seem to be aligning what is selected in the configurator. For example, average cell voltage is selected and centered in the OSD tab in the configurator but what shows on the output is a combination of km and sats.

## Steps to Reproduce
1. Flash RC 4.1 to H743 Wing V2
2. Connect HDZero Whoop to FC on any spare UART and powered off of 9v/Ground pads.
3. Enable HDZero VTX in the corresponding port in the ports tab in the configurator
4. Power on FC with a battery and let HDZero rx connect

## Expected behavior
Expected behavior would be OSD elements appearing in the goggles according to what they should be in the configurator.

## Additional context

[image]
[image]

dump

# version
# INAV/MATEKH743 4.1.0 Jan 25 2022 / 11:16:00 (ebf5b0e3)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1503 -100
servo 4 1000 2000 1503 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
safehome 0 0 0 0
safehome 1 0 0 0
safehome 2 0 0 0
safehome 3 0 0 0
safehome 4 0 0 0
safehome 5 0 0 0
safehome 6 0 0 0
safehome 7 0 0 0
[... 1521 lines cut ...]
set nav_fw_pitch2thr = 10
set nav_fw_launch_thr = 1800
set nav_fw_launch_idle_thr = 1500
set limit_cont_current = 0
set limit_burst_current = 0
set limit_burst_current_time = 0
set limit_burst_current_falldown_time = 0
set limit_cont_power = 0
set limit_burst_power = 0
set limit_burst_power_time = 0
set limit_burst_power_falldown_time = 0

# end the command batch
batch end

#

---

# version
# INAV/MATEKH743 4.1.0 Jan 25 2022 / 11:16:00 (ebf5b0e3)
# GCC-10.2.1 20201103 (release)

- FC Board name and vendor: Matek H743 Wing V2
- INAV version string: 4.1 RC1

MrD-RC, 2022-01-30:
You haven’t set the video format to HD.

joshfisk3, 2022-01-30:
Whoops posted that with my work account. Is that this setting? set osd_video_system

MrD-RC, 2022-01-30:
Just use the radio buttons on the top left of the OSD page 😉

joshfisk3, 2022-01-30:
gotcha just saw it, thank you for the help

MrD-RC, 2022-01-30:
No problem 👍🏻

closed 2022-01-31 by DzikuVx