iNavFlight/inav #7794: Matek Optical Flow & Lidar Sensor 3901-L0X for SPEEDYBEEF7V2

andreapede, 2022-02-02:
Actually there is no support for these sensors in the target of SPEEDYBEEF7V2
I mounted on my drone and I would like to use it.
At the moment I added the following line to the target.h

```
#define USE_RANGEFINDER
#define USE_RANGEFINDER_MSP
#define USE_OPFLOW
#define USE_OPFLOW_MS
```
I'm using the UART 3 of the FC.

But it doens't seems working.
Am I missing something?

andreapede, 2022-02-02:
Found what was missing. This should be typed in CLI.

`serial 2 1 115200 115200 0 115200
set rangefinder_hardware = MSP
set opflow_hardware = MSP`

DzikuVx, 2022-02-22:
rangefinder and opflow are enabled on all targets > 256kB of flash, there is no need to make any code changes

```// Allow default rangefinders
#define USE_RANGEFINDER
#define USE_RANGEFINDER_MSP
#define USE_RANGEFINDER_BENEWAKE
#define USE_RANGEFINDER_VL53L0X
#define USE_RANGEFINDER_VL53L1X
#define USE_RANGEFINDER_US42
#define USE_RANGEFINDER_TOF10120_I2C

// Allow default optic flow boards
#define USE_OPFLOW
#define USE_OPFLOW_CXOF
#define USE_OPFLOW_MSP```

closed 2022-02-22 by DzikuVx

pDR8iLQ4go, 2022-02-22:
Is there a description available on how to parameterize "align_opflow"? Both, FC and lidar are in the "normal" position. Is this the right parameter:
```
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 0
set align_opflow = CW0FLIP
```

DzikuVx, 2022-02-23:
This is described in the opflow module manual