iNavFlight/inav #7799: Incorrect Diff All after 3.02

Jetrell, 2022-02-05:
@MrD-RC 
Not sure if you remember us talking about this on Discord. I thought it was tied to the `Profiles`.

I removed the extra Profiles and the programming logic that controlled them, and the issue still remained.  

The main concern is the way it separates the below setting from the rest of the Launch values, after a `Diff All`. 
`nav_fw_launch_thr = 1850`
`nav_fw_launch_idle_thr = 1200`
Even when the _loaded_ `Diff All `had those settings in the correct location in the text file.

It places them in the `battery_profile`.
[image]
And not here
[image]

If that incorrect `Diff All` is now reloaded in to the CLI. It will not identify the launch throttle value as part of the launch settings, and makes them default.
[image]

This could be a trap for anyone who didn't notice, and found their launch throttle is under powered after a throw.

I did some testing, and found that the issue appeared days after the 3.02 release.

MrD-RC, 2022-02-05:
Ah, I see what you mean now. I think the real issue is this is not in the release notes. Having the launch idle and throttle in the battery profile makes sense. The same plane would likely need less of both on 4S compared to 3S. 

There being no warning about this is the problem. I’ll add something to the release notes.

Jetrell, 2022-02-05:
>  Having the launch idle and throttle in the battery profile makes sense. The same plane would likely need less of both on 4S compared to 3S.
> 
> There being no warning about this is the problem. I’ll add something to the release notes.

If the Battery_profile is used to change Control profiles. Each battery_profile will have those individual settings to suit that cell count.. Plus many other settings as well.

I find it odd.. that it only adds those specific ones to the battery_profile. When there is no other Battery or Control profiles used in the `Diff All`.  What do you think about this?

Thanks for replying :-)

MrD-RC, 2022-02-05:
You can't have the settings in one place, unless multiple profiles are used. They either have to be in profiles or not. That's not something that can't change. Plus, it would be more confusing if they're in one place on one model and another place on a different model with the same firmware version, don't you think? 

I think they're much more useful in profiles, as it makes they system more flexible. To me the only issue is that they have been moved, and it's not been documented in the release notes. The Battery.md document is also not up to date with these changes. It might also be a good idea to highlight the fields that are different when switching battery profiles and control profiles in Configurator. That will give folks an idea of what to check when profiles are changed.

Battery profiles and control profiles are independent by default. They can be linked by setting `controlrate_profile = 2` for example, in the battery profile.

Jetrell, 2022-02-05:
I get your point. Especially when multiple `battery_profiles` or multiple `control_profiles` are used. As you stated, the logical place to have these same setting, with different values, according to battery voltage... Should be under the `battery_profile `.

e.g. from a Multicopter. Note `ControlRate_profile` being used.

```
# profile
profile 1
set mc_p_pitch = 30
set mc_i_pitch = 43
set mc_d_pitch = 19
set mc_cd_pitch = 70
set mc_p_roll = 28
set mc_i_roll = 45
set mc_d_roll = 19
set mc_cd_roll = 65
set mc_p_yaw = 38
set mc_i_yaw = 50
set mc_d_yaw = 1
set mc_cd_yaw = 38
set mc_p_level = 26
set mc_d_level = 65
set max_angle_inclination_rll = 400
set dterm_lpf_hz = 150
set dterm_lpf_type = PT1
set dterm_lpf2_hz = 350
set yaw_lpf_hz = 10
set heading_hold_rate_limit = 60
set nav_mc_pos_xy_p = 67
set nav_mc_vel_xy_p = 43
set nav_mc_vel_xy_d = 96
set nav_mc_heading_p = 55
set mc_iterm_relax_cutoff = 20
set d_boost_max_at_acceleration =  5000.000
set d_boost_gyro_delta_lpf_hz = 95
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set tpa_rate = 15
set tpa_breakpoint = 1750
set rc_expo = 40
set roll_rate = 62
set pitch_rate = 58
set yaw_rate = 60
set manual_rc_expo = 40
set fpv_mix_degrees = 20

# profile
profile 2

set mc_p_pitch = 23
set mc_i_pitch = 39
set mc_d_pitch = 14
set mc_cd_pitch = 65
set mc_p_roll = 22
set mc_i_roll = 41
set mc_d_roll = 13
set mc_p_yaw = 33
set mc_i_yaw = 48
set mc_d_yaw = 1
set mc_cd_yaw = 35
set mc_p_level = 26
set mc_d_level = 60
set max_angle_inclination_rll = 400
set dterm_lpf_hz = 170
set dterm_lpf_type = PT1
set dterm_lpf2_hz = 370
set yaw_lpf_hz = 15
set heading_hold_rate_limit = 60
set nav_mc_pos_xy_p = 67
set nav_mc_vel_xy_p = 43
set nav_mc_vel_xy_d = 96
set nav_mc_heading_p = 55
set mc_iterm_relax_cutoff = 20
set d_boost_gyro_delta_lpf_hz = 90
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set tpa_rate = 20
set tpa_breakpoint = 1650
set rc_expo = 40
set rc_yaw_expo = 22
set roll_rate = 62
set pitch_rate = 58
set yaw_rate = 60
set manual_rc_expo = 40
set fpv_mix_degrees = 20

# profile
profile 3

# battery_profile
battery_profile 1

set bat_cells = 3
set vbat_min_cell_voltage = 340
set vbat_warning_cell_voltage = 360
set battery_capacity = 850
set battery_capacity_warning = 204
set battery_capacity_critical = 170
set controlrate_profile = 1
set throttle_idle =  3.000
set failsafe_throttle = 1240
set nav_mc_hover_thr = 1265

# battery_profile
battery_profile 2

set bat_cells = 4
set vbat_min_cell_voltage = 340
set vbat_warning_cell_voltage = 360
set battery_capacity = 700
set battery_capacity_warning = 168
set battery_capacity_critical = 140
set controlrate_profile = 2
set throttle_idle =  3.000
set failsafe_throttle = 1170
set nav_mc_hover_thr = 1190

# battery_profile
battery_profile 3

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save
```

# 

>
>Battery profiles and control_profiles are independent by default

The part I can't work out. Is why these (specific) throttle values are included under the `battery_profile`, when there is only one profile being used.
**i.e**. _battery_profile 2 & 3_ and _Control_profile 2 & 3_ are empty. Not selected for use.. And `ControlRate_profile` has never been selected for use on my fixed wings at all. They are only a fixed cell count.
Under this condition, it would seem logical to have them under the `#master`?

The only tried _Control_profile_ changes through the program logic conditions on a fixed wing. But have now removed them. And its just #Profile 1.
Does this make sense?

>
>The Battery.md document is also not up to date with these changes. It might also be a good idea to highlight the fields that are different when switching battery profiles and control profiles in Configurator. That will give folks an idea of what to check when profiles are changed.

That would be a good idea.. Thankyou

MrD-RC, 2022-02-05:
> The part I can't work out. Is why these (specific) throttle values are included under the `battery_profile`, when there is only one profile being used. **i.e**. _battery_profile 2 & 3_ and [...]

Because of the way things are coded. The CLI commands are hard coded in the settings.yaml file. Plus there are profiles files that define how the profiles are made. Dynamically changing the code is not possible, only through compiling the firmware. 

But also, it doesn't really make sense to me to have moving variables. It would just add a layer of confusion. For example, I have 2 planes, both on 4.0.0. One only uses 4S batteries, the other is using 4S and 6S (with battery profile 2 set up for 6S). It makes no sense to me to have the settings stored in # master on the first plane, but on the second plane they're in # battery profile. The settings are always related to a battery, even if you only use one battery on that model. To me, it makes more sense to have them with the battery settings.

> > The Battery.md document is also not up to date with these changes. It might also be a good idea to highlight the fields that are different when switching battery profiles and control profiles in Configurator. That will give folks an idea of what to check when profiles are changed.
> 
> That would be a good idea.. Thankyou
> 
> And if nothing can be done.. Would it be possible to some how allow Launch mode to integrate its throttle setting... regardless of were they are found in the Diff All? Like re-configuring the Diff All loading order. So #Master comes last or something...

I don't know about re-ordering the diff, or even if it would have any effect. I still think the key problem here is that it's missing from the release notes. At least then pilots could make note of the moved settings, or even modify the diff so that they are placed in the battery profile(s). Once it's done, the problem doesn't come up again.

MrD-RC, 2022-02-05:
The release notes have been updated to reflect this. The Battery.md file will be updated as soon as it passes the checks.

Jetrell, 2022-02-05:
>
> At least then pilots could make note of the moved settings, or even modify the diff so that they are placed in the battery profile(s). **Once it's done, the problem doesn't come up again**.

Please don't take this as me being argumentative... I know the results i'm seeing and its frustrating. The FC is an Matek F411wing.

It always loads these as default, 
`set nav_fw_launch_thr = 1700`
`set nav_fw_launch_idle_thr = 1300 OR 1000`
even though MY setting values are there to see under the `battery_profile`. It just ignores them in a CLI load, and loads the default anyway.
```
# battery_profile
battery_profile 1

set vbat_cell_detect_voltage = 435
set vbat_min_cell_voltage = 360
set vbat_warning_cell_voltage = 370
set battery_capacity = 650
set battery_capacity_warning = 156
set battery_capacity_critical = 130
set throttle_idle =  0.000
set fw_min_throttle_down_pitch = 20
set nav_fw_cruise_thr = 1480
set nav_fw_min_thr = 1300
set nav_fw_pitch2thr = 9
set nav_fw_launch_thr = 1850
set nav_fw_launch_idle_thr = 1200
```

[image]

But if my `Diff All` has those values removed from the `battery_profile` and placed in the `#Master`
Like this -
```# battery_profile
battery_profile 1

set vbat_cell_detect_voltage = 435
set vbat_min_cell_voltage = 360
set vbat_warning_cell_voltage = 370
set battery_capacity = 650
set battery_capacity_warning = 156
set battery_capacity_critical = 130
```
Then it loads the My values into above launch throttle settings( viewed under the advanced tuning tab).
This is the only way I can make it load those settings for launch throttle, is by going back to the old way.

Have you had the chance to load try what I've mention here? So you may understand my meaning.
Its either written words are a bad way to communicate... Or i'm totally missing something... But I have used the `Controlrate_profile` on my quads from back in June and its never been an issue. And I can't see any issue with the Diff All text file.  And I'm not a newbie to iNav either.

I'll close this issue and when I test some other FC later on. Then If need be ill come back here and chat again.
Thanks Darren

MrD-RC, 2022-02-05:
Not at all mate. It looks like Configurator and and the firmware aren't in sync in the way they're getting the settings there. Configurator 4.x should be getting the data from the battery profile. If it's not, that's an issue.

I've just tested on the latest master, and it seems to be working fine. I used the battery profile diff from your last comment.
[image]

Here are the master and profiles sections from the diff
```
# master
set gyro_main_lpf_hz = 25
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 30
set acc_hardware = MPU6000
set acczero_x = 31
set acczero_y = -5
set acczero_z = -193
set accgain_x = 4070
set accgain_y = 4101
set accgain_z = 4052
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = FAKE
set motor_pwm_protocol = STANDARD
set model_preview_type = 8
set small_angle = 180
set imu_acc_ignore_rate = 9
set imu_acc_ignore_slope = 5
set applied_defaults = 3
set airmode_type = STICK_CENTER_ONCE
set nav_wp_radius = 5000
set nav_rth_allow_landing = FS_ONLY
set nav_rth_altitude = 5000
set nav_fw_bank_angle = 45
set nav_fw_control_smoothness = 2
set osd_overspeed_alarm = 3350
set i2c_speed = 800KHZ

# profile
profile 1

set fw_p_pitch = 15
set fw_i_pitch = 5
set fw_d_pitch = 5
set fw_ff_pitch = 70
set fw_p_roll = 15
set fw_i_roll = 3
set fw_d_roll = 7
set fw_p_yaw = 20
set fw_i_yaw = 0
set fw_ff_yaw = 100
set max_angle_inclination_rll = 550
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.200
set nav_fw_pos_z_p = 15
set nav_fw_pos_z_d = 5
set nav_fw_pos_xy_p = 60
set d_boost_min =  1.000
set d_boost_max =  1.000
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 18
set pitch_rate = 9
set yaw_rate = 3

# battery_profile
battery_profile 1

set vbat_cell_detect_voltage = 435
set vbat_min_cell_voltage = 360
set vbat_warning_cell_voltage = 370
set battery_capacity = 650
set battery_capacity_warning = 156
set battery_capacity_critical = 130
set throttle_idle =  0.000
set fw_min_throttle_down_pitch = 20
set nav_fw_cruise_thr = 1480
set nav_fw_min_thr = 1300
set nav_fw_pitch2thr = 9
set nav_fw_launch_thr = 1850
set nav_fw_launch_idle_thr = 1200
```

The really strange thing is the Advanced tuning page is getting everything from direct MSP calls. So it's effectively doing `get nav_fw_launch_thr' etc. Not getting the data from any specific place. It's been like that since 3.0. Performing that get in CLI works fine too.

Just out of interest, I tried just using `set nav_fw_launch_thr = 1750` to see what happened. Nothing was put in `# master` and the value in the battery profile was updated. Advanced Tuning also showed the correct value. I think there must be something strange going. Something doesn't add up here.

MrD-RC, 2022-02-05:
Just out of interest. I performed a full chip erase, then used this diff to re-setup the FC.

```
# diff

# version
# INAV/MATEKF405SE 5.0.0 Feb  3 2022 / 21:23:55 (d7760905)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1

# servo

# safehome

# logic
logic 0 1 -1 16 2 16 0 5 0
logic 1 1 -1 14 4 0 0 1500 0
logic 2 1 -1 38 0 11 4 1 0
logic 3 0 1 0 0 0 0 0 0
logic 4 0 3 0 0 0 0 0 0
logic 5 0 1 0 0 0 0 0 0
logic 6 0 1 0 0 0 0 0 0

# gvar

# pid

# feature
feature MOTOR_STOP

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 23 0 H

# master
set gyro_main_lpf_hz = 25
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 30
set acc_hardware = MPU6000
set acczero_x = 31
set acczero_y = -5
set acczero_z = -193
set accgain_x = 4070
set accgain_y = 4101
set accgain_z = 4052
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = FAKE
set motor_pwm_protocol = STANDARD
set model_preview_type = 8
set small_angle = 180
set imu_acc_ignore_rate = 9
set imu_acc_ignore_slope = 5
set applied_defaults = 3
set airmode_type = STICK_CENTER_ONCE
set nav_wp_radius = 5000
set nav_rth_allow_landing = FS_ONLY
set nav_rth_altitude = 5000
set nav_fw_bank_angle = 45
set nav_fw_control_smoothness = 2
set osd_overspeed_alarm = 3350
set i2c_speed = 800KHZ

set throttle_idle =  0.000
set fw_min_throttle_down_pitch = 20
set nav_fw_cruise_thr = 1480
set nav_fw_min_thr = 1300
set nav_fw_pitch2thr = 9
set nav_fw_launch_thr = 1850
set nav_fw_launch_idle_thr = 1200

# profile
profile 1

set fw_p_pitch = 15
set fw_i_pitch = 5
set fw_d_pitch = 5
set fw_ff_pitch = 70
set fw_p_roll = 15
set fw_i_roll = 3
set fw_d_roll = 7
set fw_p_yaw = 20
set fw_i_yaw = 0
set fw_ff_yaw = 100
set max_angle_inclination_rll = 550
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.200
set nav_fw_pos_z_p = 15
set nav_fw_pos_z_d = 5
set nav_fw_pos_xy_p = 60
set d_boost_min =  1.000
set d_boost_max =  1.000
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 18
set pitch_rate = 9
set yaw_rate = 3
```

The Advanced Tuning page was correct, as was the diff. The values had been automatically added to the battery profile.

Jetrell, 2022-02-06:
> Not at all mate. It looks like Configurator and and the firmware aren't in sync in the way they're getting the settings there. Configurator 4.x should be getting the data from the battery profile. If it's not, that's an issue.
> 

Thanks heaps Darren for spending the time to look at this. I know how long it takes!

I ran a few dozen tests today with various flight controllers and setting variations. And had no problems with the other FC's, but only this MatekF411_SFTSRL2 board and target. 
I have had this board in a Dart for years, and its never been any issue. And still works fine.

But all my tests today proved what you stated, **Configurator and the firmware aren't in sync**.

Using WinMerge proved that it was failing to load different setting half of the time.
* Sometimes it missed some of the `battery_profile`
* Other times it fail to load the whole of `profile 1`. 
* Then other times it missed loading 15 line from the bottom of the `#Master`.
* Or it didn't load anything from the `Profile` or `battery_profile`

[image]

I use a great deal of iNav's features. So all my Diff All's are very full.  
The CPU is only showing 10% usage during Diff loads.. 

Unless you have any idea's on this. I'll just have to look hard to see what is missing each time I run a Diff All load... Which is a pain. Because this is my test plane.

Thanks again mate :+1:

MrD-RC, 2022-02-06:
- Is that board ok with the regular MATEKF411 target? 
- Is the MATEKF411_SFTSRL_2 target from the Matek site or direct install from Configurator?

I don’t think its s sync issue. Using the get and set commands show that configurator don’t care if the variables are in `# master` or `# battery profiles`. The firmware stores or retrieves the data regardless. I’m thinking the problem is with that target. I have an F411-WING here. If you post up your full diff, I’m happy to try it out on that.

Jetrell, 2022-02-06:
Thank you for offering to do this :+1: 

I haven't tried it with any other F411_wing target. Due to this model requiring both Softserials for ESC telemetry and SA. Of which it works fine, as has for many releases.

The target is an official iNav release. I mostly run '**load from file**' for all my flashes.  And have always done "**Keep current setting**"... but have used the '**airplane without a tail**' many times in these tests, to cover all bases...  But it made no difference.

Here are the two `Diff All` files. The first that always works, is the one that has the throttle profiles located in the `#master`.
While the second was copied from CLI `Diff All `of that file. It has the throttle profiles loaded by the configurator, in the `battery_profile`. It will leave stuff out 50% of the time, when loaded again after a 'CLI - **Defaults**'.

[F411_SFTSRL2- always works.txt](https://github.com/iNavFlight/inav/files/8009744/F411_SFTSRL2-.works.txt)
[F411_SFTSRL2 - diff all from version that always works.txt](https://github.com/iNavFlight/inav/files/8009743/F411_SFTSRL2.-.diff.all.from.version.that.always.works.txt)

MrD-RC, 2022-02-06:
It would be great if anyone could test that second diff on a different F411 flight controller. There's something strange going on here.

MrD-RC, 2022-02-06:
I ended up having the same results as @Jetrell with that second diff file. I thought I'd record the whole process, as videos tell a million words. So, [here's that video](https://youtu.be/wZShoXeryDM).

I found that the problem happened on the F411-WING with that 2nd diff file (F411_SFTSRL2 - diff all from version that always works.txt). It was not target dependant, but happened on the Matek F411-WING regardless. I had 3 results:
- The first saved nothing in the battery profiles
- The second saved 3 parameters
- The third saved 4 parameters

Copying in the `# battery profile` section on it's own worked fine. So nothing wrong with the section in the diff.

I also tried the diff on an F405-WSE, and the whole diff was accepted (other than the soft serial 2 port error).

I think it would be great if it could be tested on other F411 flight controllers. Maybe even F722 flight controllers. I have an F722-WPX I can dig out here. The F411 and F722 have the same flash, but the F722 has double the RAM. So maybe that could give some ideas on what' going on?

ABLomas, 2022-02-10:
Maybe related:
https://github.com/iNavFlight/inav/issues/7281
https://github.com/iNavFlight/inav/issues/7128
(IIRC there are few more EEPROM corruption issues in INAV)

MrD-RC, 2022-02-10:
I don’t think they’re related. I watched the MSP data at the bottom of the CLI window, and nothing changes while the diff is pasted in to the CLI. So I believe this is all stored in objects in the configurator until the save command is used. 

Which would also make it more strange that it only seems to happen with F411 FCs. F722 work fine, same as F405. 

Nice idea though.

Jetrell, 2022-02-11:
Thanks again Darren for such a thorough investigation. 
Did you run a test with 3.02? Configurator and firmware. I didn't have the issue with them. 
But with this firmware from the master, a day or so after 3.02, the issue appeared. It was from the 18/19th Sept.
[inav_4.0.0_MATEKF411_SFTSRL2.zip](https://github.com/iNavFlight/inav/files/8048291/inav_4.0.0_MATEKF411_SFTSRL2.zip)

MrD-RC, 2022-02-11:
Cool, so that’s before the profile switching was merged. I think we can safely tick that off the list.

Jetrell, 2022-02-13:
I forgot that I owned an F411-wse. So I gave it a quick few try. 
It also acted in the same way as the F411-wing. By loading the first file without errors. Then producing  `battery_profile `errors from its `Diff All`, when loaded again.

MrD-RC, 2022-02-13:
Interesting. I’d love to know the results from a non-Matek F411.

Jetrell, 2022-02-16:
A friend tested a GEPRC F411. And said he experienced the same issue, with my `Diff All`, but not every time. 

There has got to be something that was added to the master at this time frame, that is causing the issue.
https://github.com/iNavFlight/inav/commits/master?after=36d36c9aa6828b1afc44b00c1c0fcd0c5d722f68+419&branch=master
Unless the the problem was added to the Configurator around that same time?

If these FC's don't have this issue in 3.02. And it then turns up a couple of days later with the file I left (above) from the master. It seems conclusive.
With 3.02 loaded. If I load my second file that has the throttle profiles placed into the `battery_profile`, by the Configurator... Then check the `Diff All` in the CLI. ... It places the those throttle setting back into the `#Master `again.... Just the reverse of what it does in 4.0-4.1 releases.

Jetrell, 2022-06-23:
@MrD-RC I'll close this for a bit of house keeping... I don't think we'll find out the cause.
I've got used to just working around the problem, by loading the profiles last.

closed 2022-06-23 by Jetrell