iNavFlight/inav #7820: Graphical OSD artificial horizon does not match Airplane pitch attitude.

access3424, 2022-02-11:
## Current Behavior
See attached screenshot. Graphical OSD artificial horizon does not match Airplane pitch attitude. In level flight raw pitch attitude shows 3.4 degrees nose up (Which is correct) and graphical horizon showing 6 degrees nose down. There is also  fw_level_pitch_trim =  4.000 and  align_board_roll = 0     align_board_pitch = 25    align_board_yaw = 900

## Steps to Reproduce
The above settings show the airplane level on the bench. The fw_level_pitch_trim is for angle mode to work better
1.
[image]

2.
3.
4.

## Expected behavior
All attitude references to match. ie. the graphical pitch indicator to match reality.

## Additional context

[INAV_cli_AR_WING_20220211_115055.txt](https://github.com/iNavFlight/inav/files/8047593/INAV_cli_AR_WING_20220211_115055.txt)

---

INAV/MATEKF722WPX 4.0.0 Dec 15 2021 / 15:09:49 (c20a9852)
# GCC-10.2.1 20201103 (release)

MrD-RC, 2022-02-11:
This is the angle of your camera. iNav doesn’t have any control over that. The CLI command that StuWe posted will sort your problem.

access3424, 2022-02-11:
> This is the angle of your camera. iNav doesn’t have any control over that. The CLI command that StuWe posted will sort your problem.

I wasn’t referring to the difference between the camera horizon and the artificial horizon line. I was referring to the indicated angle (dot between 0 and -10) but i will try that cli command see what happens. 👍🏻

access3424, 2022-02-11:
I used osd_ahi_vertical_offset. it was by default -18 no idea why. After reseting to 0 i got the desired result

closed 2022-02-11 by access3424