iNavFlight/inav #7828: Customized output mapping in mixer tab possible?

chretze, 2022-02-13:
## Current Behavior

The first two outputs of the flight controller are always reserved for motors, no matter the preset.

## Desired Behavior

The option to map the first two outputs to servos instead of motors.

## Suggested Solution

Ideally, the output mapping should be customizable.

## Who does this impact? Who is this for?

Former quadcopter pilots that want to switch to airplanes, as well as airplane pilots on a limited budget. And of course, me.

## Additional context

I generally think this would be a great feature because there are many very cheap FCs out there (especially on the second hand market) that are limited to 4 output channels. Pilots on a budget could make use of these and stabilize cheap airplanes and fully enable rudder, elevator, and two aileron servos on them. Many of these cheap planes don't warrant the purchase of a more expensive flight controller with more outputs. People are also more willing to experiment with and learn about the software if they can use it on cheap hardware, and the hobby would be more accessible by m
[image]
aking more cheap FCs suitable to stabilize planes.

MrD-RC, 2022-02-13:
This is not possible. The timers on the MCU can be used for servos or motors, not both. The motor timers on motors and servos operate at different frequencies.

chretze, 2022-02-13:
> This is not possible. The timers on the MCU can be used for servos or motors, not both. The motor timers on motors and servos operate at different frequencies.
> 
> You could change the target to have the first 2 motors as servos. But then you wouldn’t have any motors at all. Fine for a glider I guess. But it would be better just to buy a cheap flight controller with 6 outputs. I’m not a fan of them, but most Omnibust F4 FCs have 6 PWM outputs.

Thank you for the explanation. I think, coincidentally, you recently explained the same thing in a youtube comment.
Although I don't know what timers and mechanics lead to some outputs being able to be mapped as servos and others not, I guess this is some technical issue that goes beyond my understanding.

Could you elaborate on what you mean by "change the target to have the first 2 motors as servos"?
Cheers

MrD-RC, 2022-02-13:
Yes, that was me on YouTube. 

Basically, the processor has inputs and outputs (I/O) pins. These I/Os are channels on timers, multiple I/Os sharing timers. 

```
DEF_TIM(TIM3,  CH3, PB0,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,               0, 0), // S1_OUT D1_ST7
DEF_TIM(TIM3,  CH4, PB1,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,               0, 0), // S2_OUT D1_ST2
DEF_TIM(TIM2,  CH4, PA3,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,               0, 1), // S3_OUT D1_ST6
DEF_TIM(TIM2,  CH3, PA2,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,               0, 0), // S4_OUT D1_ST1
DEF_TIM(TIM8,  CH3, PC8,  TIM_USE_ANY,                 0, 0), // S5_IN // pad labelled CH5 on OMNIBUSF4PRO
DEF_TIM(TIM8,  CH4, PC9,  TIM_USE_ANY,                 0, 0), // S6_IN // pad labelled CH6 on OMNIBUSF4PRO
```

That is a cut down version of the Omnibus F4 Pro definitions for the 6 outputs. The first column (`TIM8` for example) is the timer number, the `CH` is the channel of that timer, and the next number is the pin on the MCU. After that, you see the function assigned.  So `TIM3` is always set to motors, regardless of if it is a multicoptor or fixed wing. `TIM2` is motors on copters or servos on fixed wing. TIM8 can be any. But they will need to be the same thing. So you couldn’t have a motor on `TIM8 CH3` and a servo on `TIM8 CH4`.

By change the target to have the first 2 motors as servos. I mean coding a custom firmware target for that situation. But, this would mean there are no motors on that flight controller, so navigation modes would be unusable. For example RTH would have no control over the throttle to navigate safely. 

I’m sure you said you have an Omnibus FC. Most of those have more than 4 outputs. The only ones I can think of that don’t are the AIO quad specific boards. On some, the pwm 5 and pwm 6 are regular pads or through a connector block.

chretze, 2022-02-15:
Welp, that seems like a challenge to rewrite the firmware. I'll order a different flight controller, but I'll still learn how to do this as it seems like a practical skill. I downloaded GCC Arm Embedded Toolchain. I read quite a few guides on how to modify firmware for ARM chips, but they all mentioned that it would only work on linux, however they were also quite old and it might just be that the windows version of this GCC Toolchain came a little later.

I have checked all the pins on the board; There is a set of 2x5 tiny pins on the back of the board that serve as connectors to things like 4in1 ESCs and what not. 4 of those 10 pins are GND, 4 are mirroring the 4 main outputs from the top of the board, and 2 seem to generate no output at all. Hard to tell if they're actually the PC8 and PC9 outputs or something else entirely...

The FC is actually a DYS F4 PRO. The (very limited) manual just tells me to use the OMNIBUSF4 firmware for INAV. Again, I think this might be just outdated, because there's a DYSF4PRO firmware available as well. I just installed that and on first glance it seems to work identically, I'll check if the back pins work differently tomorrow.
I became sick today so I'm just slow with everything atm

MrD-RC, 2022-02-16:
If you check in the doc/development folder on here. There are documents on how to build firmware. The one to checkout is [this one](https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows%2010%20or%2011%20with%20Linux%20Subsystem.md). Works fine in Windows.

Hope you feel better soon.

chretze, 2022-02-16:
An absolute champion in the discord has shown me everything how to modify them, I am now ready to experiment with all kinds of things regarding flight controller firmware. Everything works as I had hoped!
Thank you and everyone in the discord for your help!

closed 2022-02-16 by chretze