iNavFlight/inav #7831: Pid tune via dji V1 googles

klaudius666, 2022-02-14:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1. Changing rates in googles
2.the rate change and quickly go back to initial value

3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor: mambaf722i2c
- INAV version string: 3.0

klaudius666, 2022-02-14:
Same for pids

Scavanger, 2022-02-14:
Use iNav 4. Inflight adjustments are supported via the craft name field.

closed 2022-02-16 by MrD-RC