iNavFlight/inav #7847: Omnibusf4 wont arm 3.0.2 or 4.0.2

Gando65, 2022-02-23:
## Current Behavior
When attempt is made to arm aircraft an arming disabled flag armsw appears with osd message 'disable arm switch first'..did not have issue with inav 2.6..tried 2 different frsky d16 receivers and also flash receivers. Do not have issue on matekf411wse boards.

## Steps to Reproduce

DzikuVx, 2022-02-23:
It appears the channel you use for arming is HIGH during boot. this channel has to be put LOW before it can arm

closed 2022-02-23 by DzikuVx

breadoven, 2022-03-06:
I've never had a problem with the Omnibusf4 works fine with all recent versions including V5.0. I assume you've checked "Status" in the CLI ?

Gando65, 2022-03-06:
Yeah checked status etc, maybe its a clone board with issues but flew fine on 2.6. Ive built about 20 planes with 2.4, 433mhz, and 5.8 and 1.3ghz vtxs etc, never saw this before. For whatever reason on boot the board think qrm switch is on, guessing its prob some speed issue on boot, who knows...maybe just fly it with safety disabled on this plane and test modes, it is what it is...some bit of code prob not playing nice...same on 3.02 and 4.02 but ill check newer and newer buildd as they come.

breadoven, 2022-03-06:
As far as I can glean from the code the "Arm switch" arm blocker is only active when the arm switch is in the armed position and arming is disabled for some other reason. If the arm switch is deselected the "Arm switch" arm blocker is reset so can't get stuck as enabled at boot. Something else is causing arming to be blocked which results in the "Arm switch" message when you try to arm. However, almost all the other things that can block arming should show a message in the OSD before you try and arm. The only things that don't are related to using the OSD/CMS menu.

Gando65, 2022-03-06:
Thanks for suggestions..ill poke around myself..appreciate it!