iNavFlight/inav #7862: Support for SK6812 RGB+W LEDs

jbienz, 2022-02-27:
## Current Behavior

Only WS2811 LEDs are supported. No RGB + W lighting exists.

## Desired Behavior

I'm very interested in seeing SK6812 LED support added to iNav. For those who aren't familiar, SK6812 is very close in protocol to WS2811 already supported, but SK6812 adds support for a dedicated white channel. 

Here's a 2 minute video on why SK6812 is so awesome:

https://youtu.be/QnvircC22hU?t=720

Some of the main advantages of SK6812 over WS2811 are:

- **Greater accuracy in color reproduction.** White is pure white and requires no mixing. Light colors can be produced by mixing primary colors with white rather than mixing all primary colors.
- **Reduced power consumption for colors.** Mixing a primary color with white (2 channels) requires less power than mixing all primary colors (3 channels).
- **Greatly reduced power consumption for white.** Only 1 channel is needed for pure white.
- **No rainbow effect when fast moving.** Wingtip strobes, for example, could be pure white with no rainbow on high rate rolls or fly-bys.

## Code We Can Leverage

SK6812 support was added to Betaflight recently in December. The pull request is:

https://github.com/betaflight/betaflight/pull/10762

I pulled that branch of Betaflight and compared the changed files with the same ones in iNav. Unfortunately there were far more differences than I expected and I was not comfortable with trying to perform the merge myself.

[image]

Looking at the code, here are the changes made to Betaflight:

1. **LedStrip.md -** Simple updates to documentation to mention SK6812 support.
1. **Settings.c -** This is the configuration page for LEDs. The author added a new option in the drop-down to select RGB + White as the LED type.
1. **light_ws2811strip.c -** Code was added to include white when calculating colors values if RGB + White is enabled.
1. **light_ws2811strip.h -** Main changes here were to go from using 24-bit values for color to using 32-bit values (8 bits * 3 colors + 8 bits * 1 white).

## What is Needed

I looked up the author of the Betaflight pull request (g3gg0) and unfortunately they're not a contributor to iNav. I have found an e-mail address for this individual and I plan to reach out, but they may not respond or may not be interested in iNav. I'm hoping a member of the iNav community could help migrate these Betaflight changes over to the iNav codebase.

@Linjieqiang and @Mluessi, I'm tagging the two of you because I noticed that you've contributed to both projects and are likely familiar with both codebases. @Martinbudden, I'm tagging you because I saw that in 2016 / 2017 you made some efforts to bring iNav up to date with changes from Betaflight.

If anyone is up for collaborating on this or even just taking it on, please let me know. And THANK YOU!

g3gg0, 2022-03-02:
Hi,

will have a look into it, but cannot promise.

Regards,
g3gg0

jbienz, 2022-03-13:
Hey @g3gg0, does it look like there might be any possibility of getting some assistance on this one? If not, I understand. Just trying to decide if I should wait or go for WS2811 to get the lights in my Night Timber working again.

g3gg0, 2022-07-14:
Did you have time to check the code, that I've sent you?

jbienz, 2022-07-16:
Unfortunately this project got put on hold for me, but I do have everything you sent in my notebook for as soon as I'm able to resume it. Thank you sincerely @g3gg0.