iNavFlight/inav #7863: Unstability behavior with Blackbox log files and H7 flight controller

ArnoTlse, 2022-02-27:
## Current Behavior
I'm encountering unstability issue with the Blackbox feature with Inav 4.0.0 and my Flight controller Lumenier H7 LUX flashed with matek H743. 

`(H Firmware revision:INAV 4.0.0 (c20a9852) MATEKH743)`

Yesterday, I flew few flights. I have copied/pasted the log from my SDcard to the PC. And when I have wanted to open them through INAV BlackBox tool, only the last one open. In fact, the first one gives me a message like this :

[image]

The second gives me the same message, but the third indicates nothing as error message, but the information are not well read as for example, the motors of the quad mock-up are displayed @ full throttle all along the records.

[image]

And the last one just open nicely.
[image]
 

Today, I have tried to investigate a little. So, I completly format my SD card (with overwrite option, not the fast one) with **SD CARD formatter** tool. Put it directly on the quad and go for few flights again.
This time, the first flight log opens well, but all the others are corrupted. The second one gives me the error message. The third one loads, but some parameters seem missing : 

[image]

I need to investigate more in order to find a reproductable path. Meanwhile I wanted to highlight this experience I have and if someone else has encountered similar behavior. 

## Steps to Reproduce
For now, I can't find a reproductable path of the bug. What I have noticed for now, is unstability with the SD CARD on the H7 flight controller. I will try to make some further tests.

## Expected behavior
Stable SD CARD logging like I have already experimented on my F405 STD flight controller before switching to H7.

## Suggested solution(s)
None unfortunately

## Additional context

[CLI Dump](https://pastebin.com/sXvpjUUG)
PWD = BJpsJRgM36

---

# version
INAV/MATEKH743 4.0.0 Dec 15 2021 / 15:18:48 (c20a9852)
GCC-10.2.1 20201103 (release)

stronnag, 2022-02-27:
yet you fail to supply the one thing that would be useful to assist you, i.e. the actual blackbox log ....

ArnoTlse, 2022-02-27:
Hi,

Sorry. 
Here is a log with the error message :
[LOG00006.TXT](https://github.com/iNavFlight/inav/files/8149346/LOG00006.TXT)

Here is a log without error message but missing parameters:
[LOG00003.TXT](https://github.com/iNavFlight/inav/files/8149347/LOG00003.TXT)

stronnag, 2022-02-27:
The logs are corrupt. Large sections of the headers of that enable the tools to decode them are missing.

* Use a better card
* Use a lower logging rate.

stronnag, 2022-02-27:
```
$ blackbox_decode --stdout --merge-gps > /dev/null /tmp/LOG00003.txt

Log 1 of 1, start 03:56.763, end 05:27.538, duration 01:30.775

Statistics
Looptime            947 avg          229.8 std dev (24.2%)
I frames    5632  113.5 bytes avg   639322 bytes total
P frames   84470   69.6 bytes avg  5880078 bytes total
H frames      45   10.0 bytes avg      450 bytes total
G frames     497   25.1 bytes avg    12470 bytes total
E frames       1    6.0 bytes avg        6 bytes total
S frames     407   41.0 bytes avg    16687 bytes total
Frames     90102   72.4 bytes avg  6519400 bytes total
Data rate  992Hz  72187 bytes/s     721900 baud

3 frames failed to decode, rendering 10 loop iterations unreadable. 90106 iterations are missing in total (45388ms, 50.00%)

WARNING: Missing expected metadata - check for log corruption
	Warning: No VBAT reference
	Error: No P Interval
	Error: No Acc.1G metadata

$ blackbox_decode --stdout --merge-gps > /dev/null /tmp/LOG00006.txt

Log 1 of 1, start 02:48.632, end 04:01.832, duration 01:13.199

Statistics
Looptime            947 avg          229.6 std dev (24.2%)
I frames    4542  111.6 bytes avg   506767 bytes total
P frames   65433   69.6 bytes avg  4553223 bytes total
H frames      37   10.0 bytes avg      370 bytes total
G frames     399   24.4 bytes avg     9722 bytes total
E frames       1    6.0 bytes avg        6 bytes total
Frames     69975   72.3 bytes avg  5059990 bytes total
Data rate  955Hz  72061 bytes/s     720700 baud

358 frames failed to decode, rendering 2690 loop iterations unreadable. 75346 iterations are missing in total (37952ms, 51.85%)

WARNING: Missing expected metadata - check for log corruption
	Warning: No VBAT reference
	Error: No Firmware type metadata
	Error: No Firmware revision metadata
	Error: No P Interval
	Error: No Acc.1G metadata

```

ArnoTlse, 2022-02-27:
Ok, thanks. 
I will test the second thing first as unfortunately, I haven't other SD card yet, and I know that this one was working well on F405. 

Is there any SD CARD recommandation in term of class ?

stronnag, 2022-02-27:
Seems to me it's a bit of a crap-shoot. Some cards / FC combinations just don't get on for reasons unknown.
Please do try at lower rates (at the moment we can't even read the rate from the logs, they are so damaged).

ArnoTlse, 2022-02-28:
Hi, 

New flights today. I decrease the rate from 50% to 25%. Same issue unfortunately.
Here is one log with the error message :
[LOG00009.TXT](https://github.com/iNavFlight/inav/files/8156170/LOG00009.TXT)

and here a log without error message but missing data :
[LOG00011.TXT](https://github.com/iNavFlight/inav/files/8156173/LOG00011.TXT)

Will test further lower rate next time.

closed 2023-10-27 by DzikuVx