iNavFlight/inav #7864: Wrong mixer output

dataema92, 2022-02-28:
Hello,
I have a problem with the mixer output.
In every mixer configuration with also 100% weight, the mixer sometimes stop at nearly 1800 out of 2000. Sometimes goes to 2000 sometimes 1800 without change any settings. In the receiver tab i see every time 2000.
Can someone help me?

MrD-RC, 2022-02-28:
By mixer, I’m assuming you mean servos. Are you in manual mode when testing this?

dataema92, 2022-02-28:
Yes, servos and in manual mode. I tried also to reflash the firmware and configure only the receiver and mixer tabs.

dataema92, 2022-02-28:
Sometimes I can see the value at 1800 and, without move the sticks, the value jump to 2000. A minute later i can go max to 1800 again

MrD-RC, 2022-02-28:
That sounds strange. I would expect it to be lower than 2000 in a stabilised mode. But manual should get the full 2000. It shouldn’t change uncommanded either. How close is your transmitter to the receiver? Any chance it’s getting swamped?

dataema92, 2022-02-28:
it's a brand new Matek H743 V2. the receiver is at 1 meter from the transmitter (tried also from more distant).
At this link you can see the problem.
https://youtu.be/igeWNospmBU

MrD-RC, 2022-03-01:
@dataema92 1m will be fine. To be honest 30cm would be fine. Some people have them right next to each other, so it's worth checking.

I don't see the jumping about you're talking about. I just see what I would kind of expect to see in Acro, Angle, or Horizon modes. I did see the centres jumping about on the receiver page, but this most likely means that you need to calibrate your gimbals.

dataema92, 2022-03-01:
> @dataema92 1m will be fine. To be honest 30cm would be fine. Some people have them right next to each other, so it's worth checking.
>
> I don't see the jumping about you're talking about. I just see [...]

The problem is that the servo output stop at around 1850 max and 1250 min.

dataema92, 2022-03-01:
[image]
This is the problem. It doesn't reach 2000 even as in the receiver tab is 2000.

MrD-RC, 2022-03-01:
Can you post up your modes page, hiding unused modes?

MrD-RC, 2022-03-02:
@dataema92 I think you are confusing modes here. Manual is not the same as Acro. If you select the MANUAL mode, you are in Manual and not Acro.

- To use Acro, you don’t have any other flight or navigation mode selectes. The Acro box at the bottom will light up. 
- To use Manual, you need  to have the MANUAL mode in an active range.

MrD-RC, 2022-03-02:
@dataema92 yes, which is working correctly. That’s why I asked if you were in manual when testing. 

You must have been in Acro. Only going to around 1800 is correct in Acro. You no doubt have the default settings on there and have not done an autotune yet. Once you do, you may see a little more movement in Acro, but not much. The 1800-2000 is room for the stabilisation to work.

closed 2022-03-02 by MrD-RC