iNavFlight/inav #7866: Logic control over Adjustments Tab

Jetrell, 2022-03-01:
## Current Behavior

Control over the `Adjustment` settings is defined by a `Slot`.

[image]

## Desired Behavior

Allow a Logic Condition to define if a specific `Adjustment` will operate or not.

## Who does this impact? Who is this for?

This is useful in the case that a user whats to share the two channels required for in-fight tuning, with other modes or PWM outputs.
It will allow the in-flight tuning adjustments to be disabled, so those channels can be used elsewhere, when the the in-flight OSD tuning page is not being used.
## Additional context

Presently, control can be taken away from a PWM mixed output, via a Logic Condition. So its half way there.
[image]

[image]

rts18, 2022-03-01:
Inflight tuning is a useful feature, but it wastes two channels. This should make it more appealing with limit available channels.

rts18, 2022-03-20:
@MrD-RC Can this Enhancement be added to inav? I would find it useful. At the moment I have to choose between keeping landing gear and flaps, over the inflight tuning.