iNavFlight/inav #7874: Matek 405 MiniTE

Grumpyrse, 2022-03-04:


FNNCFPV, 2022-03-05:
> 

When is that releasing?

Jetrell, 2022-03-05:
Around late June.

helicocrasher, 2022-03-16:
Would be nice to have it earlier. :-)

The F722 SE Mini is no more available and the F405 MiniTE is the best replacement.
Having it ready for maiden today with the Matek FW 4.1 build. 
One present issue with that is: entering the blackbox tab in the configurator crashes the FC. So no way to erase the the BB memory.

MATEKSYS, 2022-03-16:
@helicocrasher  fixed  http://www.mateksys.com/Downloads/FW/inav_4.1.0_MATEKF405TE.hex

helicocrasher, 2022-03-18:
@MATEKSYS 
another issue observed.
The OSD is not displaying at all the values that are supposed to flash.
For example altitude when above set alarm limit. Instead of flashing noting is displayed at all.
Any chance to reproduce and fix? :-)  
Thank you

OptimusTi, 2022-03-18:
Use
set display_force_sw_blink = ON

Some boards require this.

MATEKSYS, 2022-03-19:
No any target has "display_force_sw_blink" definitions.  All have it set to off by default in INAV fw

MrD-RC, 2022-03-19:
My point is more “Why not?” If all boards that use that target require it to be enabled (only one FC uses that target); why not add it to the target config? It would make life easier for the end user. 

The problem only starts when you have multiple FCs on the same target where one could require this, another doesn’t.

MATEKSYS, 2022-03-19:
> 

nothing to do with target,  if multiple FCs use the same target, no possible one require it, others don't. 
same theory for all other FC with other targets,  no possible F405-miniTE requires it,  others don't require it if they all don't have this definition in target.

MATEKSYS, 2022-03-19:
If need to set display_force_sw_blink = ON by default,  it is better to add this into INAV code for all targets

DzikuVx, 2022-04-20:
#7924

closed 2022-04-20 by DzikuVx