iNavFlight/inav #7878: Matek F405-WSE (3.0.1, 4.0, and 4.1) Only Enters Safe to Arm State With Battery Disconnected

clemmmdawg, 2022-03-06:
## Current Behavior
My Matek F405-WSE flight controller will only enter a safe to arm state when connected with USB and disconnected from battery voltage. Once the safe to arm state has been achieved, I am able to connect battery voltage, unplug the model from USB, arm, launch and fly. Below is the procedure I've come up with to get my model into an armed state and flying. 

## Steps to Reproduce
1. Flash firmware (I've tied 3.0.1, 4.0 and 4.1)
2. Configure plane ([diff](https://pastebin.com/71RcpW88))
3. Cold start, plug in battery
4. Wait for GPS lock (BN-220 blinks red when position locked)
5. "Unable to arm, waiting for GPS fix" displays in [OSD](https://i.imgur.com/eDF5Sot.jpg)
6. Plug in FC to USB, see "run-time calibration" and "navigation is safe" [pre-arming checks are failed](https://i.imgur.com/ebt4JPy.png)
7. Unplug battery, wait 5-10 seconds for "run-time calibration" and "navigation is safe" [pre-arming checks to have passed](https://i.imgur.com/rC9DyGu.png)
8. [Plug battery back in](https://i.imgur.com/mPU0HtH.png)
9. Disconnect from USB and arm

## Expected behavior
Model should enter safe to arm state with battery plugged in and have no need for USB connection.

## Additional context
A few Discord users suggested the problem may have to do with the HDOP value being too high. After some testing I was still able to reproduce the problem with an HDOP value between 1.5 and 2.0. This model has been flown in both GPS and stabilization modes with this convoluted arming procedure, so a hardware failure seems unlikely. 

[CLI Dump](https://pastebin.com/Dv5dp9Uv)

---

- FC Board name and vendor: INAV/MATEKF405SE 4.0.0 Dec 15 2021 / 14:57:16 (c20a9852)
- INAV version string: GCC-10.2.1 20201103 (release)

Jetrell, 2022-03-06:
I've experienced the same thing with one of my quads. Being that it has more electrical noise produced from the CRSF, VTX and camera, all being in close proximity to the GPS.
Getting a GPS lock by USB, prevents these devices from being powered. Which helps it pickup weaker satellites, that are drowned out by the noise of those devices.

There is a difference between getting 6 satellites for a general fix. And getting the required amount of satellites for a 3D fix.
It can sometimes require more than 8 Satellites to get a 3D fix in a noisy install.  While on a low electrical noise installation, a 3D fix can occur with only 4 satellites.
Without a 3D fix, the home position will not be recorded at Arming.

clemmmdawg, 2022-03-06:
The RX and GPS are both powered via 4v5 pins so they are on with either USB power or VBAT. My VTX is powered from the 8v pin that only turns on with VBAT. Physically, the RX and GPS are pretty close together, but both are on the opposite side of my model from the VTX. So the only thing that would interfere with the GPS would be the RX, both of which power on from USB. However, that is the only configuration I am able to arm in.

Getting a 3D fix is not the problem as far as I can tell. My [screenshot ](https://i.imgur.com/ebt4JPy.png) of the failed pre-arm checks shows a 3D fix.

Jetrell, 2022-03-06:
You can try setting `Nav_extra_arming_safety = BYPASS or OFF`
This will allow the FC to arm without a GPS fix. And may help you track down the problem, by seeing if it records the home position when it arms. 
This may prove if its a GPS issue, that is stopping it from arming.

breadoven, 2022-03-07:
It's the runtime calibration that's causing the problem, I'm guessing Gyro (or something related to Gravity possibly) failing to calibrate. This will cause the Navigation Safety issue (invalid Position status) and also the Manual mode issue since Manual mode is forced on until runtime calibration has completed.

Not sure why having the battery connected would cause the calibration issue however other than it being FC board related.

clemmmdawg, 2022-03-07:
> You can try setting `Nav_extra_arming_safety = BYPASS or OFF` This will allow the FC to arm without a GPS fix.

I'll give this a shot in the next day or two.

>also the Manual mode issue since Manual mode is forced on until runtime calibration has completed.

I was drawing the same conclusions, I just don't know enough about how INAV is coded to be sure. I have a feeling you're right about it being a problem with this specific board though since I have two others with zero issues. I guess if the nav_extra_arming_safety doesn't work, I'll be swapping out the FC.

clemmmdawg, 2022-03-18:
After a FC swap, everything worked fine. Must have been a hardware issue. Maybe we could get more verbose messages in the OSD or an error code alluding to the problem instead of "waiting for GPS fix" when navigation/runtime calibrations are unsafe. 

Thanks for the help!

closed 2022-03-18 by clemmmdawg