iNavFlight/inav #7879: Unable to get INAV to recognize VL53L0X sensor (I2C)

armsp, 2022-03-06:
## Current Behavior
Since the past week I have been trying everything to get INAV to recognize my VL53L0X sensor. On the i2c bus, I have a magnetometer and the VL53L0X tof sensor attached. The magnetometer works well, but inav shows red for the VL53L0X sonar sensor as the screenshot below shows -
[image]
Any idea how to ensure that inav recognizes it?

## Expected behavior
The sonar sensor is recognized and is blue in color instead of red.

## Setup Details
My flight controller is [Diatone Mamba Stack Basic F405 Mini MK3.5](https://www.team-blacksheep.com/products/product:5206)
```
# version
# INAV/MAMBAF405US_I2C 4.1.0 Feb 11 2022 / 10:39:20 (611d26f4)
# GCC-10.2.1 20201103 (release)
```

CONFIGURATOR -4.1.0
FC FIRMWARE4.1.0

DzikuVx, 2022-05-12:
Please verify that VL53L0X is configured to use `0x29` address

armsp, 2022-05-12:
@DzikuVx That helped...thanks so much Pawel! I get the sensor readings now.
On a side note, I want to attach a flight computer (R-Pi) to INAV FC, but it is not quite clear to me if INAV supports Mavlink or MultiWii and how to go about interfacing and sending commands...is there a Wiki or proper documentation for this?

closed 2022-05-12 by armsp

Killerbeans, 2024-11-21:
I'm having the same problem. I have connected a TOF 10120. Not detecting it. Where do I find the address for it? Thanks