iNavFlight/inav #7885: No dataflash chip found on Diatone Mamba MK4 H743 FC

FarmersFPV, 2022-03-09:
## Current Behavior
I have a new Diatone Mamba MK4 H743 FC that is supposed to have 128mb internal flash memory for blackbox logging.  I'm running INAV 4.1 (first time user of INAV), and it says in the top right corner that there is "No dataflash chip found".  Am I missing something as a new user?  Or is this a bug?

[image]

- FC Board name and vendor: Diatone Mamba MK4 H743 FC
- INAV version string: 4.1

dawr68, 2022-03-11:
Flash support is not currently added for this target

uralm1, 2022-03-16:
Small offtopic question on issue for MK4 fc, I tried to look into...
Do you have ADC external current meter working on this fc? It works in BF dev 4.3, but shows huge amps in inav 4.1. Not only huge amps, the values are inverted - more current -> gets lower values. Millivolts on current pad are increasing properly, so ESC is good.
My hw is: Mamba MK4 H743 + HobbyWing Xrotor 60A.
Or may be linked with c73cdc9e5759e6ca208e71b507edcc0e9aa4467e or bc4c8ffb79055d039305b72b47c90d4f7713c781

rblilja, 2022-05-30:
I'm +1 for adding dataflash support on this target :)

rblilja, 2022-05-30:
> Dataflash chip support is not currently added for this target

I'm +1 for adding dataflash support on this target :)

hogthrob, 2022-06-24:
> Small offtopic question on issue for MK4 fc, I tried to look into... Do you have ADC external current meter working on this fc? It works in BF dev 4.3, but shows huge amps in inav 4.1. Not only huge amps, the values are inverted - more current -> gets lower values. Millivolts on current pad are increasing properly, so ESC is good. My hw is: Mamba MK4 H743 + HobbyWing Xrotor 60A. Or may be linked with [c73cdc9](https://github.com/iNavFlight/inav/commit/c73cdc9e5759e6ca208e71b507edcc0e9aa4467e) or [bc4c8ff](https://github.com/iNavFlight/inav/commit/bc4c8ffb79055d039305b72b47c90d4f7713c781)

See #8146, should be fixed now.

JulioCesarMatias, 2022-07-04:
Solved https://github.com/iNavFlight/inav/pull/8166

closed 2022-07-04 by JulioCesarMatias