iNavFlight/inav #7888: Why do I need Pitch Stick to Arm Motors after Switch Arming

lasersbee, 2022-03-10:
This is my 1st Fixed wing iNav RC Build.
After 2 days of trying to figure this out I thought it was time
to ask people in the know.
When I arm my FC with a switch on my TX16S with bat power 
applied and no USB cable connected I need to push the Pitch
(elevator) stick away from me to get the Motor to spin.

Is there a reason for this and can I get rid of the required pitch
stick input?

Jerry

MrD-RC, 2022-03-10:
Sounds like a channel order mismatch. Have you check your transmitter moves the correct bars on the receiver page?

lasersbee, 2022-03-11:
Thanks for the reply.. Yes all the channels work as expected. On a Mini Talon.
Using the SF switch to arm. Can see it Arm under Modes in Configurator but
the motors wont start until I move the Pitch or Roll stick to any of 4 Max position.
Then the motor arms and runs.

Checked that only Ail, Elev, Throttle, Rudder, SF(arm), SC(modes) and SB(rates)
were used on the Radio.

Is there a CLI command that needs to be checked for proper setup?

Jerry

erstec, 2022-03-11:
> Thanks for the reply.. Yes all the channels work as expected. On a Mini Talon. I connected the USB cable to the F411-Wing and iNav Config 4.1.0. Using the SF switch to arm. I can see it Arm under Modes in Configurator but the motors wont start until in addition I move the Pitch or Roll stick to any of 4 Max position. Then the motor arms and runs.
> 
> Checked that only Ail, Elev, Throttle, Rudder, SF(arm), SC(modes) and SB(rates) were used on the Radio.
> 
> Is there a CLI command that needs to be checked for proper setup?
> 
> Jerry

Attach your diff/dump here

lasersbee, 2022-03-11:
Sorry... New to this. What is a Diff/Dump ?

jerry

erstec, 2022-03-11:
> Sorry... New to this. What is a Diff/Dump ?
> 
> jerry

Connect with Configurator, go to CLI tab, enter 'diff all' command (without quotes), press Enter and then save to file what you see in console window :)

MrD-RC, 2022-03-11:
> Thanks for the reply.. Yes all the channels work as expected. On a Mini Talon. I connected the USB cable to the F411-Wing and iNav Config 4.1.0. Using the SF switch to arm. I can see it Arm under [...]

Hi Jerry, it’s the output of the pitch and roll on the receiver page. Just want to make sure you don’t have any mixing on the transmitter.

lasersbee, 2022-03-11:
> > Sorry... New to this. What is a Diff/Dump ?
> > jerry
> 
> Connect with Configurator, go to CLI tab, enter 'diff all' command (without quotes), press Enter and then save to file what you see in console window :)

Here is the Diff Dump...
[INAV_cli_20220311_061433.txt](https://github.com/iNavFlight/inav/files/8231451/INAV_cli_20220311_061433.txt)

Jerry

erstec, 2022-03-11:
> > > Sorry... New to this. What is a Diff/Dump ?
> > > jerry
> > 
> > 
> > Connect with Configurator, go to CLI tab, enter 'diff all' command (without quotes), press Enter and then save to file what you see in console window :)
> 
> Here is the Diff Dump... [INAV_cli_20220311_061433.txt](https://github.com/iNavFlight/inav/files/8231451/INAV_cli_20220311_061433.txt)
> 
> @MrD-RC... No mixing of channels in the Radio or in iNav.
> 
> Jerry

With your config all is ok on my setup, not same FC but whatever. Arm->ThrottleUp->Motor spins
Looks like you have some mistmach on TX-RX part
What motor behaviour when you control it from Outputs tab?

lasersbee, 2022-03-11:
> With your config all is ok on my setup, not same FC but whatever. Arm->ThrottleUp->Motor spins Looks like you have some mistmach on TX-RX part What motor behaviour when you control it from Outputs [...]

On Outputs tab if I enable Master motor slider the motor works as expected.

Jerry

breadoven, 2022-03-11:
> Thanks for the reply.. Yes all the channels work as expected. On a Mini Talon. I connected the USB cable to the F411-Wing and iNav Config 4.1.0. Using the SF switch to arm. I can see it Arm under [...]

When you say it arms in Configurator, is the Arm box on the modes tab dark blue or light blue ? Needs to be light blue to show it's actually armed otherwise it just indicates the switch position is selected but the mode isn't actually active.

If you have a beeper what sound does it make when you try and arm ?

Also, what does it say regarding arming in the CLI tab if you enter "status" and hit enter ?

lasersbee, 2022-03-12:
> When you say it arms in Configurator, is the Arm box on the modes tab dark blue or light blue ? Needs to be light blue to show it's actually armed otherwise it just indicates the switch position is [...]

I'll answer in the sequence you asked..
1) When I use the Radio Switch only the ARM box goes light Blue after a sat fix.
    The Motor does not run with throttle. The receiver tab does show the throttle
    working.
    Going to the CLI and entering Status I get "Arming disabled flags : ARMSW CLI"
   
2) I don't have a beeper at this time.

3) If I then add a Pitch or Roll stick movement while (switch) armed the motor does work.
    While in this state and I go to the CLI... The CLI will not accept and input commands. 

Jerry

lasersbee, 2022-03-12:
Just added an LED to the F411-Wing FC Buzzer pads. Didn't want it
incessantly beeping in my ears. ;-)

1) Before a Sat Fix the LED flashes continually. When I get a Sat fix
    the LED stops flashing.

2) When I Arm by the radio switch only... the LED flashes 3 times with
    a small delay repetitively. No motor control with Throttle stick.

3) When I then add Pitch or Roll stick movement...  the LED stops Flashing 
    and I have Motor control with the Throttle Stick.

Jerry

MrD-RC, 2022-03-12:
Do you have launch mode permanently enabled? Moving the pitch/roll stick could be cancelling the launch mode, giving you use of the throttle.

lasersbee, 2022-03-12:
In the Configuration tab I had "Permanently enable launch mode for Fixed Wing"
enabled.
I disabled it and now I can't Arm at all and the ARM box in Modes is Dark Blue.
In CLI "status shows "Arming disabled flags :NAV ARMSW CLI

Jerry

breadoven, 2022-03-12:
I'm guessing it won't let you arm because a Nav mode is selected (this is why it lists NAV in status for arming disabled). Are any of Nav modes on switch SC selected in the modes tab ?

lasersbee, 2022-03-12:
I have RTH on the SC (Modes) switch.
I removed RTH in Modes and no change.
I then also Disabled "Permanently enable launch mode for Fixed Wing"
in Configurations. **Now it Arms with the SC Switch only**.

So how do I use RTH on my SC modes switch [Horizon-Manual-RTH] ?
And how do I auto Launch if I want to since I've Disabled "Permanently 
enable launch mode for Fixed Wing"

Jerry

breadoven, 2022-03-12:
You can have RTH on the switch but it can't be selected before you arm (all Nav modes must be off before arming).

When you have Launch permanently enabled it becomes active the moment you Arm. You would then launch (throw) the plane and the motor will start once the launch is detected by INAV. The plane will then be automatically controlled during the launch according to the launch settings you set, motor power, climb angle, max altitude etc. At the end of the launch you take control and fly normally. You can tell when launch is active by the sound the beeper makes. This is why it's much better to have a beeper fitted than an LED (which you probably can't see when you're actually trying to fly and not testing at the bench).

MrD-RC, 2022-03-12:
The real thing that needs to change is the arming ability when launch mode is permanently enabled. It should be no different to regular arming. You should never be able to arm in a nav mode, regardless of permanent autolaunch or not.

rts18, 2022-03-13:
> The real thing that needs to change is the arming ability when launch mode is permanently enabled. It should be no different to regular arming. You should never be able to arm in a nav mode, regardless of permanent autolaunch or not.

I understand for safety reasons. But what if a user wants their plane to enter RTH or Position hold, when it exits autolaunch?

breadoven, 2022-03-13:
@lasersbee  Should have clarified that you can Arm with any Nav mode active (except PosHold) IF Launch mode is active. Wasn't really sure what modes you had active when you tried before and couldn't Arm due to NAV. You might want to read the Wiki also, in particular:

https://github.com/iNavFlight/inav/wiki/%22Something%22-is-disabled----Reasons

MrD-RC, 2022-03-13:
> I understand for safety reasons. But what if a user wants their plane to enter RTH or Position hold, when it exits autolaunch?

Then they just do the same as when autolaunch is on a switch, following the standard procedure. 

1. Enable autolaunch if not permanently enabled
2. Arm
3. Select post launch mode (loiter for example)
4. Raise throttle to post launch throttle 
5. Throw the plane

lasersbee, 2022-03-13:
> @lasersbee Should have clarified that you can Arm with any Nav mode active (except PosHold) IF Launch mode is active. Wasn't really sure what modes you had active when you tried before and couldn't [...]

Thanks for the link... will be doing a bit more reading.
I want to use iNav for the OSD data when flying FPV and RTH Feature.

I've have used the Bigaole BGL-6G-AP 6-Axis Flight Controller With GPS 
many times and it arms the motors as soon as there is a Satellite Fix. And 
the RTH works flawlessly. 
I have a Throttle cut on a switch on my radio. No problems with that setup 
at all except there is no OSD Data available for FPV.

I understand that iNav is much more sophisticated and with many gotchas.
Would have been nice to be able to test the RTH functions on the bench.
I've even installed a GPS outdoors with the lines running back to the Bench
to do my testing. (my Shop is metal clad).

Will keep plugging away with my iNav setup. Since this thread I have a
much better understanding of my iNav Arm issues. 

Just want to say Thank You all for all your help.

Jerry

breadoven, 2022-03-13:
You won't be able to test any of the Nav modes except AltHold on the bench for a fixed wing because you need to have a valid heading available for the modes to activate. The heading comes from the GPS but requires forward speed to register a valid reading. Can't remember what happens on a fixed wing if you have a compass, not sure it's used even if available.

lasersbee, 2022-03-13:
> You won't be able to test any of the Nav modes except AltHold on the bench for a fixed wing because **you need to have a valid heading available for the modes to activate. The heading comes from the [...]

Good to know... Thanks

Jerry