iNavFlight/inav #7902: INAV - No OSD Needed - Compile Error

Digital256, 2022-03-14:
I have a project where I don't need OSD. When I disable all the OSD references in the target files I get an error compiling the code.

Compile error appears in src/main/rx/rx.c
      undefined reference to 'osdConfig_System'

The src/main/rx/rx.c code includes this.
       #include "rx/crsf.h"

The src/main/rx/crssf.c code includes this.
      #include "io/osd.h"

Also  src/main/rx/crssf.c has code using OSD for calculations.

Recommended Solution:
    #ifdef USE_OSD

I'm using the latest code from GitHub.
[image]

DzikuVx, 2022-03-15:
@Digital256 you are welcome to propose a pull request that fixes that. It's an open-source and seems like you know what you are doing :)

Digital256, 2022-03-15:
Will do.

PR #8860 (2023-03-23): Fix compilation if OSD is disabled
closes #7902
closed 2023-03-06 by DzikuVx