iNavFlight/inav #7910: RTH of INAV version 4.1.0 Error ?

doquangquan93, 2022-03-16:
Hello team dev
What happen RTH of INAV version 4.1.0 ? 
Why return to home not landing . That is too bad ...
The flight before RTH was successful.

My Video 
https://www.youtube.com/watch?v=rEG_pwh34eU

I using inav 4.1.0 
My  Diff 
# INAV/OMNIBUSF4V3_S5_S6_2SS 4.1.0 Feb 11 2022 / 11:23:48 (611d26f4)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# safehome

# logic
logic 0 1 -1 25 0 1 0 0 0
logic 1 1 -1 3 1 5 0 1300 0
logic 2 1 1 25 0 3 0 0 0

# gvar

# pid

# feature
feature THR_VBAT_COMP
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature PWM_OUTPUT_ENABLE

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS

# map

# serial
serial 0 2 115200 115200 0 115200
serial 5 64 115200 115200 0 115200
serial 30 2048 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1300
aux 1 1 2 1300 2100
aux 2 11 2 1700 2100
aux 3 10 1 1700 2100
aux 4 3 2 1700 2100
aux 5 13 1 1700 2100
aux 6 30 5 1700 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 8 11 H
osd_layout 0 1 1 12 V
osd_layout 0 2 0 0 V
osd_layout 0 3 8 6 V
osd_layout 0 4 8 6 V
osd_layout 0 6 23 0 V
osd_layout 0 7 14 12 V
osd_layout 0 8 18 10 H
osd_layout 0 9 23 12 H
osd_layout 0 10 22 1 V
osd_layout 0 11 7 12 V
osd_layout 0 12 18 12 V
osd_layout 0 13 3 5 V
osd_layout 0 14 19 11 V
osd_layout 0 15 22 6 V
osd_layout 0 19 3 10 H
osd_layout 0 20 1 1 V
osd_layout 0 21 1 0 V
osd_layout 0 22 15 11 V
osd_layout 0 23 13 1 V
osd_layout 0 24 13 2 V
osd_layout 0 25 28 5 H
osd_layout 0 26 23 7 V
osd_layout 0 28 12 2 H
osd_layout 0 30 2 10 V
osd_layout 0 31 25 11 H
osd_layout 0 32 1 11 V
osd_layout 0 33 23 12 V
osd_layout 0 34 10 9 H
osd_layout 0 35 2 5 H
osd_layout 0 36 3 10 H
osd_layout 0 38 3 9 H
osd_layout 0 40 13 0 V
osd_layout 0 41 24 10 H
osd_layout 0 42 24 9 H
osd_layout 0 46 24 11 H
osd_layout 0 47 24 11 H
osd_layout 0 50 23 4 H
osd_layout 0 85 3 6 V
osd_layout 0 96 22 5 V
osd_layout 0 104 25 11 H
osd_layout 0 105 27 11 H
osd_layout 0 106 24 11 H
osd_layout 0 109 24 11 V
osd_layout 0 110 7 11 V
osd_layout 0 111 25 11 H
osd_layout 0 124 24 11 H

# master
set looptime = 500
set gyro_main_lpf_hz = 110
set gyro_main_lpf_type = PT1
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 120
set setpoint_kalman_q = 200
set acc_hardware = MPU6000
set acc_lpf_hz = 102
set acczero_x = 135
set acczero_y = 2
set acczero_z = -262
set accgain_x = 4110
set accgain_y = 4108
set accgain_z = 4013
set align_mag = CW270FLIP
set mag_hardware = QMC5883
set magzero_x = -617
set magzero_y = 558
set magzero_z = 128
set maggain_x = 1346
set maggain_y = 1302
set maggain_z = 1342
set baro_hardware = BMP280
set serialrx_provider = CRSF
set blackbox_rate_denom = 2
set motor_pwm_rate = 16000
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = RTH
set align_board_pitch = 2
set vbat_scale = 1110
set current_meter_scale = 165
set current_meter_offset = 4750
set model_preview_type = 3
set small_angle = 180
set imu_acc_ignore_rate = 8
set imu_acc_ignore_slope = 5
set applied_defaults = 2
set gps_ublox_use_galileo = ON
set airmode_type = THROTTLE_THRESHOLD
set inav_reset_altitude = EACH_ARM
set nav_extra_arming_safety = OFF
set nav_user_control_mode = CRUISE
set nav_auto_speed = 400
set nav_auto_climb_rate = 200
set nav_land_maxalt_vspd = 100
set nav_rth_altitude = 2000
set osd_video_system = NTSC
set osd_time_alarm = 20
set osd_alt_alarm = 1000
set osd_crosshairs_style = TYPE7
set osd_sidebar_horizontal_offset = -1
set osd_home_position_arm_screen = OFF
set i2c_speed = 800KHZ
set dshot_beeper_tone = 5

# profile
profile 1

set mc_p_pitch = 50
set mc_i_pitch = 75
set mc_d_pitch = 30
set mc_p_roll = 55
set mc_i_roll = 60
set mc_d_roll = 32
set mc_p_yaw = 40
set mc_i_yaw = 80
set dterm_lpf_type = PT3
set heading_hold_rate_limit = 101
set nav_mc_vel_xy_p = 55
set d_boost_max =  1.500
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set smith_predictor_delay =  1.500
set tpa_rate = 20
set tpa_breakpoint = 1200
set rc_yaw_expo = 70
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# battery_profile
battery_profile 1

set throttle_idle =  7.000
set nav_mc_hover_thr = 1250

# end the command batch
batch end

breadoven, 2022-03-16:
A log would be useful, hard to say what happened otherwise.

doquangquan93, 2022-03-17:
> A log would be useful, hard to say what happened otherwise.

sorry I don't have blackbox . I will try test again and using blackbox

doquangquan93, 2022-03-17:
> A log would be useful, hard to say what happened otherwise.

I watched the dvr again. Baro is the problem? Baro doesn't work properly.

doquangquan93, 2022-03-17:
Maybe the error is coming from safehome . I don't use it . but is always enable 
[image]

RipperGOT, 2022-03-22:
Don't worry about safehome_max_distance, it's fine.

I don't have that problem.
I see a bit of toilet bowling in the video.
Correct mag alignment? From the sensor you have in the cli, I deduce that you use a BN-880Q, right? In these 3 (Bn-880Q , BN-880 and fake BN-880) I use CW90FLIP (Look at the position of the gps in the photo).
Another question, what is your nav_min_rth_distance?
Have you put some foam in the baro to avoid direct air from the propellers? 
Don't worry about safehome_max_distance, it's fine.

[image]

doquangquan93, 2022-03-23:
> Đừng lo lắng về safehome_max_distance, nó ổn.
> 
> Tôi không có vấn đề đó. Tôi thấy một chút trò chơi bowling trong nhà vệ sinh trong video. Căn chỉnh và hiệu chỉnh mag đúng ?? Từ cảm biến bạn có trong cli, tôi suy ra rằng bạn sử dụng BN-880Q và CW270FLIP, phải không? Trong 3 cái này (Bn-880Q, BN-880 và BN-880 giả) tôi sử dụng CW90FLIP (Nhìn vào vị trí của gps trong ảnh). Một câu hỏi khác, nav_min_rth_distance của bạn là gì? Bạn đã đặt một ít bọt vào baro để tránh không khí trực tiếp từ các cánh quạt chưa? Đừng lo lắng về safehome_max_distance, nó ổn.
> 
> [image]

Thank for answer
I using GPS Flywoo and mag setup oke!
After I set safehome_usage_mode = off . I no longer have that problem. I tried again so many times!
[image]

This is my test : https://www.facebook.com/groups/kythuatrcvietnam/posts/1637264429940436/

doquangquan93, 2022-03-23:
HI RipperGOT
Another question, what is your nav_min_rth_distance?
-> Using default settings
Have you put some foam in the baro to avoid direct air from the propellers?
-> Baro has a problem. then i disabled it

breadoven, 2022-03-23:
Is it definitely repeatable that with `safehome_usage_mode `= OFF then RTH works OK but with `safehome_usage_mode `= RTH it fails to loiter/land at the home point ? You don't have any Safehomes set in the Diff file so the `safehome_usage_mode `setting should be ignored and have no affect on RTH..

breadoven, 2022-03-30:
Can't help but think this caused by another case of RC input being applied during the RTH. Probably explains why the quad is flying at a funny angle during the initial phase of RTH shown the video. This will prevent INAV correctly homing in on the home point so it never gets there. It did seem to get within 1m of home but if that was actually 1.1m it wouldn't have flagged as having arrived. The quad was flown back to 1m of the home point in ACRO so manual flying seems likely. `nav_user_control_mode `= NAV_GPS_CRUISE together with stick input during the RTH will constantly redefine the desired position away from the home position preventing RTH from correctly targeting the home location. Also Safehomes can't cause this if none are defined, probably just coincidental.

In the absence of a log file and with no further info this is the most likely explanation so this issue should probably be closed. Only question is whether RC input should behave this way during RTH especially close to the home point ?

doquangquan93, 2022-03-31:
> Can't help but think this caused by another case of RC input being applied during the RTH. Probably explains why the quad is flying at a funny angle during the initial phase of RTH shown in the above video. This will prevent INAV correctly homing in on the home point so it never gets there. It did seem to get within 1m of home but if that was actually 1.1m it wouldn't have flagged as having arrived. The quad was flown back to 1m of the home point in ACRO so manual flying seems likely. `nav_user_control_mode `= NAV_GPS_CRUISE together with stick input during the RTH will constantly redefine the desired position away from the home position preventing RTH from correctly targeting the home location. Also Safehomes can't cause this if none are defined, probably just coincidental.
> 
> In the absence of a log file and with no further info this is the most likely explanation so this issue should probably be closed. Only question is whether RC input should behave this way during RTH especially close to the home point ?

I also don't know exactly what happened. after i set safehome_usage_mode = OFF . Everything is perfect. I don't have that problem anymore!
in the future if the problem persists i will upload to blackbox
Thanks for the reply!

closed 2022-06-23 by DzikuVx