iNavFlight/inav #7913: FC F765-WSE driver - IMU temperature sensor does not show any information.

jpci, 2022-03-19:
I fly without any problem with this controller but the temperature indicator of the IMU of the INAV OSD does not offer any data, it blinks as if it did not exist.
The barometer temperature value shows information but ussually  it is always much lower.
I have other matek fc's and this IMU indicator works correctly in all of them, being very indicative of the stress to which the FC is subjected.
I contacted Matek to find out if there was any problem with this new chipset and he referred me to this line of code, where the sensor shows "null".
https://github.com/iNavFlight/inav/blob/master/src/main/drivers/accgyro/accgyro_icm42605.c#L265 
I think it should already be working on it, if so, please close this case. Thanks for all this great work.
Regards,

Jetrell, 2022-03-30:
@MATEKSYS I can confirm too, that the IMU temperature is not being displayed on the F765SE variant.   

@jpci From my tests, The F765WSE, also appears to be the only FC I have, that MSC does not work on now either. Can you confirm this?

jpci, 2022-04-02:
Hello, please, can you explain a little more what is not working for you? I don't know what you mean by MSC. Thanks.

Jetrell, 2022-04-02:
> Hello, please, can you explain a little more what is not working for you? I don't know what you mean by MSC. Thanks.

MSC is _mass storage device class_, which allows the microSD to be accessed as an external drive. Rather than removing the SD card from the fight controller every time you need to retrieve log data.
Go to the CLI and type in MSC and Enter. Then it should reboot the FC, and a window for that drive should appear after after a few seconds.... But I can't get it to work with the F765-WSE.

jpci, 2022-04-10:
Hi, I don't usually use an sd card in the FC, with few exceptions. I just tried what you said and, without problem, I access the SD card with the MSC command.

Jetrell, 2022-04-10:
That's interesting. Mine comes up as a portable device, under the Device manager. But it doesn't see it as a drive.
I've also had many other issue with this FC. When using specific SD cards for logging.

What operation system are you using and what SD card speed and size?

jpci, 2022-04-10:
Hi, the operating system is windows 10 and the microsd is a generic 8gb and speed 10, with fat32 format.

jpci, 2022-04-18:
Hi, I'm using windows 10 and the SD its an ordinary Class 10,

vlamgat007, 2023-04-22:
> I fly without any problem with this controller but the temperature indicator of the IMU of the INAV OSD does not offer any data, it blinks as if it did not exist. The barometer temperature value [...]

HI.  I am having the same issue on H743-WLITE.  The IMU temp is flashing zero.  I will appreciate help fixing this.  Thanks!

vlamgat007, 2023-04-28:
> HI. I am having the same issue on H743-WLITE. The IMU temp is flashing zero. I will appreciate help fixing this. Thanks!

@jpci were you able to resolve this issue?  Thanks.

jpci, 2023-04-30:
> @jpci were you able to resolve this issue? Thanks.

Nope yet.
I begin to think that, due to cost reduction, Matek does not include this sensor in its IMU.  But frankly, it seems like a basic to me. Let me know if you get any more information.  Thank you

MATEKSYS, 2023-05-01:
> Nope yet. I begin to think that, due to cost reduction, Matek does not include this sensor in its IMU. But frankly, it seems like a basic to me. Let me know if you get any more information. Thank you

1. The IMU is not made by us. and ICM42688P made by TDK is not a cost reduction product.
2. ICM42688P has temperature sensor integrated, read the datasheet.  just INAV doesn't add its temperature sensor support in ICM42605 driver.

vlamgat007, 2023-05-02:
> datasheet. just INAV doesn't add its temperature sen

Thank you Matek.  I thought it might be a driver issue because the current firmware does not map to the exact chip.

jpci, 2023-05-02:
> 1. The IMU is not made by us. and ICM42688P made by TDK is not a cost reduction product.
> 2. ICM42688P has temperature sensor integrated, read the datasheet.  just INAV doesn't add its temperature [...]

Thanks for the information. 
You decide the components that you use in your FC's. 
And you also decide the support you offer to the community. 
This issue is more than a year old.
Thanks for your work in any case.

closed 2023-10-27 by DzikuVx

lwan288, 2024-01-03:
> Thanks for the information. You decide the components that you use in your FC's. And you also decide the support you offer to the community. This issue is more than a year old. Thanks for your work [...]

The issue hasn't been resolved yet, right?  I tried to change NULL to mpuTemperatureReadScratchpad and recompiled accgyro_icm42605.c, but it doesn't work.  : (

lwan288, 2024-01-03:
> Nope yet.
> I begin to think that, due to cost reduction, Matek does not include this sensor in its IMU. But frankly, it seems like a basic to me. Let me know if you get any more information. Thank you

Do you have the driver code to share?   

MPU6000 on MatekH743 V2 works perfectly, not sure why it is changed to ICM42688P.