iNavFlight/inav #7916: INAV 4.1 - Freeze during 45min flight.

evildronepilot, 2022-03-21:
I took my AR Wing pro for an endurance flight and as I was approaching 45 minutes in the air and at about 300-500m from home the aeroplane stopped responding to the stick commands while being in the Cruise mode (No failsafe - RSSI 99% ) 
It entered a right turn with the nose down attitude just to smack into the ground at about 100-120km/h 

## Current Behavior
1. The aeroplane was frozen. No response to any commands inc the flight mode switch.
2. RTH did not kick in and I do not believe  I lost the link. After the crash, all the electronic components are still working including servos, servo links, FC, GPS etc.

#The gear:
INAV 4.1 On Matek F405 Wing
R9MM ELRS  2.3.0 RX 
Edge TX 2.6 (stable - not nightly)

MrD-RC, 2022-03-21:
Any DVR or blackbox logs?

evildronepilot, 2022-03-21:
> 

Sorry... DJI goggles are not recording and I didn't use the blackbox in that flight. I know... It is very little to work with.

evildronepilot, 2022-03-24:
> Any DVR or blackbox logs?

Hi. Nop but...
I was investigating the hardware while rebuilding the plane and I found that the **GPS is not working**. 
The wing was in the cruise mode when it happened so the question is... 

**Can the GPS malfunction while in cruise mode cause the FC system to freeze?**

papage, 2022-03-28:
I don't know if this really worths anything, but today i was testing my 7" LR with inav (4.1) and while the drone was sitting disarmed on the ground, it froze. The OSD stopped updated (thats how i realized that it froze) and it was not responding to ANY command. I had to reboot it, in order to get it back. Sorry no logs or anything.

RoadyFPV, 2022-04-09:
maybe a kind of that problem? https://github.com/iNavFlight/inav/issues/7672

evildronepilot, 2022-04-09:
> maybe a kind of that problem? #7672
> 
> not only JETI EX BUS?

Not sure. 
I can't replicate the problem with my rig

evildronepilot, 2022-04-09:
After the investigation I narrowed down the possible reasons for the crash.

1. GPS failed mid flight what destabilised the fc and lead to the lock.
2. DJI vist running off the 9v regulator on 1000mw polling 1.6/1.8amp overhead the 9v Regulator and locked the FC 

I can't  verify  what would be the INAV behaviour In those cases but I would like to belive there should be  a safety  feature to prevent the scenario?

Also. 
I have completed 2 flight with new GPS and with vista on 12v.... no problems

MrD-RC, 2022-04-09:
How can the firmware protect from hardware issues? The DJI stuff, iNav could never deal with. That is down to the builder of the system. 

I have pulled the GPS before, and the firmware knows that the state has changed, and updates the OSD accordingly. I don’t know what happens if in a navigation mode. Maybe that could be investigated. It should just drop down to the next most logical mode, probably Angle in a navigation scenario. I would imagine if it falls through, its to Acro.

Ben-Stevenss, 2022-08-24:
If it looked like this i have the same problem.

https://youtube.com/shorts/PnNs6VeDLac?feature=share

evildronepilot, 2022-08-25:
> If it looked like this i have the same problem.
> 
> https://youtube.com/shorts/PnNs6VeDLac?feature=share
> 
> Only i have it every 2 or 3 flights, and it can happen after 10 min.

Can't see the stickers but... yeah, exactly this situation.

Ben-Stevenss, 2022-08-25:
Found a solution yet?

Ben-Stevenss, 2022-09-14:
My problem was the air unit, switched to 12v and al problems are gone.

closed 2023-10-26 by DzikuVx