iNavFlight/inav #7929: When using RTH with EXTRA, airplane is returning home at negative altitude

jlindrud, 2022-03-26:
## Current Behavior

I have RTH mode set to EXTRA, with RTH altitude set to 40m. So when RTH is initiated, the airplane immediately climbs 40m and then heads home. However, if RTH is initiated at an altitude of e.g. -100m (which is common at some spots where I fly), the airplane climbs to -60m, and then flies home level, at -60m. This obviously guarantees a crash, unless I take over.

## Steps to Reproduce
1. Set RTH mode to EXTRA and RTH altitude to 40m
2. Initiate RTH at -100m.
3. Aircraft climbs from -100m to -60m, and then continues flying home at -60m.

## Expected behavior
After initial climb, if aircraft is below RTH home altitude, aircraft should keep climbing until it reaches RTH home altitude.

---

- FC Board name and vendor:
Matek F411 Wing

- INAV version string:
INAV/MATEKF411_S5S6_SS2 4.0.0 Dec 20 2021 / 14:04:20 (GITDIR-N)

jlindrud, 2022-03-26:
The purpose of RTH is to bring the aircraft home though... The described behavior is 100% guaranteed to crash the aircraft. When would it ever make sense?

You can't specify a negative RTH altitude in Inav (for obvious reasons). If an aircraft is in RTH and finds itself below the RTH home altitude, I would think it obviously needs to climb, regardless of anything else.

MrD-RC, 2022-03-26:
I agree with @stronnag. That is how extra is intended to work. You should chose a different RTH method that is more suitable to your location, such as AT_LEAST. 

EXTRA would make sense if you are always going to be flying above your launch point, with no large obstacles; when current altitude + extra would clear any local obstacles.

closed 2022-03-26 by MrD-RC