iNavFlight/inav #7932: Unable to flash Matek 405std

Smyffy82, 2022-03-27:
## Current Behavior

## Steps to Reproduce

1.hold boot button
2.connect usb
3.firmware target Matek 405std
4.erase and flash board

## Expected behavior
 firmware flashed successfully without being stuck   #`#halfway`

## Suggested solution(s)

0crap, 2022-03-28:
It might help to give some more info.
- Did it work in the past?
- Which version do you try to flash?
- Which target do you select?
- Do you use Full chip erase?

There are no known issues flashing the Matek F405-STD
(Use one myself.)

Smyffy82, 2022-03-28:
MATEKF405STD
4.2.11 - 09-11-2021 20:48
Full chip erase
Manual baud rate 57600

Smyffy82, 2022-03-28:
Gets halfway through flashing…
Then at the top left “USB device successfully closed” but flashing yellow bar freezes

MrD-RC, 2022-03-28:
Have you tried putting the baud rate back to automatic?

Smyffy82, 2022-03-30:
Baud rate is set back to automatic but still freezes during the process 
It’s BetaFlight I’m trying to flash on to the board not Inav although this is probably the wrong place to post this it’s been suggested to me that I desolder everything off the board and start again to see if that works

MrD-RC, 2022-03-30:
If the issue is with BetaFlight, it has nothing to do with iNav. This should be posted on the [BetaFlight GitHub](https://github.com/betaflight/betaflight-configurator/issues).

closed 2022-03-30 by MrD-RC

Smyffy82, 2022-03-31:
[image]
#

MrD-RC, 2022-03-31:
Again, this is BetaFlight. Please post the issue on their GitHub, which I linked to in my last message. 

This has nothing to do with iNav.