iNavFlight/inav #7933: Add support for Holybro Kakute H7 Mini via target or other facility

thujones, 2022-03-28:
## Current Behavior
Test flashed Kakute H7 firmware onto H7 Mini board (and I recognize this isnt a supported thing to do, just testing to see if the H7 Mini version is even viable at all for iNav). All functionality on the surface appears to work, noting that the H7 full board has User1 and User2 mapped to Bluetooth and VTX Pit switch (not sure which order), whereas the H7 mini only has a pit switch on User1 in Betaflight.

Board initializes, serial data stream via MSP seems to work just fine. GPS and Receiver via UART works fine, along with I2C for magnetometer all appear to work just fine. Can calibrate the gyro and compass with success.

The detected issue for me occurs assigning USER1 (or USER2 since its defined as existing in the H7 Full target) and setting them on/off the PINIO appears mapped incorrectly and thus the VTX does not power up. I could bypass this by resoldering the power wire to the VTX however it would be nice to have this particular board supported.

Dumping out "resource" does show the H7 full config as defined in the code with what I expect to be the correct pin on the processer B11 being "free".

Also, in this configuration (and yes I realize the target is not specific to this board) the dataflash chip does not appear to be recognized, which leads me to believe various other board features may not be getting initialized/used. I expect this particular facet to be due to the actual H7 full size board using an SD card instead of a dedicated dataflash chip.

I am sure its been asked 1000x where resource remapping similar to Betaflight could potentially solve this problem, and normally I would approach it this way, so off to github I go to kindly tender this request.

Also, if there is some function I am missing to make "smaller brother" versions of a given board work properly, I am all ears.

Thanks!

## Desired Behavior
Implementation of Holybro Kakute H7 Mini for specific feature set (or CLI equivalent workaround) to allow use of USER1 via PINIO to control board's PIT Switch in addition to proper recognition of dataflash chip and any other potential features that are different from the full size Kakute H7

## Suggested Solution
Implementation of Holybro Kakute H7 Mini for board specific feature set

## Who does this impact? Who is this for?
Any user of Holybro Kakute H7 Mini board

## Additional context
It appears (and I do not profess to be a electronics expert by any means) in Betaflight this board has B11 mapped to be controlled via PINIO 1. I'm happy to provide any information needed.

thujones, 2022-04-06:
Update: I emailed HolyBro support and linked this to them, they acknowledged the request and indicate they will be reaching out at some point to facilitate getting this worked on

thujones, 2022-04-27:
[inav_4.1.0_KAKUTEH7MINI.zip](https://github.com/iNavFlight/inav/files/8573634/inav_4.1.0_KAKUTEH7MINI.zip)
[KAKUTEH7MINI.zip](https://github.com/iNavFlight/inav/files/8573667/KAKUTEH7MINI.zip)

I compiled my own version that got the USER1 working to power the VTX+ pads (although inverted and I couldnt figure out how to change that but whatever just updated the modes to accommodate), along with getting the blackbox flash working as well. Referenced the BF config of course to make sure all pins appear correct.

So while not official or sanctioned, attached the compiled hex in case anyone wants it, along with a zip of the files I used for config. If its suitable and some more developer-oriented person wants to potentially run with it, be my guest

DzikuVx, 2022-05-12:
There will be an official target soon. Thanks for contribution @thujones

PR #8033 (2022-05-16): Kakute H7 Mini target
closes #7933
closed 2022-05-16 by DzikuVx