iNavFlight/inav #7934: inav accel drift question

papage, 2022-03-28:
Hi all,

This is actually not a bug but a question.
I have problems with inav drifting in "angle" mode. I have calibrated the accel but still there is some significant drift.

What i don't really understand is why the accel is drifting in the first place. The accel is not like the gyro that can miss very slow movements and the differential error is added and magnified. The accel knows very well, where gravity points to, so if you do a one point calibration of the "horizontal level" of the drone, then you should not have any drifting problems in angle mode. At least that how i understand how accel works.

I would appreciate if someone could explain to me why we have this drifting problem of the accel, in angle mode.

Regards,
Sp

0crap, 2022-03-29:
Most of the time it's vibrations related. 
All UAV's have them, some more, some less.
The ACC can't detect a difference between drift of the UAV and vibrations.

Make sure your UAV is as low on vibrations as you can get it.

b14ckyy, 2022-03-29:
The Accellerometer is not directly keeping the craft level. 
ACC is only used as a reference to know where "down" is. All the orientation is mostly calculated only by gyro data. 
So not the accellerometer drifts but the virtual horizon drifts. 

vibration will induce more noise in the gyro and therefore it is harder to distinguish roll movements from noise. This can raise the gyro drift, an initial drift then makes the craft move to one direction causing accelleration forces the Accellerometer detects and causes the level to drift even more over timer. 

So yes vibration limiting helps but you will never get fully rid of the drifting until INAV gets a completely new AHRS system.

robustini, 2022-04-30:
I have a similar problem with 4.1 and am trying to figure out where it comes from.
Take off in Angle and the level is perfect, I switch to Acro and fly for about fifteen minutes, then I go back to Angle and the level is completely wrong, about 50 degrees of pitch forward and even a little bit of roll to the left.
I'm sorry but I have no log of this flight, the two images show the level in Angle at take off and when I switch after the flight in Acro.

[image]

[image]