iNavFlight/inav #7935: Can arm using the 10x arm switch method if there is an invalid parameter

MrD-RC, 2022-03-28:
If there is an invalid setting, such as the gyro_hardware_lpf. You can still arm and fly the model by disarming and arming 10 times, rapidly. I know that this is a shortcut to bypass the safety checks, such as the number of satellites. But surely it should still block arming if there are known invalid settings.

To test:
1. Use a current 4.1 release firmware and configurator
2. Perform the process to screw the config (Go to CLI, disconnect, reconnect, cycle USB, connect, disconnect, connect - cheat mode enabled). In CLI if you get `gyro_hardware`, you should have an invalid setting
3. Try to arm with switch - won't arm 👍
4. Try using allow_bypass: yaw right and  arm switch - won't arm 👍
5. disarm and arm quickly 10 times - it arms 👎

ThomasCr, 2022-05-09:
It's a question of the definition... When you sit in your room and want to test the esc with your remote it is all good.

MrD-RC, 2022-05-09:
> It's a question of the definition... When you sit in your room and want to test the esc with your remote it is all good.
I think you misunderstand. I'm not saying that you shouldn't be able to arm with the 10x arm toggle. Using that for testing is fine. 

What I am saying is that if there is an invalid setting, it should not arm regardless of the method. Not until the invalid setting is fixed. I've seen too many reports of people who can't arm, so do the 10x arm toggle, then fly. Next they crash and blame iNav. This is easy to do if system messages are not enabled or DJI is being used. People don't see the error, have plenty of sats, then force arm and fly.

b14ckyy, 2024-03-20:
still a thing?