iNavFlight/inav #7937: IRC TRAMP VTXs NOT working on SOFTSERIAL

flyingzzy, 2022-03-30:
**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

I use panda 5804M vtx on ominibus F4V3,with S6_SS fw,it uses irc tramp protocal .
I want to use osd to set vtx channel and power,but all 3 hardware serials have been used ,i have to connect it to softserial 2(pin:PWM 6),
but it cant work.If i switch it to uart 1/3,it works well.

## Steps to Reproduce

1.Flash F4V3 with S6_SS firmware
2.enable softserial feather in config page
3.config irc tramp on softserial in port page
4.connect vtx on pwm6 pin(s6)

## Expected behavior

Be able to use irc tramp vtxs on softserial S6 to set channel and power level.

## Suggested solution(s)

It seems to be a timing or hi/low level problem?

## Additional context

---

- FC Board name and vendor:ominibus F4V3
- INAV version string:4.1.0

Jetrell, 2022-03-30:
I'm using Soft Serial to run the Tramp protocol for a Matek VTX mini. It's worked fine for many releases.
I have the baud set to default 115200.

As you said, it may be a timing issue for that VTX.  Do you have another Tramp VTX, you could run a test with?

flyingzzy, 2022-03-30:
I have just monitored the waveform and data of the hard uart and soft uart ，with an oscilloscope and a logic analyzer. I found that the data of both uarts are the same. Hard serial ports work well, while with soft serial ports, fc and vtx can communicate with each other, but many data frames have a short low pulse before the start bit of the first byte, which can cause serial data parsing errors. If you can work properly with other vtx,  maybe the problem is the firmware of my vtx. I'll consult the manufacturer again.THKs

flyingzzy, 2022-04-01:
I fouund the glitch before the start bit of vtx data can last 3.2us，it's strange why there is no glitch on  hart uart port ?

erstec, 2022-04-02:
@flyingzzy do this 3.2us happens every packet? Is it on FC TX or came from VTX?

flyingzzy, 2022-04-03:
> @flyingzzy do this 3.2us happens every packet? Is it on FC TX or came from VTX?

not every packet. Most packets have a glitch less than 1us, only a few can last 3.2us.It always came from VTX.
Yestoday I tried to add a pull up resister of 200-700 ohms,but nothing changed.So strange.

erstec, 2022-04-03:
> > @flyingzzy do this 3.2us happens every packet? Is it on FC TX or came from VTX?
> 
> not every packet. Most packets have a glitch less than 1us, only a few can last 3.2us.It always came from VTX. Yestoday I tried to add a pull up resister of 200-700 ohms,but nothing changed.So strange.

So, F405 MCU based FC (same as yours), SoftSerial 1 configured as IRC Tramp, baudrate doesn't matter, as Tramp protocol works at 9600bps 8N1, TrampHV VTX connected (I just don't own any another Tramp protocol compatible VTX).
FC sends packet to VTX, after 40 ms VTX sends answer back (repeats every ~2 seconds)
>1 hour and I don't see anything wrong, 100% of packets are correct (from both ends) and parsed (by FC FW and same time by Data analyzer).

You metioned that you are using "**_ominibus F4V3,with S6_SS fw_**"
As I see this FC don't have SoftSerial 2
https://github.com/iNavFlight/inav/blob/c91d3a85106bf9e0d193f219f1ff8bc79a52512d/src/main/target/OMNIBUSF4/target.h#L161-L165
Explain this moment, please

flyingzzy, 2022-04-03:
> > > @flyingzzy do this 3.2us happens every packet? Is it on FC TX or came from VTX?
> > 
> > 
> > not every packet. Most packets have a glitch less than 1us, only a few can last 3.2us.It always came from VTX. Yestoday I tried to add a pull up resister of 200-700 ohms,but nothing changed.So strange.
> 
> So, F405 MCU based FC (same as yours), SoftSerial 1 configured as IRC Tramp, baudrate doesn't matter, as Tramp protocol works at 9600bps 8N1, TrampHV VTX connected (I just don't own any another Tramp protocol compatible VTX). FC sends packet to VTX, after 40 ms VTX sends answer back (repeats every ~2 seconds)
> 
> 1+ hour and I don't see anything wrong, 100% of packets are correct (from both ends) and parsed (by FC FW and same time by Data analyzer).
> 
> You metioned that you are using "**_ominibus F4V3,with S6_SS fw_**" and **_i have to connect it to softserial 2(pin:PWM 6)_** As I see this FC don't have SoftSerial 2
> 
> https://github.com/iNavFlight/inav/blob/c91d3a85106bf9e0d193f219f1ff8bc79a52512d/src/main/target/OMNIBUSF4/target.h#L161-L165
> 
> 
> Explain this moment, please

Thanks, everything you said is right( S6 is softserial 1, i typed wrong).
Today i did some tests:
1. I used a USB-UART converter to send freq and power control messages to the VTX,it works well.
2. I designed a digital filter using FPGA board,it can get rid of any glitches less than 5us, after fully test i connect it between the FC and VTX, the glitches are gone and the waveforms are very good. The FC can read back data from the VTX, but all setting messages just cant work.
3. I find the only difference is the interval between message bytes, on hardware uart the high level is ONE bit long, but on softserial it is THREE bits long. This may be a fw bug of this VTX only, i think.

Here are some pics from logic anylizer
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

flyingzzy, 2022-04-03:
I think there are 2 way to solve this problem:  reduce the interval to 1 bit time on the FC, or wait for the VTX manufactor to update its FW

erstec, 2022-04-03:
> I think there are 2 way to solve this problem: reduce the interval to 1 bit time on the FC, or wait for the VTX manufactor to update its FW I am not able to modify the inav code to change the interval for further tests, anyone could help?

We can try. But I need to understand clearly, what is a time measured between message bytes (I believe it start bit) on softresial and on hardware exactly?

P.S. I'm not sure if so deep chat is ok on GitHub Issue, we can move to discord if you prefer, till we "solve" that.

flyingzzy, 2022-04-03:
> > I think there are 2 way to solve this problem: reduce the interval to 1 bit time on the FC, or wait for the VTX manufactor to update its FW I am not able to modify the inav code to change the interval for further tests, anyone could help?
> 
> We can try. But I need to understand clearly, what is a time measured between message bytes (I believe it start bit) on softserial and on hardware exactly?
> 
> P.S. I'm not sure if so deep chat is ok on GitHub Issue, we can move to discord if you prefer, till we "solve" that.

One message contains 16 bytes, there are 15 intervals.  The interval on hard uart is just one stop bit(high level) , 104us@9600 baud. On softserial, the high level is 3 bits long(1 stop bit and 2 idle bits), about 312us

erstec, 2022-04-03:
> One message contains 16 bytes, there are 15 intervals. The interval on hard uart is just one stop bit(high level) , 104us@9600 baud. On softserial, the high level is 3 bits long(1 stop bit and 2 [...]

Ok, I see, I going deep to code, as I see now on softserial every byte are transmitted as separate parts, need to understand "where" timer is during those 300us. I will back to you after debugging.

erstec, 2022-04-03:
@flyingzzy done. Which INAV version do you prefer to test? 5.0.0 or 4.1.0? And what exact target you are using?

flyingzzy, 2022-04-03:
4.1.0 is OK, ominibus F4V3 S6 SS

erstec, 2022-04-03:
@flyingzzy check it, please, my measurements are 1.04us now. This "patch" will work only on bidirectional SoftSerial (IRC Tramp is)

[inav_4.1.0_OMNIBUSF4V3_S6_SS.hex.zip](https://github.com/iNavFlight/inav/files/8405241/inav_4.1.0_OMNIBUSF4V3_S6_SS.hex.zip)

flyingzzy, 2022-04-04:
> @flyingzzy check it, please, my measurements are 104us now. This "patch" will work only on bidirectional SoftSerial (IRC Tramp is)
> 
> [inav_4.1.0_OMNIBUSF4V3_S6_SS.hex.zip](https://github.com/iNavFlight/inav/files/8405241/inav_4.1.0_OMNIBUSF4V3_S6_SS.hex.zip)

Tested, very good!
Now it is 1 stop bit time(104us) between any message bytes on softserial, polling message and freq/power setting message all work well.
Can this patch be added to new versions?

P.S. I am not familar with github, when should i close this issue and what should i do next?

erstec, 2022-04-04:
> Tested, very good! Now it is 1 stop bit time(104us) between any message bytes on softserial, polling message and freq/power setting message all work well. Can this patch be added to new versions? [...]

Perfect, I am tested it on "genuine" TrampHV and SmartAudio 2.0 VTX too, it works ok, then I will create PR.

Issue be closed after code will be merged (or not) :)

flyingzzy, 2022-04-04:
:)
Since softserial is a half-duplex one wire bus, both sides should switch rx/tx direction frequently，that might generate glitches, and it will cause bad data on uart. So if possible, maybe it is a good idea to add a "giltch filter" in the sample code.

erstec, 2022-04-04:
> :) Since softserial is a half-duplex one wire bus, both sides should switch rx/tx direction frequently，that might generate glitches, and it will cause bad data on uart. So if possible, maybe it is a good idea to add a "giltch filter" in the sample code.

As, SoftSerial is low-speed alternative to communicate with non-mandatory peripherals and are just driven by timer it will be complicated and process will became much more CPU intensive. I don't think it is really needed till someone propore different implementation of softserial driver. I believe it is not problem if communication with for ex. VTX will be performed "not from first packet".

But you are free to propose code to improve it :) it opensource :)

erstec, 2022-04-04:
@DzikuVx I need your opinion on this case

I was done some SoftSerial  "stop-bit time" "fix" https://github.com/iNavFlight/inav/pull/7948 to make stop bit exactly 1 timer "wide", not 3x as currently is.
But this was affected some VTX'es, probably waiting for "longer" stop-bit.

I see only two ways to "solve" this issue: https://github.com/iNavFlight/inav/pull/7948#issuecomment-1087210996

What do you think?

flyingzzy, 2022-04-04:
I prefer the first way.
In fact, this problem is a bug of 5804 vtx, it should be corrected by the manufacture. Changing the stop bit from 3 to 1 is not a right way, it may be too strict for other devices.

erstec, 2022-04-05:
@flyingzzy 
maybe you can test this one?
Use `vtx_softserial_shortstop` CLI parameter, set it to `ON` to have 1x stopbit and to `OFF` to have 3x

[inav_4.1.0_OMNIBUSF4V3_S6_SS.hex.zip](https://github.com/iNavFlight/inav/files/8420067/inav_4.1.0_OMNIBUSF4V3_S6_SS.hex.zip)

flyingzzy, 2022-04-06:
@erstec 
OK，i will test it this weekend, its a good idea to add a para in cli

flyingzzy, 2022-04-11:
@erstec 
tested. It works good

erstec, 2022-04-11:
@flyingzzy is it works without your mentioned "FPGA-based filter"?

flyingzzy, 2022-04-11:
@erstec 
without other board. Just connect vtx with fc.

erstec, 2022-04-11:
Ok, as there is no suggestions I will propose PR

closed 2022-04-21 by flyingzzy